linrunner/TLP #117: `sudo tlp setcharge ` doesn't work in thinkpad x61

gansteed, 2015-01-18:
my system environment is Ubuntu 14.04 x64 with unity as desktop environment , when I type 

```bash
sudo tlp setcharge 80 95 BAT0
Setting temporary charge thresholds for BAT0:
  stop  = 95
  start = 80
Activating...done.
```

yes, it just works in the first 1 second, but after that, it is still charging.

linrunner, 2015-01-18:
Hi, 

it's a hardware problem. I read about this kind of quirk in the german ThinkPad forum. 

Try to power off to activate the thresholds after setcharge.

gansteed, 2015-01-20:
I've reboot my system, but it still doesn't work :(

linrunner, 2015-01-20:
Then your issue is a hardware problem that cannot be solved by software. I suggest you seek advice in a ThinkPad related forum.

gansteed, 2015-01-22:
All right, thanks anyway :+1:

closed 2015-01-22 by linrunner