linrunner/TLP #119: Add an option to run custom commands into configuration

Kuci, 2015-01-20:
This may be useful in a case when one would like to run a custom command or bunch of custom commands on profile change. For example, one uses acpi call and he would like to fully power off his discrete GPU.

linrunner, 2015-01-20:
Hi,

i'm not comfortable with potential stablity issues by foreign scripts running in the TLP environment. Neither am i keen on supporting people in writing proper scripts. So at the moment the answer is: no.

Kuci, 2015-01-20:
Okay, I understand. Then, what about making an option to write some stuff directly to /proc/acpi/call ?

ArchangeGabriel, 2015-01-20:
This could cause great damages if you wrote wrong things there, and it would require a dependency on acpi-call. Don’t like it personally, even if it’s optional. By the way, using acpi-call for disabling a discrete GPU is a bad idea, either use bbswitch alone or let nouveau do this for you.

Kuci, 2015-01-20:
TLP already depends on acpi-call for Sandy Bridge and newer ThinkPad battery functions. But if radeon does the same, then I do not need such an option.

closed 2015-01-20 by Kuci

linrunner, 2015-01-20:
> Then, what about making an option to write some stuff directly to /proc/acpi/call ?
No way.