linrunner/TLP #120: tlp 0.7 debian jessie

prysme01, 2015-02-02:
Hello, 

First of all, I would like to thank you for bring us TLP which is really great for laptop user.

I've just upgrade from 0.6 to 0.7 today on my debian jessie by using your release TLP0.7.tag.gz
I did a make install
then a systemctl enable tlp.service and a reboot

When checking tlp stat I'm getting this : 
***Warning: tlp.service is not enabled -- invoke "systemctl enable tlp.service" 
to correct this!
***Warning: tlp-sleep.service is not enabled -- invoke "systemctl enable tlp-sle
ep.service" to correct this!

But I did the 
systemctl enable tlp.service

And furthermore 
systemctl enable tlp-sleep.service
doesn't work because I saw on the developper page that this service is not automatically installed.

Is my setup ok ?
Do I really need the sleep service to be install and how ?

Thanks.
Prysme.

linrunner, 2015-02-02:
Hi,

maybe you missed the information that there is a Debian package repository for TLP since v0.6. Please refer to the [installation instructions](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#debian). Packages are the the only *supported* way to install TLP on Debian.

prysme01, 2015-02-02:
damn you're right, I missed it.

Now I'm just getting this warning x86_energy_perf_policy: program not installed.

linrunner, 2015-02-02:
It's not a warning just a notice.

Unlike Ubuntu Debian's package [linux-tools](https://packages.debian.org/jessie/linux-tools) doesn't provide a x86_energy_perf_policy binary. I suggest you file a bug report there.

closed 2015-02-02 by linrunner