linrunner/TLP #122: CPU_MAX_PERF_ON_BAT and CPU_BOOST_ON_BAT not respected during hardware accelerated video playback

sos-michael, 2015-02-05:
tlp conf:
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave

CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=25

CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0

System details:
Ubuntu 14.04.1 x64
"mainline" kernel 3.18.5
Dell XPS 13 L322x
Intel Core i7-3537U

os ondemand gov disabled by the following:
sudo update-rc.d -f ondemand remove 

So if you're on battery and do something CPU intensive like say sunspider

lscpu output looks like this the entire time and performance is clearly reduced:
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    2
Core(s) per socket:    2
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 58
Stepping:              9
CPU MHz:               800.000
BogoMIPS:              4988.58
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              4096K
NUMA node0 CPU(s):     0-3

But when running a hardware accelerated youtube video( Firefox 35.0.01,1080p, Flash ...using this method http://www.webupd8.org/2013/09/adobe-flash-player-hardware.html if it matters)
results in lscpu bouncing all over the place and actually reaching turbo states:

Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    2
Core(s) per socket:    2
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 58
Stepping:              9
CPU MHz:               2948.144
BogoMIPS:              4988.58
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              4096K
NUMA node0 CPU(s):     0-3

Let me know if you need anything else.

sos-michael, 2015-02-05:
Interestingly enough adding:

CPU_SCALING_MAX_FREQ_ON_AC=3100000
CPU_SCALING_MAX_FREQ_ON_BAT=1200000

Does not change the behavoir.

linrunner, 2015-02-05:
I *always* need the whole output please

    sudo tlp-stat

sos-michael, 2015-02-05:
/usr/sbin/tlp: 1: /etc/default/tlp: ------------------------------------------------------------------------------: not found
/usr/bin/tlp-stat: 1: /etc/default/tlp: ------------------------------------------------------------------------------: not found
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
 ------------------------------------------------------------------------------
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=3
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=25
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
DISK_DEVICES="sda"
DISK_APM_LEVEL_ON_AC="254"
DISK_APM_LEVEL_ON_BAT="1"
DISK_SPINDOWN_TIMEOUT_ON_BAT="1"
DISK_IOSCHED="noop"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
USB_AUTOSUSPEND=1
RESTORE_DEVICE_STATE_ON_STARTUP=1
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave

+++ System Info
System         = Dell Inc. XPS L322X
BIOS           = A10
Release        = Ubuntu 14.04.2 LTS
Kernel         = 3.18.5-031805-generic x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-3.18.5-031805-generic root=UUID=d1910d10-6445-4303-91d8-c2de735e82d7 ro intel_pstate=enable i915.enable_rc6=7 i915.enable_fbc=1 i915.semaphores=1 i915.fastboot=1 i915.disable_pch_pwm=0 i915.lvds_downclock=1 drm.vblankoffdelay=1 ath9k.ps_enable=1 pcie_aspm=force quiet splash vt.handoff=7
Init system    = upstart

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = Intel(R) Core(TM) i7-3537U CPU @ 2.00GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3100000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu1/cpufreq/scaling_max_freq  =  3100000 [kHz]

[... 99 lines cut ...]
+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
driver blacklist   = (not configured)
wwan blacklist     = disabled

Bus 004 Device 004 ID 0cf3:3004 control = auto, autosuspend_delay_ms =  2000 -- Atheros Communications, Inc.  (btusb)
Bus 004 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 004 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 003 ID 0c45:644d control = auto, autosuspend_delay_ms =  2000 -- Microdia  (uvcvideo)
Bus 003 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = SIMPLO
/sys/class/power_supply/BAT0/model_name                     = Dell
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   6390 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   3692 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2340 [mAh]
/sys/class/power_supply/BAT0/current_now                    =    919 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

linrunner, 2015-02-06:
TLP apparently applies the settings

    /sys/devices/system/cpu/intel_pstate/min_perf_pct = 25
    /sys/devices/system/cpu/intel_pstate/max_perf_pct = 25

(as for the min_perf_pct = 25 i reckon you changed the settings without **tlp start**).

This leaves us with two possibilities i think:

1. Hardware video acceleration is a GPU feature and intel_pstate is responsible for the CPU only
1. intel_pstate doesn't work as expected and you have to file a kernel bug report

None of these may be solved by TLP, sorry.

linrunner, 2015-02-06:
ps. did you monitor (and compare) the effective power consumption with

    watch -n 5 cat /sys/class/power_supply/BAT0/current_now

?

sos-michael, 2015-02-07:
Actually *perf_pct values less than 25 are ignored (perhaps a kernel related change and 25 corresponds to the lowest multiplier/voltage state on this machine)

I too think its a kernel issue actually, I was just looking for incidental insight on the matter.

power draw spike from 5-8W to upwards of 25W

sos-michael, 2015-02-07:
oddly enough, if i change them both to zero notice what happens!

/usr/sbin/tlp: 1: /etc/default/tlp: ------------------------------------------------------------------------------: not found
/usr/bin/tlp-stat: 1: /etc/default/tlp: ------------------------------------------------------------------------------: not found
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
 ------------------------------------------------------------------------------
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=3
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=0
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
DISK_DEVICES="sda"
DISK_APM_LEVEL_ON_AC="254"
DISK_APM_LEVEL_ON_BAT="1"
DISK_SPINDOWN_TIMEOUT_ON_BAT="1"
DISK_IOSCHED="noop"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
USB_AUTOSUSPEND=1
RESTORE_DEVICE_STATE_ON_STARTUP=1
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave

+++ System Info
System         = Dell Inc. XPS L322X
BIOS           = A10
Release        = Ubuntu 14.04.2 LTS
Kernel         = 3.18.5-031805-generic x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-3.18.5-031805-generic root=UUID=d1910d10-6445-4303-91d8-c2de735e82d7 ro intel_pstate=enable i915.enable_rc6=7 i915.enable_fbc=1 i915.semaphores=1 i915.fastboot=1 i915.disable_pch_pwm=0 i915.lvds_downclock=1 drm.vblankoffdelay=1 ath9k.ps_enable=1 pcie_aspm=force quiet splash vt.handoff=7
Init system    = upstart

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = Intel(R) Core(TM) i7-3537U CPU @ 2.00GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3100000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   800000 [kHz]
[... 101 lines cut ...]
+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
driver blacklist   = (not configured)
wwan blacklist     = disabled

Bus 004 Device 012 ID 0cf3:3004 control = auto, autosuspend_delay_ms =  2000 -- Atheros Communications, Inc.  (btusb)
Bus 004 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 004 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 003 ID 0c45:644d control = auto, autosuspend_delay_ms =  2000 -- Microdia  (uvcvideo)
Bus 003 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = SIMPLO
/sys/class/power_supply/BAT0/model_name                     = Dell
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   6390 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   3692 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2451 [mAh]
/sys/class/power_supply/BAT0/current_now                    =   1151 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

linrunner, 2015-02-07:
Doesn't happen here (with 3.19rc7):

    +++ System Info
    System         = LENOVO ThinkPad X220 42914CG
    Release        = Ubuntu 14.04.1 LTS
    Kernel         = 3.19.0-031900rc7-generic #201502020035 SMP Mon Feb 2 05:36:49 UTC 2015 x86_64
    
    +++ Processor
    CPU Model      = Intel(R) Core(TM) i7-2640M CPU @ 2.80GHz
    
    /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
    /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
    /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
    /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3500000 [kHz]    

    /sys/devices/system/cpu/intel_pstate/min_perf_pct      = 5
    /sys/devices/system/cpu/intel_pstate/max_perf_pct      = 24
    /sys/devices/system/cpu/intel_pstate/no_turbo          = 0

I think you should file a kernel bug report.

dywisor, 2015-02-07:
Unrelated to the issue at hand, but it seems that you've removed the ``#`` in the first line of ``/etc/default/tlp``. As a result, TLP tries to run this line as command and you get these warnings:
```
/usr/sbin/tlp: 1: /etc/default/tlp: ------------------------------------------------------------------------------: not found
/usr/bin/tlp-stat: 1: /etc/default/tlp: ------------------------------------------------------------------------------: not found
```

=> Change it back to
```
# ------------------------------------------------------------------------------
```

linrunner, 2015-02-07:
I now remember that this is not the first issue about difficulties in writing to intel_pstate sysfs --> https://github.com/linrunner/TLP/pull/115

pyamsoft, 2015-04-07:
Apologies for bumping a rather old issue.

Please note that the ability for the intel_pstate driver to control CPU frequencies has significantly improved in kernel versions 3.19

Also note from the kernel documentation about the intel_pstate driver:

*For contemporary Intel processors, the frequency is controlled by the
processor itself and the P-states exposed to software are related to
performance levels.  The idea that frequency can be set to a single
frequency is fiction for Intel Core processors. Even if the scaling
driver selects a single P state the actual frequency the processor
will run at is selected by the processor itself.*

Any values stated in the configuration file which are then read by intel_pstate will NOT be exactly used. pstate will try its best to hover around those values, but if the driver decides that more CPU power is needed it will scale as it sees fit.

Please also note that scaling_max_freq is reported as the Turbo frequency when using the intel_pstate driver, which may be the reason that even when no_turbo is set to 1, turbo frequencies are achieved.

This may explain why the output of lscpu is giving back weird values even when you have requested through TLP for lower frequencies. This appears to be, according to the intel_pstate documentation, by design.

linrunner, 2015-04-07:
Thanks for your hints.

closed 2015-04-07 by linrunner