linrunner/TLP #123: tlp-stat reports some warnings related to uninitialized variables access.

swojtasiak, 2015-02-08:
TLP Version: 0.7
System: Gentoo
Hardware: ThinkPad T430

Some general logs together with the "emerge --info" which should give you a general overview of the used system configuration:

**Log from tlp-stats (tlp-stats -s):**
>  ~tas  tlp-stat -s
--- TLP 0.7 --------------------------------------------

>+++ System Info
System         = LENOVO ThinkPad T430 2349EJ3
BIOS           = G1ET41WW (1.16 )
Kernel         = 3.17.8-gentoo-r1 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-3.17.8-gentoo-r1 root=/dev/sda7 ro thinkpad_acpi.fan_control=1
Init system    = openrc

>+++ System Status
TLP power save = enabled
power source   = battery

**lsusb**
>  ~tas  lsusb
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 046d:c03e Logitech, Inc. Premium Optical Wheel Mouse (M-BT58)
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**lsusb 2> /dev/null**
>  ~tas  lsusb 2> /dev/null 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 046d:c03e Logitech, Inc. Premium Optical Wheel Mouse (M-BT58)
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**ls -l /sys/bus/usb/devices**
>  ~tas  ls -l /sys/bus/usb/devices
total 0
lrwxrwxrwx 1 root root 0 Feb  8  2015 1-0:1.0 -> ../../../devices/pci0000:00/0000:00:1a.0/usb1/1-0:1.0
lrwxrwxrwx 1 root root 0 Feb  8  2015 1-1 -> ../../../devices/pci0000:00/0000:00:1a.0/usb1/1-1
lrwxrwxrwx 1 root root 0 Feb  8  2015 1-1:1.0 -> ../../../devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1:1.0
lrwxrwxrwx 1 root root 0 Feb  8  2015 1-1.2 -> ../../../devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2
lrwxrwxrwx 1 root root 0 Feb  8  2015 1-1.2:1.0 -> ../../../devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0
lrwxrwxrwx 1 root root 0 Feb  8  2015 2-0:1.0 -> ../../../devices/pci0000:00/0000:00:1d.0/usb2/2-0:1.0
lrwxrwxrwx 1 root root 0 Feb  8  2015 2-1 -> ../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1
lrwxrwxrwx 1 root root 0 Feb  8  2015 2-1:1.0 -> ../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1:1.0
lrwxrwxrwx 1 root root 0 Feb  8  2015 usb1 -> ../../../devices/pci0000:00/0000:00:1a.0/usb1
lrwxrwxrwx 1 root root 0 Feb  8  2015 usb2 -> ../../../devices/pci0000:00/0000:00:1d.0/usb2

**emerge --info**
>  ~tas  emerge --info
Portage 2.2.14 (python 2.7.9-final-0, default/linux/amd64/13.0/desktop, gcc-4.8.3, glibc-2.19-r1, 3.17.8-gentoo-r1 x86_64)
System uname: Linux-3.17.8-gentoo-r1-x86_64-Intel-R-_Core-TM-_i5-3320M_CPU_@_2.60GHz-with-gentoo-2.2
KiB Mem:     3766816 total,   2945100 free
KiB Swap:    2097148 total,   2097148 free
Timestamp of tree: Sat, 07 Feb 2015 22:15:01 +0000
ld GNU ld (Gentoo 2.24 p1.4) 2.24
app-shells/bash:          4.2_p53
dev-java/java-config:     2.2.0
dev-lang/perl:            5.18.2-r2
dev-lang/python:          2.7.9-r1, 3.3.5-r1
dev-util/cmake:           2.8.12.2-r1
dev-util/pkgconfig:       0.28-r1
sys-apps/baselayout:      2.2
sys-apps/openrc:          0.13.9
sys-apps/sandbox:         2.6-r1
sys-devel/autoconf:       2.13, 2.69
sys-devel/automake:       1.13.4
sys-devel/binutils:       2.24-r3
sys-devel/gcc:            4.8.3
sys-devel/gcc-config:     1.7.3
sys-devel/libtool:        2.4.4
sys-devel/make:           4.0-r1
sys-kernel/linux-headers: 3.16 (virtual/os-headers)
sys-libs/glibc:           2.19-r1
Repositories: gentoo tlp
ACCEPT_KEYWORDS="amd64"
ACCEPT_LICENSE="* -@EULA"
CBUILD="x86_64-pc-linux-gnu"
CFLAGS="-march=core-avx-i -O2 -pipe "
CHOST="x86_64-pc-linux-gnu"
CONFIG_PROTECT="/etc /usr/share/gnupg/qualified.txt"
CONFIG_PROTECT_MASK="/etc/ca-certificates.conf /etc/dconf /etc/env.d /etc/fonts/fonts.conf /etc/gconf /etc/gentoo-release /etc/revdep-rebuild /etc/sandbox.d /etc/terminfo"
CXXFLAGS="-march=core-avx-i -O2 -pipe "
DISTDIR="/usr/portage/distfiles"
FCFLAGS="-O2 -pipe"
FEATURES="assume-digests binpkg-logs config-protect-if-modified distlocks ebuild-locks fixlafiles merge-sync news parallel-fetch preserve-libs protect-owned sandbox sfperms strict unknown-features-warn unmerge-logs unmerge-orphans userfetch userpriv usersandbox usersync"
FFLAGS="-O2 -pipe"
GENTOO_MIRRORS="rsync://gentoo.prz.rzeszow.pl/gentoo http://gentoo.prz.rzeszow.pl"
LANG="en_US.utf8"
LDFLAGS="-Wl,-O1 -Wl,--as-needed"
MAKEOPTS="-j5"
PKGDIR="/usr/portage/packages"
PORTAGE_CONFIGROOT="/"
PORTAGE_RSYNC_OPTS="--recursive --links --safe-links --perms --times --omit-dir-times --compress --force --whole-file --delete --stats --human-readable --timeout=180 --exclude=/distfiles --exclude=/local --exclude=/packages"
PORTAGE_TMPDIR="/var/tmp"
PORTDIR="/usr/portage"
PORTDIR_OVERLAY="/var/lib/layman/tlp"
SYNC="rsync://rsync3.pl.gentoo.org/gentoo-portage"
USE="X a52 aac acl acpi alsa amd64 berkdb bindist bluetooth branding bzip2 cairo cdda cdr cli consolekit cracklib crypt cups cxx dbus dri dts dvd dvdr emboss encode exif fam firefox flac fortran gdbm gif glamor gpm gtk iconv ipv6 jpeg lcms ldap libav libnotify mad mmx mng modules mp3 mp4 mpeg multilib ncurses nls nptl ogg opengl openmp pam pango pcre pdf png policykit ppds pulseaudio qt3support qt4 readline sdl session spell sse sse2 sse4_1 sse4_2 ssl ssse3 startup-notification svg tcpd tiff truetype udev udisks unicode upower usb vorbis wxwidgets x264 xcb xml xv xvid zlib" ABI_X86="64" ALSA_CARDS="ali5451 als4000 atiixp atiixp-modem bt87x ca0106 cmipci emu10k1x ens1370 ens1371 es1938 es1968 fm801 hda-intel intel8x0 intel8x0m maestro3 trident usb-audio via82xx via82xx-modem ymfpci" APACHE2_MODULES="authn_core authz_core socache_shmcb unixd actions alias auth_basic authn_alias authn_anon authn_dbm authn_default authn_file authz_dbm authz_default authz_groupfile authz_host authz_owner authz_user autoindex cache cgi cgid dav dav_fs dav_lock deflate dir disk_cache env expires ext_filter file_cache filter headers include info log_config logio mem_cache mime mime_magic negotiation rewrite setenvif speling status unique_id userdir usertrack vhost_alias" CALLIGRA_FEATURES="kexi words flow plan sheets stage tables krita karbon braindump author" CAMERAS="ptp2" COLLECTD_PLUGINS="df interface irq load memory rrdtool swap syslog" CPU_FLAGS_X86="mmx sse sse2" ELIBC="glibc" GPSD_PROTOCOLS="ashtech aivdm earthmate evermore fv18 garmin garmintxt gpsclock itrax mtk3301 nmea ntrip navcom oceanserver oldstyle oncore rtcm104v2 rtcm104v3 sirf superstar2 timing tsip tripmate tnt ublox ubx" INPUT_DEVICES="synaptics evdev" KERNEL="linux" LCD_DEVICES="bayrad cfontz cfontz633 glk hd44780 lb216 lcdm001 mtxorb ncurses text" LIBREOFFICE_EXTENSIONS="presenter-console presenter-minimizer" OFFICE_IMPLEMENTATION="libreoffice" PHP_TARGETS="php5-5" PYTHON_SINGLE_TARGET="python2_7" PYTHON_TARGETS="python2_7 python3_3" RUBY_TARGETS="ruby19 ruby20" USERLAND="GNU" VIDEO_CARDS="intel" XTABLES_ADDONS="quota2 psd pknock lscan length2 ipv4options ipset ipp2p iface geoip fuzzy condition tee tarpit sysrq steal rawnat logmark ipmark dhcpmac delude chaos account"
Unset:  CPPFLAGS, CTARGET, EMERGE_DEFAULT_OPTS, INSTALL_MASK, LC_ALL, PORTAGE_BUNZIP2_COMMAND, PORTAGE_COMPRESS, PORTAGE_COMPRESS_FLAGS, PORTAGE_RSYNC_EXTRA_OPTS, USE_PYTHON

dywisor, 2015-02-08:
What's the output of:
```
grep "" /sys/bus/usb/devices/*/power/autosuspend_delay_ms
```

Does it also happen when you run tlp-usblist as root?

swojtasiak, 2015-02-08:
Here it is:

>  ~tas  rep "" /sys/bus/usb/devices/*/power/autosuspend_delay_ms 
zsh: no matches found: /sys/bus/usb/devices/*/power/autosuspend_delay_ms

Yes it does.

dywisor, 2015-02-08:
You forgot the asterisk/wildcard char ``*`` (or copy-paste messed it up?)

swojtasiak, 2015-02-08:
No, I forgot to escape it in the markdown :)

dywisor, 2015-02-08:
ok - your kernel lacks the necessary entries in /sys, might be related to config or custom patches.

You may also try:
```
find /sys/bus/usb/devices/*/. -path "*/power/autosuspend*"
```

swojtasiak, 2015-02-08:
Ok I've checked it and it gives no output at all.

Anyway, it seems that those are the only files with the autosuspend in their names:

>  tas  find /sys -name "*autosuspend*"
/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.2/1-1.2:1.0/supports_autosuspend
/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1:1.0/supports_autosuspend
/sys/devices/pci0000:00/0000:00:1a.0/usb1/1-0:1.0/supports_autosuspend
/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1:1.0/supports_autosuspend
/sys/devices/pci0000:00/0000:00:1d.0/usb2/2-0:1.0/supports_autosuspend

dywisor, 2015-02-08:
You'd usually find 500-1k files with this command :-o Make sure that ``CONFIG_PM_RUNTIME`` is set.

swojtasiak, 2015-02-08:
You're right. I missed this option when I was configuring the kernel. After enabling it and rebuilding the kernel everything works like a harm. It would be great it tlp-usblist/tlp-stat reported a kind of meaningful warning in such a situation :)

linrunner, 2015-02-08:
Thanks. Code is not hardened enough yet. I'll see into that.

Please keep that kernel for further testing please.

swojtasiak, 2015-02-08:
I will, thanks.

linrunner, 2015-02-12:
I need the output of

     find /sys/bus/usb/devices/*/. -path "*/power"

with your non-PM_RUNTIME Kernel.

swojtasiak, 2015-02-12:
Here you are:

>  ~tas  find /sys/bus/usb/devices/*/. -path "*/power"
/sys/bus/usb/devices/1-0:1.0/./ep_81/power
/sys/bus/usb/devices/1-0:1.0/./power
/sys/bus/usb/devices/1-0:1.0/./usb1-port1/power
/sys/bus/usb/devices/1-0:1.0/./usb1-port2/power
/sys/bus/usb/devices/1-0:1.0/./usb1-port3/power
/sys/bus/usb/devices/1-1/./1-1.2/1-1.2:1.0/ep_81/power
/sys/bus/usb/devices/1-1/./1-1.2/1-1.2:1.0/power
/sys/bus/usb/devices/1-1/./1-1.2/1-1.2:1.0/0003:046D:C03E.0001/input/input8/power
/sys/bus/usb/devices/1-1/./1-1.2/1-1.2:1.0/0003:046D:C03E.0001/input/input8/event6/power
/sys/bus/usb/devices/1-1/./1-1.2/1-1.2:1.0/0003:046D:C03E.0001/input/input8/mouse0/power
/sys/bus/usb/devices/1-1/./1-1.2/1-1.2:1.0/0003:046D:C03E.0001/power
/sys/bus/usb/devices/1-1/./1-1.2/1-1.2:1.0/0003:046D:C03E.0001/hidraw/hidraw0/power
/sys/bus/usb/devices/1-1/./1-1.2/ep_00/power
/sys/bus/usb/devices/1-1/./1-1.2/power
/sys/bus/usb/devices/1-1/./ep_00/power
/sys/bus/usb/devices/1-1/./power
/sys/bus/usb/devices/1-1/./1-1:1.0/1-1-port1/power
/sys/bus/usb/devices/1-1/./1-1:1.0/1-1-port2/power
/sys/bus/usb/devices/1-1/./1-1:1.0/1-1-port3/power
/sys/bus/usb/devices/1-1/./1-1:1.0/1-1-port4/power
/sys/bus/usb/devices/1-1/./1-1:1.0/1-1-port5/power
/sys/bus/usb/devices/1-1/./1-1:1.0/1-1-port6/power
/sys/bus/usb/devices/1-1/./1-1:1.0/ep_81/power
/sys/bus/usb/devices/1-1/./1-1:1.0/power
/sys/bus/usb/devices/1-1:1.0/./1-1-port1/power
/sys/bus/usb/devices/1-1:1.0/./1-1-port2/power
/sys/bus/usb/devices/1-1:1.0/./1-1-port3/power
/sys/bus/usb/devices/1-1:1.0/./1-1-port4/power
/sys/bus/usb/devices/1-1:1.0/./1-1-port5/power
/sys/bus/usb/devices/1-1:1.0/./1-1-port6/power
/sys/bus/usb/devices/1-1:1.0/./ep_81/power
/sys/bus/usb/devices/1-1:1.0/./power
/sys/bus/usb/devices/1-1.2/./1-1.2:1.0/ep_81/power
/sys/bus/usb/devices/1-1.2/./1-1.2:1.0/power
/sys/bus/usb/devices/1-1.2/./1-1.2:1.0/0003:046D:C03E.0001/input/input8/power
/sys/bus/usb/devices/1-1.2/./1-1.2:1.0/0003:046D:C03E.0001/input/input8/event6/power
/sys/bus/usb/devices/1-1.2/./1-1.2:1.0/0003:046D:C03E.0001/input/input8/mouse0/power
/sys/bus/usb/devices/1-1.2/./1-1.2:1.0/0003:046D:C03E.0001/power
/sys/bus/usb/devices/1-1.2/./1-1.2:1.0/0003:046D:C03E.0001/hidraw/hidraw0/power
/sys/bus/usb/devices/1-1.2/./ep_00/power
/sys/bus/usb/devices/1-1.2/./power
/sys/bus/usb/devices/1-1.2:1.0/./ep_81/power
/sys/bus/usb/devices/1-1.2:1.0/./power
/sys/bus/usb/devices/1-1.2:1.0/./0003:046D:C03E.0001/input/input8/power
/sys/bus/usb/devices/1-1.2:1.0/./0003:046D:C03E.0001/input/input8/event6/power
/sys/bus/usb/devices/1-1.2:1.0/./0003:046D:C03E.0001/input/input8/mouse0/power
/sys/bus/usb/devices/1-1.2:1.0/./0003:046D:C03E.0001/power
/sys/bus/usb/devices/1-1.2:1.0/./0003:046D:C03E.0001/hidraw/hidraw0/power
/sys/bus/usb/devices/2-0:1.0/./ep_81/power
/sys/bus/usb/devices/2-0:1.0/./power
/sys/bus/usb/devices/2-0:1.0/./usb2-port1/power
/sys/bus/usb/devices/2-0:1.0/./usb2-port2/power
/sys/bus/usb/devices/2-0:1.0/./usb2-port3/power
/sys/bus/usb/devices/2-1/./ep_00/power
/sys/bus/usb/devices/2-1/./power
/sys/bus/usb/devices/2-1/./2-1:1.0/ep_81/power
/sys/bus/usb/devices/2-1/./2-1:1.0/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port1/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port2/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port3/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port4/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port5/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port6/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port7/power
/sys/bus/usb/devices/2-1/./2-1:1.0/2-1-port8/power
/sys/bus/usb/devices/2-1:1.0/./ep_81/power
/sys/bus/usb/devices/2-1:1.0/./power
[... 28 lines cut ...]
/sys/bus/usb/devices/usb1/./power
/sys/bus/usb/devices/usb1/./1-0:1.0/ep_81/power
/sys/bus/usb/devices/usb1/./1-0:1.0/power
/sys/bus/usb/devices/usb1/./1-0:1.0/usb1-port1/power
/sys/bus/usb/devices/usb1/./1-0:1.0/usb1-port2/power
/sys/bus/usb/devices/usb1/./1-0:1.0/usb1-port3/power
/sys/bus/usb/devices/usb2/./2-1/ep_00/power
/sys/bus/usb/devices/usb2/./2-1/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/ep_81/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port1/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port2/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port3/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port4/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port5/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port6/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port7/power
/sys/bus/usb/devices/usb2/./2-1/2-1:1.0/2-1-port8/power
/sys/bus/usb/devices/usb2/./ep_00/power
/sys/bus/usb/devices/usb2/./power
/sys/bus/usb/devices/usb2/./2-0:1.0/ep_81/power
/sys/bus/usb/devices/usb2/./2-0:1.0/power
/sys/bus/usb/devices/usb2/./2-0:1.0/usb2-port1/power
/sys/bus/usb/devices/usb2/./2-0:1.0/usb2-port2/power
/sys/bus/usb/devices/usb2/./2-0:1.0/usb2-port3/power

linrunner, 2015-02-12:
So the directories are there but not the autosuspend* files.

Danke :-)

swojtasiak, 2015-02-12:
You're welcome :)

linrunner, 2015-02-12:
Please replace tlp-stat, tlp-usblist and tlp-pcilist with the new versions from [devel](https://github.com/linrunner/TLP/tree/devel) and show the full output of

    tlp-stat

swojtasiak, 2015-02-12:
Looks promising :)

```
>  ~tas  tlp-stat 
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/tlp.conf
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_DRIVER_BLACKLIST="usbhid"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO ThinkPad T430 2349EJ3
BIOS           = G1ET41WW (1.16 )
Kernel         = 3.17.8-gentoo-r1 #11 SMP Sat Feb 7 18:39:27 CET 2015 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-3.17.8-gentoo-r1.tlp root=/dev/sda7 ro thinkpad_acpi.fan_control=1
Init system    = openrc

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = userspace
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =  1200000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2601000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies = 2601000 2600000 2500000 2400000 2300000 2200000 2100000 2000000 1900000 1800000 1700000 1600000 1500000 1400000 1300000 1200000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = userspace
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =  1200000 [kHz]
[... 122 lines cut ...]
Bus 001 Device 003 ID 046d:c03e (autosuspend not available) -- Logitech, Inc. Premium Optical Wheel Mouse (M-BT58) (usbhid)
Bus 001 Device 002 ID 8087:0024 (autosuspend not available) -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 (autosuspend not available) -- Linux Foundation 2.0 root hub (hub)

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 45N1001
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  47520 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  44800 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =   8680 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   9006 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

tpacpi-bat.BAT0.startThreshold                              =     96 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

+++ Suggestions
* Reconfigure your Linux kernel with PM_RUNTIME=y to reduce your laptop's power consumption.
```

linrunner, 2015-02-12:
Yes :-)

closed 2015-08-12 by linrunner