linrunner/TLP #124: "/usr/sbin/tlp: 260: [: -eq: unexpected operator" When Recalibrating Battery

lasso012, 2015-02-08:
When I try to recalibrate my battery I get the following error:
```bash
$ sudo tlp recalibrate bat0
Currently discharging battery BAT0:
voltage            =   9698 [mV]
remaining capacity =     10 [mWh]
remaining percent  =      0 [%]
remaining time     =      0 [min]
power              =  11880 [mW]
Press Ctrl+C to cancel.

Done: battery BAT0 was completely discharged.
/usr/sbin/tlp: 260: [: -eq: unexpected operator
```

dywisor, 2015-02-08:
Typo in [tlp#L260](https://github.com/linrunner/TLP/blob/049978573dd8c01302f486309f598267c7cca3fe/tlp#L260): ``$rc`` should be ``$exitcode``

Cosmetical issue as it affects output messages only, recalibrating itself should work fine.

Cougar, 2015-03-20:
Why isn't this merged? Doesn't look very proffesional if master repo is intentionally borken more than month already. I don't think is it just a "cosmetical issue", even if it doesn't change the code behaviour (exit code of the script is still wrong). This is very imoprtant message for user that AC needs to remain connected. Not eveyone is able to read the code.

closed 2015-08-12 by linrunner