linrunner/TLP #125: Machine freezes on boot / panics on suspend

rmetrich, 2015-02-13:
Hi,
Since upgrading to latest TLP for ubuntu 12.04 (0.7-1~precise), my machine freezes quickly after Xsession comes up when booting on battery (no issue when on AC).
Also, when suspending while on battery, a kernel panic occurs.
[image]

These behaviours are gone when disabling TLP.
This is 100% reproducible.

-----------------------------
TLP package:

Package: tlp
Priority: optional
Section: utils
Installed-Size: 278
Maintainer: Thomas Koch <linrunner@gmx.net>
Architecture: all
Version: 0.7-1~precise
Recommends: tlp-rdw, ethtool, linux-tools, linux-tools-common, linux-tools-generic, smartmontools
Suggests: tp-smapi-dkms, acpi-call-dkms
Depends: hdparm, iw | wireless-tools, pciutils, pm-utils, rfkill, usbutils
Conflicts: laptop-mode-tools
Filename: pool/main/t/tlp/tlp_0.7-1~precise_all.deb
Size: 58922
MD5sum: 7a723085f47aad4d4bac1e659e362d45
SHA1: c0282f2c3c487d87d4497fe5d0f26e0f32d7c7fa
SHA256: 8d31a972723a6cf15004face647a56ff7e2022badcd54b47acec1aef6a562a39

-----------------------------
Stats while on AC:

--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=0
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
DISK_IOSCHED="cfq deadline"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=N
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
[... 426 lines cut ...]
+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
driver blacklist   = usbhid
wwan blacklist     = enabled (cdc_acm cdc_wdm cdc_ether hso qcserial sierra)

Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 002 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 009 ID 0cf3:3005 control = on,   autosuspend_delay_ms =  2000 -- Atheros Communications, Inc. AR3011 Bluetooth (btusb)
Bus 001 Device 004 ID 13d3:5122 control = auto, autosuspend_delay_ms =  2000 -- IMC Networks  (uvcvideo)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTek
/sys/class/power_supply/BAT0/model_name                     = N61--52
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  57200 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  54142 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  54296 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  17919 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

linrunner, 2015-02-13:
Hi, try with this setting:

    RUNTIME_PM_ALL=0

rmetrich, 2015-02-13:
Thanks, this works!

linrunner, 2015-02-13:
I suspect the discrete part of your hybrid graphics (Nvidia?):

     /sys/bus/pci/devices/0000:01:00.0/power/control = auto (0x030000, VGA compatible controller, no driver)

Unfortunately it has no driver loaded, so TLP can't blacklist it automatically. Please try again with:

    RUNTIME_PM_ALL=1
    RUNTIME_PM_BLACKLIST="03:00.0"

rmetrich, 2015-02-13:
Doesn't help (after replacing 03:00.0 by the right id 01:00.0)

linrunner, 2015-02-13:
I see. You may a. revert to RUNTIME_PM_ALL=0 for the sake of simplicity or b. try the other PCI devices in RUNTIME_PM_BLACKLIST one by one.

closed 2015-02-23 by linrunner