linrunner/TLP #126: udev-rules incompatibility with Microsoft Kinect

gabm, 2015-02-17:
I experienced a constant loop of usb connects/disconnects from the Microsoft Kinect device.

According to the issue described in [1] it seems to be related to TLP's udev rules... As a fast fix, I removed TLP which solved the problem for now...

The VIDs and PIDs of the Kinect are:

$ lsusb
Bus 001 Device 006: ID 045e:02bf Microsoft Corp. 
Bus 001 Device 005: ID 045e:02be Microsoft Corp. 
Bus 001 Device 003: ID 045e:02c2 Microsoft Corp.

If you need any more information and/or debuggung, please contact me!

[1] https://github.com/OpenKinect/libfreenect/issues/348

linrunner, 2015-02-17:
Hi,
1. Please show **tlp-stat -u**
2. Does blacklisting all three devices help?
3. Does TLP_ENABLE=0 help?
4. Does *libfreenect* itself provide any udev rules?

gabm, 2015-02-21:
**1. Please show tlp-stat -u**
gabm@gabm-PC ~ % tlp-stat -u
--- TLP 0.7 --------------------------------------------

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
driver blacklist   = usbhid
wwan blacklist     = enabled (cdc_acm cdc_wdm cdc_ether hso qcserial sierra)

Bus 002 Device 003 ID 05c6:9205 control = on,   autosuspend_delay_ms =  2000 -- Qualcomm, Inc.  (qmi_wwan, qcserial)
Bus 002 Device 002 ID 8087:0020 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 003 ID 17ef:4816 control = auto, autosuspend_delay_ms =  2000 -- Lenovo  (uvcvideo)
Bus 001 Device 038 ID 045e:02c2 control = auto, autosuspend_delay_ms =     0 -- Microsoft Corp.  (hub)
Bus 001 Device 002 ID 8087:0020 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

**2. Does blacklisting all three devices help?**
yes

**3. Does TLP_ENABLE=0 help?**
yes

**4. Does libfreenect itself provide any udev rules?**
yes -> https://github.com/OpenKinect/libfreenect/blob/master/platform/linux/udev/51-kinect.rules

linrunner, 2015-02-21:
> 2. Does blacklisting all three devices help? yes

So, the issue here is: the kinect devices don't like autosuspend – something to be expected from surveillance hardware, isn't it?

The preferred solution for TLP is

     USB_BLACKLIST="045e:02bf 045e:02be 045e:02c2"

or am i missing something here? 

Of course you could try to throw a kernel bug report.

gabm, 2015-02-21:
ok, i suppose the autosuspend blacklist is the "right" solution here - its a hardware problem, because the kinect doesn't like usb autosuspend...

I dont think its a kernel bug, so no bug report necessary ... also its the "old" kinect, i dont know about kinect2!?

Whats wrong with mode 666 for a usb device? is it a security flaw? I dont know that much about udev, what is it supposed to be? maybe we should file a bug report or commit a better solution? 

Thanks linrunner for caring!

linrunner, 2015-02-21:
For the sake of perfection you could try to leave the hub device out of the blacklist

     USB_BLACKLIST="045e:02bf 045e:02be"

I'm not aware of any hub that doesn't support autosuspend.

OT: in this case 666 means the raw device is world writable i.e. all active users in your running system can access it, which i suppose is not secure enough for this kind of device. You won't find many raw devices under /dev allowing users even read access.

closed 2015-02-21 by linrunner