linrunner/TLP #127: Most Likely a bug with TLP and Broadwell CPU's (SMAP Feature)

TrevorScheitrum, 2015-02-19:
Hello, I installed TLP on a new work machine, a Dell Latitude e7450 with Broadwell, and it stopped Mint 17.1 from booting correctly, I had to boot to the recovery console and apt-get remove tlp for it boot to desktop.

I would suspect its related to this bug I just detected and logged with the guys over at Virtualbox here: https://www.virtualbox.org/ticket/13820 which stopped me from being able to boot virtual machines.

"Broadwell CPUs introduce a new feature called 'supervisor mode access prevention' which allows to restrict kernel code from directly accessing usermode code. The intention is to catch bugs in kernel code because normally the kernel would access user-level memory only under rare conditions (mostly when copying parameters for system calls from/to user memory prior/after the system call is executed)." 

i can get you guys any type of logs you may need, just tell what you need and how to obtain them.

linrunner, 2015-02-19:
Hi,  please try to boot with

    TLP_ENABLE=0

TrevorScheitrum, 2015-02-19:
Will do!  I can report back with results at the end of the work day

TrevorScheitrum, 2015-02-19:
Sorry, this wasn't related to what I thought it was. Since TLP interacts with the CPU, i thought it might be related to the virtualbox bug I had with a broadwell CPU, but i narrowed it down to this line

SATA_LINKPWR_ON_BAT=min_power

since i'm using an SSD, i assume that it wasn't providing enough power and the hard drive was shutting down.   Changing it to medium_power seems to be okay, but i'll probably leave it at max_performance, just to be safe.

Thanks for the help, and sorry for a false alarm.

closed 2015-02-19 by TrevorScheitrum

linrunner, 2015-02-20:
Thanks for the info. One should *never* draw conclusions in the subject of a bug report ...