linrunner/TLP #128: btrfs and TLP

owerner, 2015-02-22:
I am using btrfs on archlinux 3.18.* and have massive issues with TLP. After installing TLP and rebooting it a few times I get random file corruptions until my filesystem breaks completely. I cant supply any logs since I needed to reinstall my system every time (maybe this could be done, but I am using LUKS so, after some file curruptions I cant open the container) but I am pretty sure its connected to TLP because it brokge my system a few times. (Took me some time to figure out its TLP). 

This error is reproducable maybe even in a VM. 

Does anybody know anything about this issue?

linrunner, 2015-02-22:
Hi, sorry to hear that. 

On physical hardware SATA ALPM is a probable cause (on battery power for the default config). I would change the config immediately after install

    SATA_LINKPWR_ON_BAT=max_performance 

and [check for SATA Errors](http://linrunner.de/en/tlp/docs/tlp-faq.html#warnings) – eventually recheck with more aggressive settings.

> This error is reproducable **maybe** even in a VM. 

Maybe?! This would be a mystery. TLP doesn't touch any btrfs related settings. The only filesystem related settings are shown in tlp-stat output, "File System" section.

Please show full

    tlp-stat

output before the first reboot. You may start TLP without reboot:

   tlp start

owerner, 2015-02-26:
Hey, thank you for your help. Luckily I had a backup....

I changed the config file and disabled SATA_LINKPWR (commeted it out), just to be sure. 

```
# SATA aggressive link power management (ALPM):
#   min_power, medium_power, max_performance
#SATA_LINKPWR_ON_AC=max_performance
#SATA_LINKPWR_ON_BAT=min_power
```

`tlp-stat` before first reboot/first start of tlp:

```
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_DRIVER_BLACKLIST="usbhid"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO ThinkPad X240 20AMS21C00
BIOS           = GIET77WW (2.27 )
Release        = "Arch Linux"
Kernel         = 3.18.6-1-ARCH x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-linux root=UUID=9446519c-1fe2-438a-b940-1d1d596bc00f rw rootflags=subvol=__active cryptdevice=/dev/sda5:crypt
Init system    = systemd

+++ System Status
TLP power save = enabled
***Warning: tlp.service is not enabled -- invoke "systemctl enable tlp.service" to correct this!
***Warning: tlp-sleep.service is not enabled -- invoke "systemctl enable tlp-sleep.service" to correct this!
power source   = ac

+++ Processor
[... 147 lines cut ...]
/sys/class/power_supply/BAT0/energy_full                    =  22270 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  21730 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown

tpacpi-bat.BAT0.startThreshold                              =     96 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = SANYO
/sys/class/power_supply/BAT1/model_name                     = 45N1737
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  71280 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  68230 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  67770 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown

tpacpi-bat.BAT1.startThreshold                              =     96 [%]
tpacpi-bat.BAT1.stopThreshold                               =    100 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0
```

I just started tlp like you suggested with `tlp start` but I did not reboot yet. Now I wait and reboot a few times. Lets see what happens.....

linrunner, 2015-02-26:
Your install is not finished yet:

> ***Warning: tlp.service is not enabled -- invoke "systemctl enable tlp.service" to correct this!
> ***Warning: tlp-sleep.service is not enabled -- invoke "systemctl enable tlp-sleep.service" to correct

Please enable the services. See https://wiki.archlinux.org/index.php/TLP too.

owerner, 2015-03-03:
I forgot to mention it, but of course I started the systemd service as explained. 

After a couple of days I can say for sure that my system is still running fine, thanks to your help. 

I changed the arch wiki and added a part about btrfs. Maybe it helps other people. 

Thank you and keep up the good work.

linrunner, 2015-03-03:
You're welcome.

closed 2015-03-03 by linrunner

strelec, 2015-10-31:
Was there any extended research as to why this has happened? It seems like that from now on, everybody with btrfs disables power saving, just out of caution.

In my case, the system power usage is about 4W higher and I am not sure whether I should turn the option back on.

strelec, 2015-11-30:
@m0nk33: Can you post the output of `lspci`?

m0nk33, 2015-12-03:
``` 
00:00.0 Host bridge: Intel Corporation Haswell-ULT DRAM Controller (rev 0b)
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
00:03.0 Audio device: Intel Corporation Haswell-ULT HD Audio Controller (rev 0b)
00:14.0 USB controller: Intel Corporation 8 Series USB xHCI HC (rev 04)
00:16.0 Communication controller: Intel Corporation 8 Series HECI #0 (rev 04)
00:16.3 Serial controller: Intel Corporation 8 Series HECI KT (rev 04)
00:19.0 Ethernet controller: Intel Corporation Ethernet Connection I218-LM (rev 04)
00:1b.0 Audio device: Intel Corporation 8 Series HD Audio Controller (rev 04)
00:1c.0 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 6 (rev e4)
00:1c.1 PCI bridge: Intel Corporation 8 Series PCI Express Root Port 3 (rev e4)
00:1d.0 USB controller: Intel Corporation 8 Series USB EHCI #1 (rev 04)
00:1f.0 ISA bridge: Intel Corporation 8 Series LPC Controller (rev 04)
00:1f.2 SATA controller: Intel Corporation 8 Series SATA Controller 1 [AHCI mode] (rev 04)
00:1f.3 SMBus: Intel Corporation 8 Series SMBus Controller (rev 04)
02:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS5227 PCI Express Card Reader (rev 01)
03:00.0 Network controller: Intel Corporation Wireless 7260 (rev 83)
 ```

m0nk33, 2016-04-19:
Is this fixed yet? 
FYI:
https://www.reddit.com/r/archlinux/comments/4f5xvh/saving_power_is_the_btrfs_dataloss_warning_still/

linrunner, 2016-04-19:
TLP can't fix this. It's a kernel issue.

strelec, 2016-04-19:
Does anybody know to what extent this applies to the Core 2 Duo era of laptops?

datamaan, 2016-04-30:
Matthew Garrett - <https://mjg59.dreamwidth.org/42156.html>:

> First, these probably only have relevance to users of mobile Intel parts in the U or S range (/proc/cpuinfo will tell you - you're looking for a four-digit number that starts with 4 (Haswell), 5 (Broadwell) or 6 (Skylake) and ends with U or S), and won't do anything unless you have SATA drives (including PCI-based SATA).

ActualizeInMaterial, 2016-06-07:
Note: this workaround is not required on Lenovo T400 (Intel(R) Core(TM)2 Duo CPU     T9400  @ 2.53GHz)

Tested via powertop (put everything from Bad into Good mode, while on AC not battery - does it matter? also did re-run `tlp start` with `SATA_LINKPWR_ON_AC=min_power` prior and even after that) sporting btrfs root(I've nothing else, other than swap) on kernel Linux mighty4 4.5.6-1-MANJARO #1 SMP PREEMPT Thu Jun 2 17:33:22 UTC 2016 x86_64 GNU/Linux
The disk is an 80G SSD INTEL SSDSA2M080G2GC  Firmware Revision:  2CV102M3
	Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6

```
processor	: 1
vendor_id	: GenuineIntel
cpu family	: 6
model		: 23
model name	: Intel(R) Core(TM)2 Duo CPU     T9400  @ 2.53GHz
stepping	: 6
microcode	: 0x610
cpu MHz		: 800.000
cache size	: 6144 KB
physical id	: 0
siblings	: 2
core id		: 1
cpu cores	: 2
apicid		: 1
initial apicid	: 1
fpu		: yes
fpu_exception	: yes
cpuid level	: 10
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx lm constant_tsc arch_perfmon pebs bts nopl aperfmperf pni dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm sse4_1 lahf_lm tpr_shadow vnmi flexpriority dtherm
bugs		:
bogomips	: 5056.78
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:
```

Konfekt, 2017-05-01:
The article http://refugeeks.com/use-tlp-to-optimize-the-power-consumption-in-ubuntu/ tells us that Ubuntu disables ALPM by default because it causes data corruption on hard discs. This thread confirms it. Perhaps TLP should keep it safe as well and disable it by default? If not, warn the user.

linrunner, 2017-05-02:
@Konfekt : are you affected or just asking?

tlp-stat issues a [warning message](https://github.com/linrunner/TLP/blob/0.9/tlp-stat.in#L1240) when it detects related SATA errors.

Konfekt, 2017-05-03:
I might have been affected since the BTRFS and to a lesser extent XFS-file systems of my hard disc  were being continually corrupted. I was also extensively using hibernation.

Since switching to EXT4 the problem did not reoccur.

linrunner, 2017-05-03:
Did you check for SATA error warnings with 

    tlp-stat

or

    tlp-stat -w

?

Konfekt, 2017-05-03:
Yes, no errors at all at any point. Then suddenly an error of the form `parent transid verify failed` at which point a re-installation is the only viable path to restoration.

snj33v, 2017-08-16:
its been advised to disable ssd optimization too

marcmerlin, 2021-01-14:
Sata ALPM has been safe since 4.15: https://www.spinics.net/lists/linux-btrfs/msg101833.html

linrunner, 2021-01-14:
Thanks for the info. I get it that this statement comes from the btrfs dev David Sterba

nibazshab, 2022-08-12:
@linrunner hi, i use Arch Linux from btrfs now, i want to install tlp, at now, do i need add `SATA_LINKPWR_ON_BAT=max_performance` in tlp.conf?

linrunner, 2022-08-12:
@nibazshab Why don't you just read the two posts above you and also look at the corresponding [FAQ entry](https://linrunner.de/tlp/faq/disks.html#does-sata-alpm-cause-disk-corruption-with-btrfs)?

nibazshab, 2022-08-12:
> @nibazshab Why don't you just read the two posts above yours and also look at the corresponding [FAQ entry](https://linrunner.de/tlp/faq/disks.html#does-sata-alpm-cause-disk-corruption-with-btrfs)?

ok, in fact i've already read those and i just worry for my computer, if it bothers you i'm so sorry

linrunner, 2022-08-12:
@nibazshab If you've already read the posts, what's your point then? The underlying kernel problem was solved 5 years ago according to the BTRFS developer. 

I'm not going to give you an assurance for your system, which by the way I don't know. All known facts are publicly mentioned or linked in this issue. You will have to decide for yourself.

Backups of important data are mandatory anyway.