linrunner/TLP #130: TLP leads to SATA Hard Disk / SSD Lockup

nohn, 2015-03-07:
This behaviour is reproducable on a Thinkpad T420 running Ubuntu 12.04 as well as 14.04:

When TLP is installed an running, every few minutes the system locks up, HDD led is constantly on for around a minute after that the system continues running normally.

Log messages:

Mar  7 08:24:28 bnibws03 kernel: [  375.143122] ata1.00: exception Emask 0x0 SAct 0x3ffffff8 SErr 0x50000 action 0x6 frozen
Mar  7 08:24:28 bnibws03 kernel: [  375.143127] ata1: SError: { PHYRdyChg CommWake }
Mar  7 08:24:28 bnibws03 kernel: [  375.143129] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143132] ata1.00: cmd 61/08:18:20:89:4c/00:00:0e:00:00/40 tag 3 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143132]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143133] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143134] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143136] ata1.00: cmd 61/08:20:28:89:4c/00:00:0e:00:00/40 tag 4 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143136]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143137] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143138] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143140] ata1.00: cmd 61/08:28:30:89:4c/00:00:0e:00:00/40 tag 5 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143140]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143141] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143142] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143144] ata1.00: cmd 61/08:30:38:89:4c/00:00:0e:00:00/40 tag 6 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143144]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143145] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143146] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143148] ata1.00: cmd 61/08:38:40:89:4c/00:00:0e:00:00/40 tag 7 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143148]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143149] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143150] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143152] ata1.00: cmd 61/08:40:48:89:4c/00:00:0e:00:00/40 tag 8 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143152]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143153] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143153] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143155] ata1.00: cmd 61/08:48:50:89:4c/00:00:0e:00:00/40 tag 9 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143155]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143156] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143157] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143159] ata1.00: cmd 61/08:50:58:89:4c/00:00:0e:00:00/40 tag 10 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143159]          res 40/00:80:00:00:00/00:00:00:00:00/40 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143160] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143161] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143163] ata1.00: cmd 61/08:58:60:89:4c/00:00:0e:00:00/40 tag 11 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143163]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143164] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143165] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143167] ata1.00: cmd 61/08:60:68:89:4c/00:00:0e:00:00/40 tag 12 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143167]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143168] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143169] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143171] ata1.00: cmd 61/08:68:70:89:4c/00:00:0e:00:00/40 tag 13 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143171]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143172] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143173] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143175] ata1.00: cmd 61/08:70:78:89:4c/00:00:0e:00:00/40 tag 14 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143175]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143176] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143177] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143179] ata1.00: cmd 61/08:78:80:89:4c/00:00:0e:00:00/40 tag 15 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143179]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143180] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143180] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143182] ata1.00: cmd 61/08:80:88:89:4c/00:00:0e:00:00/40 tag 16 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143182]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143183] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143184] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143186] ata1.00: cmd 61/08:88:90:89:4c/00:00:0e:00:00/40 tag 17 ncq 4096 out
Mar  7 08:24:28 bnibws03 kernel: [  375.143186]          res 40/00:00:00:00:00/00:00:00:00:00/00 Emask 0x4 (timeout)
Mar  7 08:24:28 bnibws03 kernel: [  375.143187] ata1.00: status: { DRDY }
Mar  7 08:24:28 bnibws03 kernel: [  375.143188] ata1.00: failed command: WRITE FPDMA QUEUED
Mar  7 08:24:28 bnibws03 kernel: [  375.143190] ata1.00: cmd 61/08:90:98:89:4c/00:00:0e:00:00/40 tag 18 ncq 4096 out
[... 64 lines cut ...]
Mar  7 08:24:29 bnibws03 kernel: [  375.503278] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503282] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503285] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503288] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503292] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503295] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503299] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503302] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503306] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503309] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503313] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503316] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503320] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503323] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503326] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503330] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503333] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503337] ata1.00: device reported invalid CHS sector 0
Mar  7 08:24:29 bnibws03 kernel: [  375.503340] ata1.00: device reported invalid CHS sector 0
[... cut ...]
Mar  7 08:24:29 bnibws03 kernel: [  375.503373] ata1: EH complete

On the Ubuntu 12.04 machine using a HDD, this started around 1 or 2 weeks ago (writing this on 2015-03-07) after running tlp for years without such problems, so I suspect a recently introduced change that cause this.

I suspected a defective HDD and replaced the HDD with an SSD and installed a fresh Ubuntu 14.04. Problems were gone. After installing tlp using the linrunner/tlp PPA, the problems re-appeared. After uninstalling tlp, the problems are gone again.

linrunner, 2015-03-07:
SATA ALPM is your prime suspect. Have a look at the [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#warnings) please.

I *always* need complete output from:

    sudo tlp-stat

nohn, 2015-03-07:
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_DRIVER_BLACKLIST="usbhid"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO ThinkPad T420 4236NGG
BIOS           = 83ET53WW (1.23 )
Release        = Ubuntu 14.04.2 LTS
Kernel         = 3.16.0-30-generic x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-3.16.0-30-generic root=/dev/mapper/bnibws04--vg-root ro
Init system    = upstart

+++ System Status
TLP power save = enabled
power source   = ac

+++ Processor
CPU Model      = Intel(R) Core(TM) i7-2620M CPU @ 2.70GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = performance
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3400000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = performance
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu1/cpufreq/scaling_max_freq  =  3400000 [kHz]

/sys/devices/system/cpu/cpu2/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor  = performance
[... 116 lines cut ...]
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ ThinkPad Extended Battery Functions
tp-smapi   = active
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/devices/platform/smapi/BAT0/manufacturer               = SANYO
/sys/devices/platform/smapi/BAT0/model                      = 42T4799
/sys/devices/platform/smapi/BAT0/manufacture_date           = 2010-10-21
/sys/devices/platform/smapi/BAT0/first_use_date             = 2011-07-02
/sys/devices/platform/smapi/BAT0/cycle_count                =    492
/sys/devices/platform/smapi/BAT0/design_capacity            =  86580 [mWh]
/sys/devices/platform/smapi/BAT0/last_full_capacity         =  67060 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_capacity         =  67060 [mWh]
/sys/devices/platform/smapi/BAT0/remaining_percent          =    100 [%]
/sys/devices/platform/smapi/BAT0/remaining_running_time_now = not_discharging [min]
/sys/devices/platform/smapi/BAT0/remaining_charging_time    = not_charging [min]
/sys/devices/platform/smapi/BAT0/power_now                  =      0 [mW]
/sys/devices/platform/smapi/BAT0/power_avg                  =      0 [mW]

tpacpi-bat.BAT0.startThreshold                              =     96 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

linrunner, 2015-03-09:
Did my suggestion to disable ALPM help?

nohn, 2015-03-10:
Did  not try yet. I'll do it next weekend.

closed 2015-03-24 by nohn

gorghino, 2016-11-04:
I'm facing this bug with my HP Omen 15 15-5003nl and ElementaryOS Loki 
Linux 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

```
ltp start log:

--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = Hewlett-Packard Type1ProductConfigId HP OMEN Notebook PC 15
BIOS           = F.38
Release        = elementary OS 0.4 Loki
Kernel         = 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.4.0-45-generic.efi.signed root=UUID=84702b2a-962f-47d1-b6f4-3077cb281632 ro quiet splash vt.handoff=7
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = AC

+++ Processor
CPU Model      = Intel(R) Core(TM) i7-4710HQ CPU @ 2.50GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3500000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
[... 139 lines cut ...]
Bus 001 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 004 ID 1bcf:2c7e control = auto, autosuspend_delay_ms =  2000 -- Sunplus Innovation Technology Inc.  (uvcvideo)
Bus 002 Device 003 ID 8087:07dc control = on,   autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 002 Device 005 ID 04f3:2016 control = on,   autosuspend_delay_ms =  2000 -- Elan Microelectronics Corp.  (usbhid)
Bus 002 Device 002 ID 046d:c52f control = on,   autosuspend_delay_ms =  2000 -- Logitech, Inc. Unifying Receiver (usbhid)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = 13142
/sys/class/power_supply/BAT0/model_name                     = RR04058
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  57152 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  57152 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  31099 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  28657 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

```

I didn't get how to disable SATA ALPM.
In the webpage there's  "to disable this setting completely, insert a '#' in the first column.". Where? Which column? 

Thank you!

linrunner, 2016-11-04:
It's described at the top of the [page you're referring to](http://linrunner.de/en/tlp/docs/tlp-configuration.html).

TLP's configuration is stored in **/etc/default/tlp**. Edit the file as root. Change the lines as shown: 

    #SATA_LINKPWR_ON_AC=max_performance
    #SATA_LINKPWR_ON_BAT=min_power

Save. Reboot.