linrunner/TLP #131: Problem with disable on battery and not in use

orschiro, 2015-03-07:
Dear @linrunner,

I wanted to disable my Bluetooth if not in use, or if on battery. My settings:

```
DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth"
DEVICES_TO_DISABLE_ON_BAT="bluetooth"
```

I use a Thinkpad X230 and version `0.7-1~trusty` on elementaryOS Freya Beta 2. 

All my settings are working except for this one. Output of tlp-stat:

```
~ ❯❯❯ sudo tlp-stat                                                                      ⏎
[sudo] password for orschiro: 
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_DRIVER_BLACKLIST="usbhid"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0
DEVICES_TO_ENABLE_ON_AC="bluetooth"
DEVICES_TO_DISABLE_ON_BAT="bluetooth"
DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth"
START_CHARGE_THRESH_BAT0=67
STOP_CHARGE_THRESH_BAT0=100

+++ System Info
System         = LENOVO ThinkPad X230 23352CG
BIOS           = G2ETA3WW (2.63 )
Release        = elementary OS Freya
Kernel         = 3.19.0-031900-generic x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-3.19.0-031900-generic root=UUID=08033b90-8c1f-4bd2-98cf-34abaa345fdd ro i915.i915_enable_rc6=1 i915.i915_enable_fbc=1 i915.lvds_downclock=1 quiet splash vt.handoff=7
[... 145 lines cut ...]

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' load error)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 45N1175
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  85860 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  83660 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  83520 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown

tpacpi-bat.BAT0.startThreshold                              =     67 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0
```

Do you have any idea why my settings are not used?

Thankfully,

Robert

linrunner, 2015-03-07:
Disabling bluetooth when not in use is possible on battery only. The documentation and the names of the variables are quite clear on that. There is no point in saving power on AC. 

Please consider that switching happens upon change of power source only.

orschiro, 2015-03-08:
> Please consider that switching happens upon change of power source only.

This was the point that was not fully clear to me. I forgot that TLP is not working as a constantly running background daemon. 

> ps. you may use both DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE and DEVICES_TO_DISABLE_ON_BAT but then the latter always wins.

I decided to only use `DEVICES_TO_DISABLE_ON_BAT`.

Thank you!

Robert

linrunner, 2015-03-09:
Clarified the docs. Thanks for your hint.

closed 2015-03-09 by linrunner