linrunner/TLP #132: Thinkpad T400 -  tp-smapi   = inactive (kernel module 'tp_smapi' load error)

japa-fi, 2015-03-07:
Hi.
Can't get the tp_smapi module to load. I have Thinkpad T400, with a battery which is not original lenovo battery (win 7 pops up a notitication that battery is not genuine thinkpad battery). I've restarted my laptop couple of times and also reinstalled the .rpm packages twice too.
Checked https://github.com/linrunner/TLP/issues/114 and here are some of the output you asked in that case.

[root@localhost ~]# modprobe -v tp_smapi
insmod /lib/modules/3.18.7-200.fc21.x86_64/extra/tp_smapi/thinkpad_ec.ko 
modprobe: ERROR: could not insert 'tp_smapi': No such device or address

[root@localhost ~]# ls -l  /lib/modules/3.18.7-200.fc21.x86_64/extra/tp_smapi/
total 104
-rw-r--r--. 1 root root 35712 Mar  7 22:32 hdaps.ko
-rw-r--r--. 1 root root 14552 Mar  7 22:32 thinkpad_ec.ko
-rw-r--r--. 1 root root 49192 Mar  7 22:32 tp_smapi.ko

[root@localhost ~]# cat /var/cache/akmods/aksmod.log
cat: /var/cache/akmods/aksmod.log: No such file or directory

[root@localhost ~]# cat /var/cache/akmods/acpi_call/.last.log
2015/03/07 22:31:42 akmods: Building RPM using the command '/bin/akmodsbuild --target x86_64 --kernels 3.18.7-200.fc21.x86_64 /usr/src/akmods/acpi_call-kmod.latest'
2015/03/07 22:31:45 akmodsbuild: * Rebuilding /usr/src/akmods/acpi_call-kmod.latest for kernel(s) 3.18.7-200.fc21.x86_64: warning: user build does not exist - using root
2015/03/07 22:31:45 akmodsbuild: warning: group build does not exist - using root
2015/03/07 22:31:45 akmodsbuild: warning: user build does not exist - using root
2015/03/07 22:31:45 akmodsbuild: warning: group build does not exist - using root
2015/03/07 22:31:45 akmodsbuild: warning: user build does not exist - using root
2015/03/07 22:31:45 akmodsbuild: warning: group build does not exist - using root
2015/03/07 22:31:45 akmodsbuild: Installing /usr/src/akmods/acpi_call-kmod.latest
2015/03/07 22:31:45 akmodsbuild: Building target platforms: x86_64
2015/03/07 22:31:45 akmodsbuild: Building for target x86_64
2015/03/07 22:31:45 akmodsbuild: Executing(%prep): /bin/sh -e /var/tmp/rpm-tmp.Bi36f3
2015/03/07 22:31:45 akmodsbuild: + umask 022
2015/03/07 22:31:45 akmodsbuild: + cd /tmp/akmodsbuild.i8P0gfK9//BUILD
2015/03/07 22:31:45 akmodsbuild: + kmodtool --target x86_64 --repo rpmfusion --kmodname acpi_call-kmod --for-kernels 3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: %package       -n kmod-acpi_call-3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: Summary:          acpi_call kernel module(s) for 3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: Group:            System Environment/Kernel
2015/03/07 22:31:45 akmodsbuild: Provides:         kernel-modules-for-kernel = 3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: Provides:         acpi_call-kmod = %{?epoch:%{epoch}:}%{version}-%{release}
2015/03/07 22:31:45 akmodsbuild: Requires:         acpi_call-kmod-common >= %{?epoch:%{epoch}:}%{version}
2015/03/07 22:31:45 akmodsbuild: Requires(post):   /usr/sbin/depmod
2015/03/07 22:31:45 akmodsbuild: Requires(postun): /usr/sbin/depmod
2015/03/07 22:31:45 akmodsbuild: Requires:         kernel-uname-r = 3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: BuildRequires:	  kernel-devel-uname-r = 3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: %post          -n kmod-acpi_call-3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: /usr/sbin/depmod -aeF /boot/System.map-3.18.7-200.fc21.x86_64 3.18.7-200.fc21.x86_64 > /dev/null || :
2015/03/07 22:31:45 akmodsbuild: %postun        -n kmod-acpi_call-3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: /usr/sbin/depmod  -aF /boot/System.map-3.18.7-200.fc21.x86_64 3.18.7-200.fc21.x86_64 &> /dev/null || :
2015/03/07 22:31:45 akmodsbuild: 
2015/03/07 22:31:45 akmodsbuild: %description  -n kmod-acpi_call-3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: This package provides the acpi_call kernel modules built for the Linux
2015/03/07 22:31:45 akmodsbuild: kernel 3.18.7-200.fc21.x86_64 for the %{_target_cpu} family of processors.
2015/03/07 22:31:45 akmodsbuild: %files        -n kmod-acpi_call-3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: %defattr(644,root,root,755)
2015/03/07 22:31:45 akmodsbuild: %dir /usr/lib/modules/3.18.7-200.fc21.x86_64/extra
2015/03/07 22:31:45 akmodsbuild: /usr/lib/modules/3.18.7-200.fc21.x86_64/extra/acpi_call/
2015/03/07 22:31:45 akmodsbuild: 
2015/03/07 22:31:45 akmodsbuild: 
2015/03/07 22:31:45 akmodsbuild: 
2015/03/07 22:31:45 akmodsbuild: %global kmodinstdir_prefix  /usr/lib/modules/
2015/03/07 22:31:45 akmodsbuild: %global kmodinstdir_postfix /extra/acpi_call/
2015/03/07 22:31:45 akmodsbuild: %global kernel_versions     3.18.7-200.fc21.x86_64___%{_usrsrc}/kernels/3.18.7-200.fc21.x86_64
2015/03/07 22:31:45 akmodsbuild: 
2015/03/07 22:31:45 akmodsbuild: + cd /tmp/akmodsbuild.i8P0gfK9/BUILD
2015/03/07 22:31:45 akmodsbuild: + rm -rf acpi_call-kmod-1.1.0
2015/03/07 22:31:45 akmodsbuild: + /usr/bin/mkdir -p acpi_call-kmod-1.1.0
2015/03/07 22:31:45 akmodsbuild: + cd acpi_call-kmod-1.1.0
2015/03/07 22:31:45 akmodsbuild: + /usr/bin/gzip -dc /tmp/akmodsbuild.i8P0gfK9/SOURCES/acpi_call-1.1.0.tar.gz
2015/03/07 22:31:45 akmodsbuild: + /usr/bin/tar -xf -
2015/03/07 22:31:45 akmodsbuild: + STATUS=0
2015/03/07 22:31:45 akmodsbuild: + '[' 0 -ne 0 ']'
2015/03/07 22:31:45 akmodsbuild: + /usr/bin/chmod -Rf a+rX,u+w,g-w,o-w .
[... 115 lines cut ...]
Installed:
  kmod-acpi_call-3.18.7-200.fc21.x86_64.x86_64 0:1.1.0-3.fc21                   

Complete!
2015/03/07 22:31:57 akmods: Successful.

[root@localhost ~]# uname -a
Linux localhost.localdomain 3.18.7-200.fc21.x86_64 #1 SMP Wed Feb 11 21:53:17 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

[root@localhost ~]# ls -l /usr/lib/modules/*/extra/acpi_call/acpi_call.ko
-rw-r--r--. 1 root root 11424 Mar  7 22:31 /usr/lib/modules/3.18.7-200.fc21.x86_64/extra/acpi_call/acpi_call.ko

[root@localhost ~]# modinfo /usr/lib/modules/*/extra/acpi_call/acpi_call.ko
filename:       /usr/lib/modules/3.18.7-200.fc21.x86_64/extra/acpi_call/acpi_call.ko
license:        GPL
depends:        
vermagic:       3.18.7-200.fc21.x86_64 SMP mod_unload 
[root@localhost ~]# modinfo /usr/lib/modules/*/extra/acpi_call/acpi_call.ko
filename:       /usr/lib/modules/3.18.7-200.fc21.x86_64/extra/acpi_call/acpi_call.ko
license:        GPL
depends:        
vermagic:       3.18.7-200.fc21.x86_64 SMP mod_unload 

To my limited knowledge, I can't find anything wrong with above output.
Comments appreciated.

linrunner, 2015-03-07:
Hi.

> [root@localhost ~]# modprobe -v tp_smapi
> insmod /lib/modules/3.18.7-200.fc21.x86_64/extra/tp_smapi/thinkpad_ec.ko
> modprobe: ERROR: could not insert 'tp_smapi': No such device or address

This is not a problem with TLP nor is it a problem with the akmod-tp_smapi package provided in the TLP repo. It's just that your T400 refuses to load tp-smapi's kernel modules. I suggest you use adequate forums to resolve this issue with tp-smapi.

closed 2015-03-07 by linrunner

japa-fi, 2015-03-08:
On 2015-03-07 23:23, linrunner notifications-at-github.com
|github/Example Allow| wrote:
> Hi.
>
>     [root@localhost ~]# modprobe -v tp_smapi
>     insmod /lib/modules/3.18.7-200.fc21.x86_64/extra/tp_smapi/thinkpad_ec.ko
>     modprobe: ERROR: could not insert 'tp_smapi': No such device or [...]

Thank you for you comment.
Btw. tlp works fine with my other laptop (T430, using akmod-acpi_call)
thanks for creating tpl.

Update (1 hour later):
https://github.com/evgeni/tp_smapi states:
=======================================================
Module parameters
-----------------

thinkpad_ec module:
  force_io=1 lets thinkpad_ec load on some recent ThinkPad models
  (e.g., T400 and T500) whose BIOS's ACPI DSDT reserves the ports we need.
=======================================================

Added options thinkpad_ec force_io=1' to thinkpad_ec.conf
Now I got the module loaded. Now, lets start experimenting with the
battery parameters :).