linrunner/TLP #133: Change autosuspend time

Marsfan, 2015-03-17:
I was looking at issue #35. Just like the user in that post, one of my usb ports autosuspend is set to 0 on delay. I would like to change this to 2000 and see if that fixes a slow boot issue just as in #35. How do I do that

linrunner, 2015-03-19:
Hi,

unfortunately this is not possible without a code change. I'd like to see

    tlp-stat -s
    tlp-stat -u

and please mark the device in question!

Btw. TLP doesn't touch the delays since 0.7, these are the kernel presets.

Marsfan, 2015-03-23:
Ok. I'll get it sometime this week. I found out that my computer's boot is slowed down by tlp. Disabling usb autosuspend helps, but not the xusb thing in #35. All I can think of is that one autosuspend.

Marsfan, 2015-03-23:
tlp-stat -s:
```
--- TLP 0.7 --------------------------------------------

+++ System Info
System         = Dell Inc. A04 Inspiron 1012
BIOS           = A04
Release        = Linux Mint 17.1 Rebecca
Kernel         = 3.13.0-37-generic x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-3.13.0-37-generic root=UUID=3102334f-9909-49fd-84f2-d0eec7247487 ro quiet splash vt.handoff=7
Init system    = upstart

+++ System Status
TLP power save = enabled
power source   = ac
```
tlp-stat -u:
```
--- TLP 0.7 --------------------------------------------

+++ USB
autosuspend        = disabled
device whitelist   = (not configured)
device blacklist   = (not configured)
driver blacklist   = usbhid
wwan blacklist     = enabled (cdc_acm cdc_wdm cdc_ether hso qcserial sierra)

Bus 001 Device 002 ID 0c45:641d control = auto, autosuspend_delay_ms =  2000 -- Microdia 1.3 MPixel Integrated Webcam (uvcvideo)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 005 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 004 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 003 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 002 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
```

I think that the issue is pretty clear. All of my Linux Foundation root hubs (whatever the heck they are) have a zero autosuspend.

linrunner, 2015-03-23:
Imho this is a kernel issue, because control = auto, autosuspend_delay_ms = 0 is the kernel preset for all hubs. So it shouldn't matter writing "auto" again.

As your netbook actually has very few usb devices you could stick with 

    USB_AUTOSUSPEND=0

as a proper workaround. Or you could blacklist only the hubs

    USB_BLACKLIST="1d6b:0001 1d6b:0002"

I'll change the code to only write "auto" if not already present. So please stay tuned for testing!

Marsfan, 2015-03-23:
What does this new commit do in stupid person terms? Also, how do I blacklist those devices?

linrunner, 2015-03-24:
> Also, how do I blacklist those devices?

Pardon? I gave you the exact config line in my above post.

> What does this new commit do in stupid person terms?

For you convenience i uploaded packages incorporating the commit [here](http://download.linrunner.de/packages/). Please remove blacklisting and re-enable autosuspend before testing boot.

Marsfan, 2015-03-24:
Sorry, I meant what file do I need to add this to, and do I just put it at the end of the file, or somewhere in the middle?

linrunner, 2015-03-24:
http://linrunner.de/en/tlp/docs/tlp-configuration.html
http://linrunner.de/en/tlp/docs/tlp-configuration.html#usb

Marsfan, 2015-03-24:
Awesome, thanks.

closed 2015-03-24 by Marsfan

linrunner, 2015-03-24:
You're welcome. 

Did you blacklist or use the new packages? As you're the only one who can reproduce this, i need your feedback on the packages (without blacklist config!).