linrunner/TLP #135: stopThreshold is ignored

Cygn, 2015-03-23:
tlp sets the same value for charge start and stop and ignores stop threshold of setcharge.

[yo@yogi ~]$ sudo tlp setcharge 45 75 BAT0
Setting temporary charge thresholds for BAT0:
  stop  = 75
  start = 45
[yo@yogi ~]$ sudo tlp-stat -b
--- TLP 0.7 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' load error)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SONY
/sys/class/power_supply/BAT0/model_name                     = 45N1705
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  47060 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  44280 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  29470 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown

tpacpi-bat.BAT0.startThreshold                              =     45 [%]
tpacpi-bat.BAT0.stopThreshold                               =     45 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

[yo@yogi ~]$ rpm -qa | grep tlp
tlp-release-1.0.2-1.fc21.noarch
tlp-0.7-2.fc22.noarch
tlp-rdw-0.7-2.fc22.noarch
[yo@yogi ~]$ uname -a
Linux yogi 3.19.1-201.fc21.x86_64 #1 SMP Wed Mar 18 04:29:24 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
(3.18.9-200.fc21.x86_64 did this too)

Cygn, 2015-03-23:
hm, it may be just the output; the stop/start thresholds seems respected.

linrunner, 2015-03-23:
Hi, this is a (well known issue](http://linrunner.de/en/tlp/docs/tlp-faq.html#wrong-threshold-values) with the ThinkPad Edge series fiirmware. Please always check the FAQ before opening an issue.

closed 2015-03-23 by linrunner