linrunner/TLP #136: Devices to enable on Radioswitch

orschiro, 2015-04-19:
Hi all, 

I recall that in previous version there used to be the entry: 

```
DEVICES_TO_ENABLE_ON_RADIOSW
```

Is this still possible with the latest version of TLP? I would like to only switch Wifi on but keep Bluetooth disabled when using the radio (hardware) switch.

Thanks ahead!

Robert

dywisor, 2015-04-20:
> Is this still possible with the latest version of TLP?

No: ``DEVICES_TO_ENABLE_ON_RADIOSW`` has been removed from the config file because the underlying kernel functionality has been removed in 3.12 or so (and was deprecated before).

In more recent kernels, you'd need a daemon listening on input events, which is not provided by TLP.
You may try [input-event-daemon](https://github.com/gandro/input-event-daemon) with a --untested-- config file as posted [here (de)](http://thinkpad-forum.de/threads/180227-Kubuntu-14-04-inaktive-FN-Tastenkombinationen-toggle?p=1815399&viewfull=1#post1815399), but this project is no longer actively maintained.

Btw, it appears that you've asked this question before at [thinkpad-forum (de)](http://thinkpad-forum.de/threads/164169-Hardware-Radio-Switch-Bluetooth-deaktiviert-lassen-wenn-Wlan-aktiviert).

orschiro, 2015-04-20:
@dywisor 

Thanks for your response! I assume it is deprecated for a reason. Thus I will no further bother to hack around it. 

> Btw, it appears that you've asked this question before at thinkpad-forum (de).

There it was! I knew that I had asked that question somewhere before. :)

Thank you!

Robert

linrunner, 2015-04-20:
TLP's features do not disappear by accident, they are dismissed in the changelog:

> +++ 0.6 --- 06.10.2014 +++
>    - Remove DEVICES_TO_ENABLE_ON_RADIOSW from configuration because
>      it's deprecated: works with Ubuntu 12.04/Kernel 3.2 only

closed 2015-04-20 by linrunner