linrunner/TLP #139: tlp block system to suspend

gansteed, 2015-05-11:
I'm using Archlinux(up-to-date), and after enable `tlp.service` and `tlp-sleep.service`, if I close the lid, my notebook will not fall to sleep.  It will sleep after disable and stop tlp service by systemd. Is there any solution for it?

linrunner, 2015-05-11:
Hi,

1. of all, i always need the full output of:

     tlp-stat

2. Does this happen on ac or battery or both?
3. Did you try to disable tlp-sleep.service only?
4. Did you try with an older kernel?

gansteed, 2015-05-12:
thanks for point that out, here is the result for testing:

1, here it is

```bash
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_DRIVER_BLACKLIST="usbhid"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO Lenovo IdeaPad Y400 20192
BIOS           = 6BCN44WW(V2.04)
Kernel         = 4.0.1-1-ARCH x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=ea52e4ea-15e1-4166-8117-d75b6cdd9f39 rw quiet
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =  1200000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3100000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =  1200000 [kHz]
[... 1194 lines cut ...]
[  197.116381] pcieport 0000:00:1c.1: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 03] add_size 200000
[  197.116395] pci_bus 0000:04: Allocating resources
[  197.116434] pcieport 0000:00:1c.3: bridge window [io  0x1000-0x0fff] to [bus 04] add_size 1000
[  197.116438] pcieport 0000:00:1c.3: bridge window [mem 0x00100000-0x000fffff 64bit pref] to [bus 04] add_size 200000
[  197.116449] pcieport 0000:00:1c.0: res[15]=[mem 0x00100000-0x000fffff 64bit pref] get_res_add_size add_size 200000
[  197.116452] pcieport 0000:00:1c.1: res[15]=[mem 0x00100000-0x000fffff 64bit pref] get_res_add_size add_size 200000
[  197.116456] pcieport 0000:00:1c.3: res[15]=[mem 0x00100000-0x000fffff 64bit pref] get_res_add_size add_size 200000
[  197.116459] pcieport 0000:00:1c.1: res[13]=[io  0x1000-0x0fff] get_res_add_size add_size 1000
[  197.116462] pcieport 0000:00:1c.3: res[13]=[io  0x1000-0x0fff] get_res_add_size add_size 1000
[  197.116487] pcieport 0000:00:1c.0: BAR 15: assigned [mem 0xc3600000-0xc37fffff 64bit pref]
[  197.116505] pcieport 0000:00:1c.1: BAR 15: assigned [mem 0xc3800000-0xc39fffff 64bit pref]
[  197.116523] pcieport 0000:00:1c.3: BAR 15: assigned [mem 0xc3a00000-0xc3bfffff 64bit pref]
[  197.116530] pcieport 0000:00:1c.1: BAR 13: assigned [io  0x5000-0x5fff]
[  197.116535] pcieport 0000:00:1c.3: BAR 13: assigned [io  0x6000-0x6fff]
[  197.116745] pci_bus 0000:01: Allocating resources
```

3,  yes, I've, I will append a message about this after reboot.

4, no, I've not. Here is the kernel message:

```bash
$ uname -a
Linux idea 4.0.1-1-ARCH #1 SMP PREEMPT Wed Apr 29 12:00:26 CEST 2015 x86_64 GNU/Linux
```

gansteed, 2015-05-12:
hi, after reboot, it block system to suspend both in ac and battery mode. here is `dmesg` after reboot

```bash
$ dmesg 
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.0.1-1-ARCH (builduser@tobias) (gcc version 5.1.0 (GCC) ) #1 SMP PREEMPT Wed Apr 29 12:00:26 CEST 2015
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=ea52e4ea-15e1-4166-8117-d75b6cdd9f39 rw quiet
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000087fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000088000-0x00000000000bffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000a95affff] usable
[    0.000000] BIOS-e820: [mem 0x00000000a95b0000-0x00000000aa9affff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000aa9b0000-0x00000000aefbefff] usable
[    0.000000] BIOS-e820: [mem 0x00000000aefbf000-0x00000000af04dfff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af04e000-0x00000000af04ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af050000-0x00000000af065fff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af066000-0x00000000af066fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af067000-0x00000000af07efff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af07f000-0x00000000af07ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af080000-0x00000000af08afff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af08b000-0x00000000af08bfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af08c000-0x00000000af090fff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af091000-0x00000000af091fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af092000-0x00000000af09ffff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af0a0000-0x00000000af0a0fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af0a1000-0x00000000af0f1fff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af0f2000-0x00000000af136fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af137000-0x00000000af1befff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000af1bf000-0x00000000af6befff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000af6bf000-0x00000000af7befff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000af7bf000-0x00000000af7fefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000af7ff000-0x00000000af7fffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000af800000-0x00000000afffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f3ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feb00000-0x00000000feb03fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed10000-0x00000000fed19fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed1c000-0x00000000fed1ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffb80000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000024effffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.31 by INSYDE Corp.
[    0.000000] efi:  ACPI=0xaf7fe000  ACPI 2.0=0xaf7fe014  SMBIOS=0xaf6bef98 
[    0.000000] efi: mem00: [Boot Code          |   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000000000000-0x0000000000001000) (0MB)
[    0.000000] efi: mem01: [Conventional Memory|   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000000001000-0x000000000006f000) (0MB)
[    0.000000] efi: mem02: [Boot Data          |   |  |  |  |   |WB|WT|WC|UC] range=[0x000000000006f000-0x0000000000070000) (0MB)
[    0.000000] efi: mem03: [Conventional Memory|   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000000070000-0x0000000000080000) (0MB)
[    0.000000] efi: mem04: [Boot Code          |   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000000080000-0x0000000000082000) (0MB)
[    0.000000] efi: mem05: [Conventional Memory|   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000000082000-0x0000000000088000) (0MB)
[    0.000000] efi: mem06: [Runtime Data       |RUN|  |  |  |   |WB|WT|WC|UC] range=[0x0000000000088000-0x00000000000a0000) (0MB)
[    0.000000] efi: mem07: [Loader Data        |   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000000100000-0x0000000000f28000) (14MB)
[    0.000000] efi: mem08: [Conventional Memory|   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000000f28000-0x000000003792e000) (874MB)
[    0.000000] efi: mem09: [Loader Data        |   |  |  |  |   |WB|WT|WC|UC] range=[0x000000003792e000-0x0000000037c8f000) (3MB)
[    0.000000] efi: mem10: [Conventional Memory|   |  |  |  |   |WB|WT|WC|UC] range=[0x0000000037c8f000-0x000000007d3f2000) (1111MB)
[    0.000000] efi: mem11: [Loader Data        |   |  |  |  |   |WB|WT|WC|UC] range=[0x000000007d3f2000-0x00000000a79c0000) (677MB)
[    0.000000] efi: mem12: [Boot Data          |   |  |  |  |   |WB|WT|WC|UC] range=[0x00000000a79c0000-0x00000000a79e0000) (0MB)
[    0.000000] efi: mem13: [Conventional Memory|   |  |  |  |   |WB|WT|WC|UC] range=[0x00000000a79e0000-0x00000000a85d4000) (11MB)
[    0.000000] efi: mem14: [Boot Data          |   |  |  |  |   |WB|WT|WC|UC] range=[0x00000000a85d4000-0x00000000a95b0000) (15MB)
[    0.000000] efi: mem15: [Reserved           |   |  |  |  |   |WB|WT|WC|UC] range=[0x00000000a95b0000-0x00000000aa9b0000) (20MB)
[... cut ...]
[    4.329774] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150204/nsarguments-95)
[    4.329783] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150204/nsarguments-95)
[    4.635772] ACPI Warning: \_SB_.PCI0.PEG0.PEGP._DSM: Argument #4 type mismatch - Found [Buffer], ACPI requires [Package] (20150204/nsarguments-95)
[    5.096287] alx 0000:02:00.0 enp2s0: NIC Up: 100 Mbps Full
[   11.407263] fuse init (API version 7.23)
```

linrunner, 2015-05-13:
Please test 4.

linrunner, 2015-06-01:
I close here. Feel free to reopen if you have tested with an older kernel.

closed 2015-06-01 by linrunner