linrunner/TLP #140: TLP not working after suspend [ubuntu 15.04]

matthias-koenig, 2015-05-13:
Hi,

I have a problem with TLP on my DELL XPS 13 (2015 model) running Ubuntu 15.04. After I waking the laptop from suspend, the cpu governor is set back to performance. Running TLP stat gives me the following two warnings:

***Warning: tlp.service is not enabled -- invoke "systemctl enable tlp.service" to correct this!
***Warning: tlp-sleep.service is not enabled -- invoke "systemctl enable tlp-sleep.service" to correct this!

I guess that these services (particularly the second one) would normally control the behaviour of TLP after suspend but they don't seem to work.  Typing the two commands into a terminal does not solve the problem. While the first one can be exectued, the warning still shows up. The second one (tlp-sleep.service) fails because the service file does not even exist on my machine.

Am I missing something? Help is appreciated!

linrunner, 2015-05-13:
Hi,

maybe your installed packages are for an older Ubuntu version.

Please show
```
apt-cache policy tlp
```
Make shure the TLP PPA for vivid is enabled in your package sources

Reinstall the packages
```
sudo apt-get update
sudo apt-get install --reinstall tlp/vivid tlp-rdw/vivid
```

matthias-koenig, 2015-05-18:
This might have been the problem. However, I saw this too late, I downloaded the missing files and installed them manually. Thank you anyway!

closed 2015-05-18 by linrunner