linrunner/TLP #143: tlp incorrectly reports Yoga 2 Pro is not working

venik212, 2015-06-06:
I am using tlp on a Yoga 2 Pro with dual boot (windows 8.1 and Lubuntu 15.04).  tlp reports correctly the temperature, but always says that the fan speed is 9, even when I can hear the fan working and the temperature is >80 deg. C.
What do I do?  Should I install the Thinkpad specific portion of tlp?  Yoga 2 Pro is ann Ideapad, not a Thinkpad.

linrunner, 2015-06-06:
Hi,

you're aware that tlp-stat only *displays* temperature and fan speed? There is no further functionality about temps and fans in TLP. So your issue subject "is not working" is plain wrong.

Please show:

    sudo tlp-stat -s -t

venik212, 2015-06-07:
If tlp reports fan speed of 0 when the fan is audibly spinning, then it must be displaying/reporting speed in units I do not understand.  In any event, here is what I get with sudo tlp-stat -s -t:
--- TLP 0.7 --------------------------------------------

+++ System Info
System         = LENOVO Lenovo Yoga 2 Pro 20266
BIOS           = 76CN43WW
Release        = Ubuntu 15.04
Kernel         = 3.19.0-18-generic x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-3.19.0-18-generic.efi.signed root=UUID=86dbd065-f89b-48ea-8511-e06a23d93745 ro quiet splash acpi_backlight=vendor vt.handoff=7
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = ac

+++ Temperatures
CPU temp               =    61 [°C]
Fan speed (fan1)       =     0 [/min]

udi@udi-Yoga-2-Pro:~$

linrunner, 2015-06-07:
For fan speed tlp-stat displays the contents of kernel sysfiles (iterates over /sys/class/hwmon/hwmon*/device/fan?_input) without any conversion or interpretation. It's a kernel or ACPI issue. TLP can't do anything to correct this.

venik212, 2015-06-08:
I naively thought that tlp reads some sensor(s) directly.  I was wrong once before, and this must be (at least) the second time. I have known for some time that Lenovo's Yoga 2 Pro does not support Linux fully (you have to disable their ideapad_laptop module, there are issues with the screen resolution and with sound, etc.), and I guess that this is another manifestation of that neglect.

linrunner, 2015-06-09:
Imho it's not possible to do device io from user space, you *always* need a kernel driver. This is why tp-smapi or acpi-call is necessary fopr ThinkPad battery functions.

I suggest you file a kernel bug report for your problem with ideapad_laptop (or check if there is one already). No improvement without reports. 

Btw: ThinkPad Edge (E/S) series reports wrong fan speed too.

closed 2015-06-09 by linrunner