linrunner/TLP #146: Can not boot in battery mode!

bdgong, 2015-06-21:
Hi,  I'm using Ubuntu14.04.2,  last few days I installed TLP for longer use in battery mode.  And yesterday when I was sitting in library try to boot my computer (Acer V3-551G) without power plugged in, oops! It's seems that the XServer can not start normally at all.  The UBUNTU LOGO just stand there and stand there forever...What's the problem?  Appreciate for your answers.

linrunner, 2015-06-21:
Hi, i *always* need the output of

    sudo tlp-stat

(in battery mode)

Did you uninstall TLP and check if it works then?

closed 2015-07-25 by linrunner

bdgong, 2015-10-22:
Thanks for your answer, I am using Debian jessie now, but that problem still exists, and I have to uninstall it.^_^