linrunner/TLP #147: TLP does not switch CPU Governor when plugged in.

duschendestroyer, 2015-07-01:
Below is the output of tlp stat after plugging in the ac adapter.
tlp knows that I'm on ac power but the governor is still powersave even though the configuration states that it should switch to performance.
When I use tlp ac to switch to ac mode manually, everything works as expected.
The only logical reason I can think of is another service setting it back to powersave on the ac event, but I don't think I have any other power management tool installed.

Distribution: Archlinux
Hardware: Thinkpad X250, intel i5 5200U

--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_DRIVER_BLACKLIST="usbhid"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = LENOVO ThinkPad X250 20CLS06A00
BIOS           = N10ET28W (1.05 )
Kernel         = 4.0.5-1-ARCH x86_64
/proc/cmdline  = initrd=\initramfs-linux.img cryptdevice=/dev/sda6:vgroup:allow-discards resume=/dev/mapper/vgroup-swap video.use_native_backlight=1 root=/dev/mapper/vgroup-root ro
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = ac

+++ Processor
CPU Model      = Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
[... 133 lines cut ...]
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = inactive (kernel module 'acpi_call' not installed)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 45N1773
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  23200 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  18270 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  14540 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown

+++ Battery Status
/sys/class/power_supply/BAT1/manufacturer                   = SANYO
/sys/class/power_supply/BAT1/model_name                     = 45N1775
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  23200 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  20540 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  11980 [mWh]
/sys/class/power_supply/BAT1/power_now                      =  26500 [mW]
/sys/class/power_supply/BAT1/status                         = Charging

+++ Suggestions
* Install tp-smapi kernel modules for ThinkPad advanced battery functions

linrunner, 2015-07-01:
Hi, i suspect the kernel fails to write the change. Please try to change the according sysfile by hand (in a root shell)

    echo "performance" > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor

and then show
 
    tlp-stat -p

duschendestroyer, 2015-07-01:
this does not change the output of tlp stat.
Calling "tlp ac" does change it to performance, however.

# echo "performance" > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
# tlp stat -p
--- TLP 0.7 --------------------------------------------

+++ Processor
CPU Model      = Intel(R) Core(TM) i5-5200U CPU @ 2.20GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2700000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu1/cpufreq/scaling_max_freq  =  2700000 [kHz]

/sys/devices/system/cpu/cpu2/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu2/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu2/cpufreq/scaling_max_freq  =  2700000 [kHz]

/sys/devices/system/cpu/cpu3/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu3/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu3/cpufreq/scaling_max_freq  =  2700000 [kHz]

/sys/devices/system/cpu/intel_pstate/min_perf_pct      = 18
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100
/sys/devices/system/cpu/intel_pstate/no_turbo          = 1

x86_energy_perf_policy.cpu0                            = powersave
x86_energy_perf_policy.cpu1                            = powersave
x86_energy_perf_policy.cpu2                            = powersave
x86_energy_perf_policy.cpu3                            = powersave

/proc/sys/kernel/nmi_watchdog                          = 0

+++ Undervolting
PHC kernel not available.

linrunner, 2015-07-02:
> this does not change the output of tlp stat. 
> Calling "tlp ac" does change it to performance, however.

Strange ...    Please show the full output of 

    tlp-stat

on **battery** and back on **ac** again. I suggest you use https://gist.github.com/ for the lengthy output.

duschendestroyer, 2015-07-02:
Here are three outputs:
1. on battery
2. after plugging in ac adapter
3. after calling tlp ac

https://gist.github.com/duschendestroyer/fb4e35fcf60988027cc6

niki-eng, 2015-07-03:
I have the same problem on Lenovo ThinkPad E540 with kernel 4.0.0-2-amd64.

linrunner, 2015-07-03:
The outputs show that the automatic transition from ac to bat fails. Please give me a trace for that transition --> http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#support

linrunner, 2015-07-03:
I also need (with AC plugged in)

    grep ".*" /sys/class/power_supply/*/{type,online,present}

duschendestroyer, 2015-07-06:
% grep ".*" /sys/class/power_supply/*/{type,online,present}                                               
/sys/class/power_supply/AC/type:Mains
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT1/type:Battery
/sys/class/power_supply/AC/online:1
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT1/present:1

duschendestroyer, 2015-07-06:
Jul 06 16:32:39 uljanov tlp[22726]: power_source=bat
Jul 06 16:32:44 uljanov tlp[22791]: +++ mode=stat -t (0.7) ++++++++++++++++++++++++++++++++++++++++
Jul 06 16:32:44 uljanov tlp[22792]: PATH=/home/dude/matlab/bin/:/opt/intel/bin/:/opt/intel/composerxe/bin/intel64:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/intel/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
Jul 06 16:32:44 uljanov tlp[22793]: power_source=bat
Jul 06 16:32:45 uljanov tlp[22830]: +++ mode=auto (0.7) ++++++++++++++++++++++++++++++++++++++++
Jul 06 16:32:45 uljanov tlp[22831]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin
Jul 06 16:32:45 uljanov tlp[22832]: power_source=ac
Jul 06 16:32:47 uljanov tlp[22865]: +++ mode=stat -t (0.7) ++++++++++++++++++++++++++++++++++++++++
Jul 06 16:32:47 uljanov tlp[22866]: PATH=/home/dude/matlab/bin/:/opt/intel/bin/:/opt/intel/composerxe/bin/intel64:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/intel/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl
Jul 06 16:32:47 uljanov tlp[22867]: power_source=ac

linrunner, 2015-07-06:
I'm not yet sure what happens. I need the full trace, i.e. configured with `TLP_DEBUG="bat disk lock nm path pm rf run sysfs udev usb" 

Please: use  **tlp-stat -t**  *not*  tlp stat -t (the latter just clutters the trace).

Because of the duplicate #149 show traces for both power state transitions:

1. AC --> BAT
2. BAT --> AC

duschendestroyer, 2015-07-06:
https://gist.github.com/duschendestroyer/4da4169c9807869fb7a5

I started with ac plugged in and unplugged (twice).
The trace doesn't show any more info despite adding the config line above.

linrunner, 2015-07-06:
1. Does  `tlp auto`  work?
2. What is your default shell (/bin/sh)?

duschendestroyer, 2015-07-06:
1. yes
2. zsh

duschendestroyer, 2015-07-06:
1. yes
2. bash (as is default in archlinux)

linrunner, 2015-07-06:
Summary so far:
* `tlp auto` called from udev executed by bash does not work
* `tlp auto` manually executed in a zsh works

linrunner, 2015-07-06:
Please try to remove the sub shell in the auto section [1] in **/usr/bin/tlp**: 

    auto) # set mode depending on state (called by udev rule)
        check_root
        apply_common_settings $pwrmode
        [ "$pwrmode" = "1" ] && poweroff_drivebay 0
        set_radio_device_states $pwrmode
        ;;

[1] https://github.com/linrunner/TLP/blob/master/tlp#L140

duschendestroyer, 2015-07-06:
this works :)

linrunner, 2015-07-06:
OK, this was the easy part. The difficult part will be to find out why the subshell fails ... :(

BLuFeNiX, 2015-07-07:
Hi, I have run the trace. It seems to be detecting the change:

    Jul 07 19:18:32 carbide tlp[5454]: +++ mode=auto (0.7) ++++++++++++++++++++++++++++++++++++++++
    Jul 07 19:18:32 carbide tlp[5455]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin
    Jul 07 19:18:32 carbide tlp[5456]: power_source=bat
    Jul 07 19:18:35 carbide tlp[5501]: +++ mode=auto (0.7) ++++++++++++++++++++++++++++++++++++++++
    Jul 07 19:18:35 carbide tlp[5502]: PATH=/usr/local/sbin:/usr/local/bin:/usr/bin
    Jul 07 19:18:35 carbide tlp[5503]: power_source=ac

But it doesn't actually change the CPU settings (governor, etc.), unless I run `tlp start` or `tlp auto`

linrunner, 2015-07-14:
Arch: 0.7-3 with Patches is in the repositories now. 
Other distros: will get the fix with 0.8 (including Debian Sid).

closed 2015-07-14 by duschendestroyer

Delvien, 2018-09-09:
This issue exists in Fedora 28 as well.

linrunner, 2018-09-10:
@Delvien : if you want support, please open your own issue and show proper tlp-stat outputs.