linrunner/TLP #148: /proc/sys/fs/xfs/age_buffer_centisecs: No such file or directory

rabin-io, 2015-07-05:
On Fedora 22 I'm seeing this message when I look on the journal logs, 

    # journalctl --unit tlp.service -b0
    Jul 05 10:47:52 One systemd[1]: Started TLP system startup/shutdown.
    Jul 05 10:47:52 One systemd[1]: Starting TLP system startup/shutdown...
    Jul 05 10:47:52 One tlp[21673]: Loading kernel modules...done.
    Jul 05 10:47:52 One tlp[21673]: Applying power save settings.../usr/lib/tlp-pm/tlp-functions: line 468: /proc/sys/fs/xfs/age_buffer_centisecs: No such file or directory
    Jul 05 10:47:52 One tlp[21673]: /usr/lib/tlp-pm/tlp-functions: line 470: /proc/sys/fs/xfs/xfsbufd_centisecs: No such file or directory
    Jul 05 10:47:52 One tlp[21673]: done.
    Jul 05 10:47:52 One tlp[21673]: Setting battery charge thresholds...done.

linrunner, 2015-07-05:
Hi, this is already known and fixed for 0.8: https://github.com/linrunner/TLP/commit/91311307a0a5dd7fc84026a13ff0e0536e722370

rabin-io, 2015-07-05:
Oh, thanks, 
I did search in the open&closed issues, and didn't find any thing, so i opened this bug report.

linrunner, 2015-08-12:
Released.

closed 2015-08-12 by linrunner