linrunner/TLP #149: TLP does not switch between AC/BAT modes

BLuFeNiX, 2015-07-06:
Running TLP on a Lenovo y50-70. User must run `sudo tlp start` after (un)plugging laptop, otherwise TLP will not change to the proper mode.

tlp-stat output:

--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=30
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
DISK_SPINDOWN_TIMEOUT_ON_AC="0 0"
DISK_SPINDOWN_TIMEOUT_ON_BAT="0 0"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau nvidia"
USB_AUTOSUSPEND=1
USB_DRIVER_BLACKLIST="usbhid"
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0
DEVICES_TO_DISABLE_ON_STARTUP="bluetooth wwan"
DEVICES_TO_ENABLE_ON_STARTUP="wifi"
DEVICES_TO_ENABLE_ON_AC="bluetooth wifi"

+++ System Info
System         = LENOVO Lenovo Y50-70 20378
BIOS           = 9ECN37WW(V2.01)
Release        = "Arch Linux"
Kernel         = 4.0.7-2-ARCH x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-linux root=UUID=7f8c0d1c-5137-486e-bafe-a60a3456f77c rw resume=UUID=a71800d8-5753-4da6-9b6e-d03eae0b7119 quiet
Init system    = systemd

+++ System Status
TLP power save = enabled
[... 151 lines cut ...]
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
driver blacklist   = usbhid
wwan blacklist     = enabled (cdc_acm cdc_wdm cdc_ether hso qcserial sierra)

Bus 002 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 002 ID 8087:8008 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 003 Device 004 ID 05a7:1020 control = on,   autosuspend_delay_ms =  2000 -- Bose Corp.  (snd-usb-audio, usbhid)
Bus 003 Device 003 ID 8087:07dc control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 003 Device 002 ID 174f:14b8 control = auto, autosuspend_delay_ms =  2000 -- Syntek  (uvcvideo)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT1/manufacturer                   = LENOVO 
/sys/class/power_supply/BAT1/model_name                     = PABAS0241231
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  55480 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  61030 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  25320 [mWh]
/sys/class/power_supply/BAT1/power_now                      =  37800 [mW]
/sys/class/power_supply/BAT1/status                         = Charging

linrunner, 2015-07-06:
Hi, this seems a duplicate of #147. Please post your traces as described there.

vahdani, 2015-07-10:
I have the exact same problem on debian (sid)

linrunner, 2015-07-14:
See #147 for the solution.

closed 2015-07-14 by linrunner