linrunner/TLP #150: Maximum cpu boost frequency?

spacepluk, 2015-07-15:
Hi, is it possible to set a maximum frequency for the CPU_BOOST option?  For some reason when the boost is enabled it ignores these settings...

````
CPU_SCALING_MIN_FREQ_ON_AC=0
CPU_SCALING_MAX_FREQ_ON_AC=3000000
CPU_SCALING_MIN_FREQ_ON_BAT=0
CPU_SCALING_MAX_FREQ_ON_BAT=3000000
````
... and goes beyond 3Ghz, is that the intended behavior?

linrunner, 2015-07-19:
Hi, please give the full output of

    sudo tlp-stat

(via https://gist.github.com/)

spacepluk, 2015-07-19:
Here it is: https://gist.github.com/spacepluk/28f59d1032419cf1cd92

linrunner, 2015-07-19:
TLP applies your settings,

    /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
    /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3000000 [kHz]

so the behaviour must stem from either intel_pstate or the processor hardware itself. I honestly don't know.

However some kind of perfomance limit is effective implicitly:

    /sys/devices/system/cpu/intel_pstate/max_perf_pct      = 83

TLP has related settings (CPU_MAX_PERF_ON_*).

I suggest you dig into intel_pstate docs [1] or ask in an adequate Linux forum.

[1] https://www.kernel.org/doc/Documentation/cpu-freq/intel-pstate.txt

spacepluk, 2015-07-19:
Cool, thanks for you help!

closed 2015-07-25 by linrunner