linrunner/TLP #151: Mint 17.2 locks up with blank screen after installed TLP

stevebinns, 2015-07-15:
After installing TLP on a thinkpad T60 (using the instructions from the TLP web page) the system locks with a blank screen probably after screen save. Sytem info follows then output from tlp-stat
-Computer-
Processor		: 2x Intel(R) Core(TM)2 CPU         T5600  @ 1.83GHz
Memory		: 2063MB (1551MB used)
Operating System		: Linux Mint 17.2 Rafaela
User Name		: steve (Steve Binns)
Date/Time		: Wed 15 Jul 2015 13:51:30 BST
-Display-
Resolution		: 1400x1050 pixels
OpenGL Renderer		: Gallium 0.4 on ATI RV515
X11 Vendor		: The X.Org Foundation
-Multimedia-
Audio Adapter		: HDA-Intel - HDA Intel
Audio Adapter		: ThinkPad EC - ThinkPad Console Audio Control
-Input Devices-
 Lid Switch
 Sleep Button
 Power Button
 AT Translated Set 2 keyboard
 Video Bus
 SynPS/2 Synaptics TouchPad
 Microsoft Basic Optical Mouse
 TPPS/2 IBM TrackPoint
 ThinkPad Extra Buttons
-Printers (CUPS)-
Canon-MG5300		: <i>Default</i>
Canon-MG6400
NewCanon
PDF
-SCSI Disks-
HL-DT-ST DVDRAM GSA-4083N
ATA HITACHI HTS54161
--------------------------------- tlp-stat
--- TLP 0.7 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
[... 125 lines cut ...]
/sys/bus/pci/devices/0000:03:00.0/power/control = on   (0x028000, Network controller, ath9k)
/sys/bus/pci/devices/0000:15:00.0/power/control = on   (0x060700, CardBus bridge, yenta_cardbus)

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
driver blacklist   = usbhid
wwan blacklist     = enabled (cdc_acm cdc_wdm cdc_ether hso qcserial sierra)

Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 005 Device 003 ID 0483:2016 control = auto, autosuspend_delay_ms =  2000 -- STMicroelectronics Fingerprint Reader (no driver)
Bus 005 Device 002 ID 0a5c:2110 control = auto, autosuspend_delay_ms =  2000 -- Broadcom Corp. BCM2045B (BDC-2) [Bluetooth Controller] (btusb)
Bus 005 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 004 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 003 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 002 Device 002 ID 045e:0083 control = on,   autosuspend_delay_ms =  2000 -- Microsoft Corp. Basic Optical Mouse (usbhid)
Bus 002 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)

+++ ThinkPad Extended Battery Functions
tp-smapi   = active
tpacpi-bat = inactive (unsupported hardware)

+++ Suggestions
* Install ethtool to disable Wake On LAN (WOL)

linrunner, 2015-07-19:
Hi, please clarify your description:

> the system locks with a blank screen probably(?) after screen save

What are you doing and what happens exactly?

I suggest to uninstall TLP and check if the symptom disappears.

stevebinns, 2015-07-19:
Unfortunately it's not particularly reproducible. I did remove TLP and the system behaved OK. I've now re-installed TLP and seen the following.
1 System seems OK with battery removed, with external power. 
2 With battery installed and no external power, starting a lock screen produces a blank screen (not the usual flying star field) and moving the mouse gives a very jerky movement of the cursor and the password request does not appear. Putting external power on fixes this problem.
3 I then left the system with battery and charger operating and came back to aforementioned black screen with no way to exit except turn of power and remove battery.
4 After this tried to reboot with just battery installed and got background screen with no icons or aplets and no response to mouse or keyboard.

Sorry it's all a bit vague. 
2 With battery installed and no external power, starting a lock screen produces a blank screen (not the usual flying star field) and moving the mouse gives a very jerky movement of the cursor and the password request does not appear. Putting external power on fixes this problem.
3 I then left the system with battery and charger operating and came back to aforementioned black screen with no way to exit except turn of power and remove battery.
4 After this tried to reboot with just battery installed and got background screen with no icons or aplets and no response to mouse or keyboard.

Sorry it's all a bit vague.

linrunner, 2015-07-19:
Your BIOS version is *way* outdated. Update to the [latest version](http://support.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t60/downloads/DS013722).

stevebinns, 2015-07-19:
OK thanks, I've now done that. 

Behaviour a bit better but a long way from perfect. Typical behaviour running on battery:
start up system, start chrome browser - takes about 1 min. Close chrome and restart it, it creates an icon on the panel but won't display the window. If I click on the icon it's as if I double click. It clears the desktop image then quickly redraws it. The chrome window never appears. With battery and charger chrome starts up in about 20 odd secs and operated normally - it's what I'm using now. I'll remove TLP and try again

stevebinns, 2015-07-19:
Now removed TLP and running on battery, chrome starts up in less than 20 secs and can be stopped and restarted with no apparent problem. Started lock screen with no problem. Anything else to try?

linrunner, 2015-07-19:
I can only guess at this stage:

(1) You should get the latest updates for kernel 3.13, as Mint (sadly) defaults to not update kernels:

    sudo apt-get install linux-generic

or your may try 3.16 with

    sudo apt-get install linux-generic-lts-utopic

(2) Related to your performance issue i would experiment with less aggressive values for DISK_APM_LEVEL_ON_BAT and SATA_LINKPWR_ON_BAT – see [TLP settings docs](http://linrunner.de/en/tlp/docs/tlp-configuration.html#disks) 

(3) Regarding graphics i would experiment with
a. comment out RUNTIME_PM_ON_* 
b. choose less aggressive radeon settings or try to enable DPM – see [docs](http://linrunner.de/en/tlp/docs/tlp-configuration.html#graphics)

Remark: the ATI X1400 is nice especially in winter (*huge* savings on your heating bill) and the radeon driver is particularly buggy for that old chip. I had a similarly specced T60 some years ago. Maybe this is the root of your problems with power saving.

I happen to have installed an Intel Graphics T60 (Hitachi HDD) with Xubuntu 14.04.2 this morning and observed none of your symptoms.

stevebinns, 2015-07-20:
Many thanks for this info. I'll pursue a bit further but as you say the problems may well be all down to the graphics set - the system does get very hot. On a separate issue (as you seem very knowledgeable!)  I've noticed that I can't get the screen brightness on battery power up to the same level as when on external power. Can this be fixed?

linrunner, 2015-07-20:
In the BIOS: Config --> Display --> Brightness

stevebinns, 2015-07-22:
OK many thanks for your help. I've upgraded the kernel to 3.13 and enabled DPM. I still have the freeze problem with TLP on battery. Typically I have a chrome window I can move the cursor around but clicking anywhere has no effect. I have to hold down the power button for 5 secs for it to halt. I can't see that tweaking settings would alter that behaviour so I've removed TLP for now. The screen is brighter now too. Thanks again.

closed 2015-07-22 by stevebinns