linrunner/TLP #160: T450s and tp_smapi

dmaglio, 2015-09-10:
I have installed tlp from website, but the module tp_smapi is always inactive
output tlp stat -b:
```
--- TLP 0.8 --------------------------------------------

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' load error)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SONY
/sys/class/power_supply/BAT0/model_name                     = 45N1111
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  23200 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  23200 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  23200 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

tpacpi-bat.BAT0.startThreshold                              =     40 [%]
tpacpi-bat.BAT0.stopThreshold                               =     50 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = LGC
/sys/class/power_supply/BAT1/model_name                     = 45N1127
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  23480 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  23460 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  21050 [mWh]
/sys/class/power_supply/BAT1/power_now                      =  15681 [mW]
/sys/class/power_supply/BAT1/status                         = Charging

tpacpi-bat.BAT1.startThreshold                              =     96 [%]
tpacpi-bat.BAT1.stopThreshold                               =    100 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0
```
output modprobe tp_smapi:
```
modprobe: ERROR: could not insert 'tp_smapi': No such device or address
```
output uname -a:
```
Linux thinkpad-t450s 4.1.6-200.fc22.x86_64 #1 SMP Mon Aug 17 19:54:31 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux
```
I use Fedora 22 x86_64

linrunner, 2015-09-10:
Hi, tp-smapi is incompatible with your hardware (since *30 Thinkpad generation). You may safely ignore the message.

closed 2015-09-10 by linrunner

kas, 2016-08-06:
Hi @linrunner, 

I found this issue from googling a problem I ran into

I have to remove tp-smapi from my arch linux T450s because it is not letting systemd-modules-load.service load with systemctl.

I noticed that I currently have acpi_call installed (https://github.com/mkottman/acpi_call).

Should TLP function fine or should I install tcapi-bat as well?

I appreciate your help in advance, and hope that my question will help others who have the same problem as me.

linrunner, 2016-08-06:
tp-smapi does not support the T450s, so you're just fine with acpi-call. tpacpi-bat comes as part of the tlp package.

kas, 2016-08-06:
@linrunner Awesome, thank you for getting back to me and thank you for creating this software. It has immensely improved how long I can use my laptop away from the outlet!!