linrunner/TLP #161: Can't mange battery on Fedora 22 with Kernel 4.1.7-200.  Source RPM missing patch

davejohnston, 2015-10-02:
TLP was installed on the latest Fedora 22 running kernel 4.1.7-200.

Trying to manage battery charge levels failed due to acpi_call not bieng available.
The kernel modules from  http://repo.linrunner.de/fedora/tlp/repos/releases/22/x86_64/os/

Are built for 4.0.4-301.fc22.x86_64

Rebuilding the src RPM

http://repo.linrunner.de/fedora/tlp/repos/releases/22/sources/SRPMS/acpi_call-kmod-1.1.0-4.fc22.src.rpm

fails with the following error:

error: File /root/rpmbuild/SOURCES/acpi_call-1.1.0.fc22-x86_64.patch is smaller than 13 bytes
error: Failed build dependencies:
	buildsys-build-rpmfusion-kerneldevpkgs-current-x86_64 is needed by acpi_call-kmod-1.1.0-4.fc22.x86_64

linrunner, 2015-10-02:
Hi, i've forwarded your problem to the package's maintainer and will keep you posted.

linrunner, 2015-10-13:
Seems you didn't install **a**kmod-acpi_call (right) but kmod-acpi_call (wrong).

If you follow the instructions

    dnf install akmod-acpi_call kernel-devel

closely, an kmod-acpi_call package matching your kernel will be generated and installed automatically and all build dependencies installed as well.

closed 2015-11-17 by linrunner