linrunner/TLP #162: RUNTIME_PM makes suspend buggy

prafiny, 2015-10-19:
Running a asus netbook (UX303LN) with TLP on arch, I noticed a strange bug when using runtime power management for PCI.

I don't know if other people do get this issue around here.

linrunner, 2015-10-19:
Hi, this is a (common) driver/kernel issue. I suggest to:

1. Try a newer/older kernel
2. Disable completely on battery via `RUNTIME_PM_ON_BAT=on`
3. Blacklist PCIe devices *one by one* in RUNTIME_PM_BLACKLIST= and re-check to isolate the offending device

prafiny, 2015-11-15:
I blacklisted the discrete graphic card (GeForce 840M) and I achieved a 3 days uptime (3 days without i.e. having the bug while suspending and being bound to reboot)

I will let it a few more days to be sure but this seems to be a good track.

To be noted that this nvidia card is taken care of by bbswitch (which keeps it off most of the time) and bumblebee (which enables it on demand.)

linrunner, 2015-11-16:
Thanks for your feedback. 

Do you use the proprietary driver `nvidia`? Because `nouveau` is blacklisted by default already within RUNTIME_PM_DRIVER_BLACKLIST.

prafiny, 2015-11-16:
I use it but through bumblebee and bbswitch. I guess it's not loaded 
unless I use optirun/primusrun, is it ?

Thanks for your support

linrunner, 2015-11-16:
Please show 

    tlp-stat -e

before activating the discrete card.

prafiny, 2015-11-16:
```
/sys/bus/pci/devices/0000:00:02.0/power/control = auto (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:03.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:1b.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.3/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.4/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, lpc_ich)
/sys/bus/pci/devices/0000:00:1f.2/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x0c0500, SMBus, no driver)
/sys/bus/pci/devices/0000:00:1f.6/power/control = auto (0x118000, Signal processing controller, no driver)
/sys/bus/pci/devices/0000:02:00.0/power/control = auto (0x028000, Network controller, iwlwifi)
/sys/bus/pci/devices/0000:03:00.0/power/control = auto (0x030200, 3D controller, no driver)
```

prafiny, 2015-11-17:
The bug  has not occured since 5 days so I guess the device that was staying on and blocking the suspend process was the discrete graphics card.

prafiny, 2015-11-17:
I should have let it longer before saying anything. In fact the bug did occur again so there might are something else.

linrunner, 2015-11-17:
From your posts i gather that blacklisting the discrete graphics mitigates the problem i.e. it occurs less often. Right? If yes, it's still a valid approach.

The alternative would be to file a bug report against Nvidia.

prafiny, 2015-11-19:
I might have understood that it is the hdd that doesn't stop. I blacklisted it and after a few hours and suspends, the laptop doesn't want to suspend anymore. Here is the log it yields : 

http://pastebin.com/x26GKiVf

linrunner, 2015-11-19:
Did you try to uninstall TLP competely yet?

prafiny, 2015-11-19:
No but why are you suggesting that ? Because when I deactivate RUNTIME_PCI, the bug is gone.

linrunner, 2015-11-19:
Why do you write about blacklisting the hdd (drive) then?

This is a kernel / driver bug by nature and not an issue with TLP itself. Can't help you any further than what i already wrote in post #2.

prafiny, 2015-11-19:
A bug has already been filed to the kernel bugtracker.

https://bugzilla.kernel.org/show_bug.cgi?id=105301

prafiny, 2015-11-20:
BTW, when I said I blacklisted, I meant blacklisted the device in RUNTIME_PM

linrunner, 2015-11-20:
OK, you mean the SATA controller not the HDD.

Did you try to disable ALPM?

prafiny, 2015-11-21:
Setting SATA_LINKPWR_ON_BAT to max_performance ?

linrunner, 2015-11-21:
Yes or comment both lines so TLP doesn't touch the setting at all.

prafiny, 2015-12-02:
Bug still there. I guess part or all of the pci controllers do trigger the bug.

closed 2016-01-10 by linrunner