linrunner/TLP #166: systemctl mask systemd-rfkill@.service

Zverushko, 2015-11-03:
unit name systemd-rfkill@.service is not valid 

fedora 23

how fix that?

thx!

dywisor, 2015-11-03:
Due to changes in systemd v227, you now need to mask systemd-rfkill.service and its corresponding .socket unit:

```systemctl mask systemd-rfkill.socket systemd-rfkill.service```

refs:
* [release notes item](https://github.com/systemd/systemd/blob/v227/NEWS#L162-L167)
* [commit that introduced this change](https://github.com/systemd/systemd/commit/d35c1bb1f4b90993124f817f6f54a933e3be5f6b)

linrunner, 2015-11-04:
@dywisor: F23 has systemd v222 afaik. 

@Zverushko: Please provide:

    systemctl list-unit-files | grep rfkill
    rpm -q systemd
    rom -ql systemd

linrunner, 2015-11-06:
The error message is wrong, the service is masked correctly.

linrunner, 2015-11-06:
Installation instructions updated.

closed 2015-11-06 by linrunner

brianpepino, 2020-06-11:
I am running Fedora 32 and TLP 1.3.1 and I am still seeing this notice - "Notice: systemd-rfkill.socket is not masked -- invoke "systemctl mask systemd-rfkill.socket" to correct this!"

Any update or guidance? Thank you in advance.