linrunner/TLP #167: wifi power_save off after boot/resume

berte78, 2015-11-08:
Hi,

After upgrading Ubuntu 15.04->15.10 the power_save of my wifi is not set to on anymore after boot and resume. My wifi card does support the power_save and it was working well in Ubuntu 15.04. Moreover, if I restart tlp manually (sudo tlp start), power_save gets enabled. All other options are OK.
I already checked with kernel 3.19 and I observed the same behaviour. I don't think I have any other power management tool enabled, at least not that I'm aware.

Thank you in advance for looking at this issue.

Here are some details of my system:

**Computer:** Asus UX305FA, Intel® Core™ M-5Y71 CPU @ 1.20GHz × 4
**OS:** Ubuntu 15.10, 4.2.0-17-generic

**lspci** => Network controller: Intel Corporation Wireless 7265 (rev 59)

**sudo tlp-stat**  =>

--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = ASUSTeK COMPUTER INC. 1.0 UX305FA
BIOS           = UX305FA.206
Release        = Ubuntu 15.10
Kernel         = 4.2.0-17-generic #21-Ubuntu SMP Fri Oct 23 19:56:16 UTC 2015 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.2.0-17-generic.efi.signed root=UUID=d121a7e0-3d34-4974-aa89-364659ebe63b ro quiet splash vt.handoff=7
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = Intel(R) Core(TM) M-5Y71 CPU @ 1.20GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2900000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu1/cpufreq/scaling_max_freq  =  2900000 [kHz]

/sys/devices/system/cpu/cpu2/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu2/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu2/cpufreq/scaling_max_freq  =  2900000 [kHz]

/sys/devices/system/cpu/cpu3/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu3/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu3/cpufreq/scaling_max_freq  =  2900000 [kHz]

/sys/devices/system/cpu/intel_pstate/min_perf_pct      = 17
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100
/sys/devices/system/cpu/intel_pstate/no_turbo          = 0

x86_energy_perf_policy.cpu0                            = powersave
x86_energy_perf_policy.cpu1                            = powersave
x86_energy_perf_policy.cpu2                            = powersave
x86_energy_perf_policy.cpu3                            = powersave

/proc/sys/kernel/nmi_watchdog                          = 0

+++ Undervolting
PHC kernel not available.

+++ Temperatures
CPU temp               =    40 [°C]
Fan speed (fan1)       =     0 [/min]

+++ File System
/proc/sys/vm/laptop_mode               =     2
/proc/sys/vm/dirty_writeback_centisecs =  6000
/proc/sys/vm/dirty_expire_centisecs    =  6000
/proc/sys/vm/dirty_ratio               =    20
/proc/sys/vm/dirty_background_ratio    =    10
/proc/sys/fs/xfs/age_buffer_centisecs  = (not available)
/proc/sys/fs/xfs/xfssyncd_centisecs    = (not available)
/proc/sys/fs/xfs/xfsbufd_centisecs     = (not available)

+++ Storage Devices
/dev/sda:
          Model     = Micron_M600_MTFDDAV256MBF               
          Firmware  = MA01    
          APM Level = 128
          Status    = active/idle
          TRIM      = supported
          Scheduler = deadline

        SMART info:
            5 Reallocated_Sector_Ct     =        0 
            9 Power_On_Hours            =      173 [h]
          194 Temperature_Celsius       =       34 (Min/Max 17/57)  [°C]

+++ SATA Aggressive Link Power Management
/sys/class/scsi_host/host0/link_power_management_policy  = min_power
/sys/class/scsi_host/host1/link_power_management_policy  = min_power

+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = default (using bios preferences)

+++ Intel Graphics
/sys/module/i915/parameters/powersave        = (not available)
/sys/module/i915/parameters/enable_rc6       =  1 (enabled)
/sys/module/i915/parameters/enable_fbc       = -1 (use per-chip default)
/sys/module/i915/parameters/lvds_downclock   =  0 (disabled)
/sys/module/i915/parameters/semaphores       = -1 (use per-chip default)

+++ Wireless
bluetooth = off (software)
wifi      = on
wwan      = none (no device)

hci0(btusb)         : not connected
wlan0(iwlwifi)      : connected, power management = off

+++ Audio
/sys/module/snd_hda_intel/parameters/power_save            = 1
/sys/module/snd_hda_intel/parameters/power_save_controller = Y

+++ Runtime Power Management
device classes   = all
device blacklist = (not configured)
driver blacklist = radeon nouveau

/sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, bdw_uncore)
/sys/bus/pci/devices/0000:00:02.0/power/control = auto (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:03.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:1b.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.3/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, lpc_ich)
/sys/bus/pci/devices/0000:00:1f.2/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x0c0500, SMBus, no driver)
/sys/bus/pci/devices/0000:00:1f.6/power/control = auto (0x118000, Signal processing controller, no driver)
/sys/bus/pci/devices/0000:02:00.0/power/control = auto (0x028000, Network controller, iwlwifi)

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 003 ID 064e:9700 control = auto, autosuspend_delay_ms =  2000 -- Suyin Corp. Asus Integrated Webcam (uvcvideo)
Bus 001 Device 002 ID 8087:0a2a control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = ASUS Battery
/sys/class/power_supply/BAT0/cycle_count                    =     72
/sys/class/power_supply/BAT0/energy_full_design             =  45030 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  42601 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  40287 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   7945 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

**iw wlan0 get power_save** => Power save: off

**sudo tlp start && iw wlan0 get power_save** =>
TLP started in battery mode.
Power save: on

linrunner, 2015-11-08:
Hi, 

please enable trace mode [1] and show the output for a. system startup and b. suspend / resume – preferably via Gist [2].

Furthermore i need the output of

    systemctl status tlp.service
    systemctl status tlp-sleep.service

[1] http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#support
[2] https://gist.github.com/

berte78, 2015-11-08:
Hi,

Thank you for your fast answer.

Below is the info you requested:

[trace_startup.txt] (https://gist.github.com/berte78/eab54c2b4d76ba4370a5#file-trace_startup-txt)
[trace_suspend-resume.txt] (https://gist.github.com/berte78/71600343c1ef32d8c888#file-trace_suspend-resume-txt)
[tlp-service.txt] (https://gist.github.com/berte78/a5a5c4b5e1fdb2db85d2#file-tlp-service-txt)
[file-tlp-sleep-service.txt] (https://gist.github.com/berte78/a8f5caab73eb3c05b5a9#file-tlp-sleep-service-txt)

linrunner, 2015-11-09:
The traces look fine.

I suspect the iwlwifi driver may drop power save mode when connecting and/or disconnecting wifi. Colud you check both cases please?

berte78, 2015-11-09:
Your hypothesis is correct. With power_save on, if I disable wifi, then power_save is off. After re-enabling wifi, power_save stays off. So this issue seems more related to the iwlwifi driver than TLP.
I will report this bug under launchpad/Ubuntu.

I suppose I can close this issue now. Thank you very much for your help.
Regards.

closed 2015-11-09 by berte78