linrunner/TLP #169: Battery not charging after recalibrate

davidebasilio, 2015-11-12:
I have a thinkpad T450 and I tried to recalibrate the removable battery (BAT1).

The discharge went succesfully, but the battery is reportedly not charging anymore now.
This is what I get:

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = SANYO
/sys/class/power_supply/BAT1/model_name                     = 45N1775
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  23200 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  22530 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =      0 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown (threshold effective)

tpacpi-bat.BAT1.startThreshold                              =     96 [%]
tpacpi-bat.BAT1.stopThreshold                               =    100 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

linrunner, 2015-11-12:
Hi,

batteries not charging are *never* an issue with TLP itself.

You're aware that ThinkPads charge only one battery at a time? I reckon the internal battery is currently charging. Just be patient or show the output of `tlp-stat -b` for both batteries please :-)

davidebasilio, 2015-11-12:
OK, so the catch was that the controller will always only operate on one battery at a time.
The other one was actually discharging (I was recalibrating it, too), so BAT1 would not charge.

Thanks a lot for the quick feedback and for the great tool!

closed 2015-11-12 by davidebasilio

linrunner, 2015-11-13:
Thanks for your feedback. Led me to improve the docs [1].

[1] http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#recalibrate