linrunner/TLP #170: Recalibration does not terminate after discharge

Emmes90, 2015-11-12:
I have a Thinkpad E550. I experienced erratic battery behavior after changing the thresholds, so I reset them and tried to recalibrate the battery as recommended in the FAQ.

When the battery reaches 1% it starts charging again, but the recalibration command does not terminate. It still says that it's discharging, but the percentage, the remaining capacity and the time estimate keep going up.

The System itself recognizes correctly that the battery is charging.

linrunner, 2015-11-13:
Hi.
> When the battery reaches 1% it starts charging again, but the recalibration command does not terminate. 
Please provide the output of

    sudo tlp-stat -s -b

for this point in time  (in another terminal).

The tlp discharge routine just waits until the hardware resets the forceDischarge flag. Considering that the E-series' firmware (BIOS) has a history of such strange bugs (imho), i'm not surprised. But let's see.

Emmes90, 2015-11-24:
Hey, sorry to keep you waiting. I haven't been using my laptop all that much in the past few days.

It is as you probably already suspected: forceDischarge stays at 1, even when the battery is charging again.

linrunner, 2015-12-24:
I changed the discharge routine to wait for status to become != "Discharging" instead of forceDischarge = 0. Will be released with TLP 0.9.

linrunner, 2016-08-19:
Released with 0.9.

closed 2016-08-19 by linrunner