linrunner/TLP #172: TLP does not change profile even though it noticed power source change

sos-michael, 2015-11-16:
Basically tlp does not change state until I run "sudo tlp start" after unplugging/plugging in

Let walk through this... 

****************
my system config (fedora 23 with nothing to fancy)
****************
+++ System Info
System         = VAIO Corporation Reserved VJZ12A
BIOS           = R1060VE
Release        = "Fedora release 23 (Twenty Three)"
Kernel         = 4.2.5-300.fc23.x86_64 #1 SMP Tue Oct 27 04:29:56 UTC 2015 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.2.5-300.fc23.x86_64 root=UUID=32425441-f260-4ee9-9000-36e45445f011 ro rd.md.uuid=df3c3c2d:a16312a2:d7690ba4:3b4c9a13 rd.md.uuid=fef1dc37:4adde613:69a636f5:65f484fe modprobe.blacklist=int3403_thermal add_efi_memmap intel_pstate=enable i915.modeset=1 i915.lvds_use_ssc=1 i915.semaphores=1 i915.enable_rc6=7 i915.enable_fbc=1 i915.disable_pch_pwm=0 i915.lvds_downclock=1 drm.vblankoffdelay=1 pcie_aspm=force rhgb quiet
Init system    = systemd

*****************
my tlp config
*****************

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=120
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=23
CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="1 1"
DISK_IOSCHED="noop noop"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=1

*************************************************************
Step 1) I unplug, system(and TLP) notices I'm battery, but TLP does not change state: (output from tlp-stat)
*************************************************************

--- TLP 0.8 --------------------------------------------
+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
[... 734 lines cut ...]

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 002 Device 002 ID 0bda:56c0 control = auto, autosuspend_delay_ms =  2000 -- Realtek Semiconductor Corp.  (uvcvideo)
Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 003 ID 064e:9917 control = auto, autosuspend_delay_ms =  2000 -- Suyin Corp.  (uvcvideo)
Bus 001 Device 002 ID 2a7e:0100 control = on,   autosuspend_delay_ms =  2000 -- <unknown> (usbhid)
Bus 001 Device 006 ID 1b96:0f0c control = on,   autosuspend_delay_ms =  2000 -- N-Trig  (usbhid)
Bus 001 Device 005 ID 0483:91d1 control = on,   autosuspend_delay_ms =  2000 -- STMicroelectronics  (usbhid)
Bus 001 Device 004 ID 8087:0a2a control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = VAIO Corp.
/sys/class/power_supply/BAT0/model_name                     = VJ8BPS47
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  65970 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  65970 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  32730 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown

linrunner, 2015-11-16:
Hi,

> fedora 23 with nothing too fancy

i'm unable to reproduce this here with F23. So maybe there is something less fancy  about your system-udevd (udevd detects change of power source and calls tlp)? 

1. I'd like to see

    rpm -qa | egrep 'systemd|tlp' | sort

2. Enable [trace mode](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#trace) and show the trace output for a. ac --> bat, b. bat --> ac

What about the settings when you start the laptop on battery?

Please show all outputs via [Gist](https://gist.github.com/) to spare us excessive scrolling ...

sos-michael, 2015-11-16:
You got it:
[sos@ursus ~]$ rpm -qa | egrep 'systemd|tlp' | sort
python3-systemd-230-1.fc23.x86_64
rpm-plugin-systemd-inhibit-4.13.0-0.rc1.6.fc23.x86_64
systemd-222-8.fc23.x86_64
systemd-compat-libs-222-8.fc23.x86_64
systemd-libs-222-8.fc23.x86_64
tlp-0.8-1.fc23.noarch

Here is the trace file in gist form. I unplugged, plugged in, unplugged, plugged in(_without_ calling sudo tlp start inbetween) but didn't notice the log accumulate any new data.. perhaps I did not do something correctly...

https://gist.github.com/sos-michael/ee339975a7ff46d26e2f#file-gistfile1-txt

I'll do a reboot an edit this in a second.

Thanks again for looking into this!

sos-michael, 2015-11-17:
Did a fresh install, just installed fedora, updated, and then installed and configured tlp. No go.

linrunner, 2015-11-17:
Sorry, but reinstalling before the analysis has finished is counterproductive.

TLP depends on udev events when plugging/unplugging the ac adapter (events originate in the ACPI BIOS). I suspect your laptop may not send these events. Enable udev logging (as root) via

    udevadm monitor --udev --property --subsystem-match=power_supply

and plug / unplug your ac adapter and show the output.. 

You should see events like this:

    UDEV  [3529.742322] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:08/PNP0C09:00/ACPI0003:00/power_supply/AC (power_supply)
    ACTION=change
    DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:08/PNP0C09:00/ACPI0003:00/power_supply/AC
    POWER_SUPPLY_NAME=AC
    POWER_SUPPLY_ONLINE=1`
    SEQNUM=3186
    SUBSYSTEM=power_supply
    USEC_INITIALIZED=29433252

sos-michael, 2015-11-17:
I know, but I had gotten it in my head that I may have changed something obscure without realizing it and I have no real desire to send you on some obscure goose chase. But back to the point, I do get udev messages, which is what expected, seeing as gnome recognizes when i change power states....

https://gist.github.com/sos-michael/d69ae1c0d8597166a184#file-udev_output

Its worth note now that my power supply status is unknown when I re-plugin because my battery is set to not charge until a specific threshold is reached... it is also weird that I get double messages.

Just spit-balling... I  know fedora 23 doesn't use on-demand and just lets the kernel do its business, but could there be something else I need to disable?

linrunner, 2015-11-17:
Your output shows events from a battery discharging, but TLP by design works with AC plug/unplug events only (refer to TLP's [udev rule](https://github.com/linrunner/TLP/blob/master/tlp.rules.in#L7)). Compare yours with my example, where POWER_SUPPLY_NAME=AC. 

Imho these battery events do occur continously while a battery is discharging and therefore are not feasible to detect AC plug/unplug.

sos-michael, 2015-11-17:
Well, that's at least something I can tackle... There is no way I could turn this into a feature request where you make TLP a bit more robust in here? 

Alternatively, Am I correct in understanding that the current udev rule does the following:

On a udev power supply change that effects the attribute "mains"(which is to say, the A/C source) run the tlp.

So my modified version should read something along the lines of: 

On udev power supply changes that effect WHATEVER_ATTRIBUTE_NAME_FOR_BATTERY_IS, check to see if POWER_SUPPLY_STATUS has changed then run tlp.

Mind pointing me in the right direction for udev stuff without having me read whatever crazy long thing the manual might be? Is the Arch Wiki sufficiently complex?

Also to answer your second question, I was just shooting in the dark really, I though maybe fedora was being "extra safe" and re-initializing the boot time power state configuration on changes to ensure some type of continuity. Not exactly the most reasonable idea I've ever had, but like I said spit-balling.

linrunner, 2015-11-18:
Try to shorten the udev rule to

    ACTION=="change", SUBSYSTEM=="power_supply", RUN+="@TLP_SBIN@/tlp auto"

and check via TLP's trace how often it is called.

sos-michael, 2015-11-19:
It works and, on my machine, I know for a fact that specific udev message is only fired off about once every 70 seconds(I have no idea why so slow) on battery state changes. That said, perhaps a really idea solution(if it is not already in place) would that TLP acts a bit gracefully and when requested to change power states that are not real state changes IE from battery to battery or from ac to ac...

For myself I might have to properly write the udev rule so that its a bit more picky, just to save myself the trouble.

Still seeing the kernel power pref not being set... gonna file another bug report for that...

Thanks again

linrunner, 2015-11-19:
One user / one buggy BIOS (did you update?) is not enough evidence to consider a change at this time.

I suggest you duplicate the .rules file, change the rule there and leave the original file untouched so it doesn't get overwritten by updates of TLP.

Please post your final workound here.

moesasji, 2015-12-23:
I experience exactly the same issue as described in this bugreport on a fully up-to-date Slackware Current, with a 4.4.0-rc5 kernel. This is with tlp-0.8 as installed from Slackbuilds, so with the default udev rule. 

Not sure what information I need to provide in addition to help resolve this:

1) I tested with udev-adm. It shows the correct events on unplug; plug
```
UDEV  [2687.587705] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:18/PNP0C09:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:18/PNP0C09:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=99
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=6190000
POWER_SUPPLY_CHARGE_FULL_DESIGN=6300000
POWER_SUPPLY_CHARGE_NOW=6137000
POWER_SUPPLY_CURRENT_NOW=850000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=TPS
POWER_SUPPLY_MODEL_NAME=S10
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=7400000
POWER_SUPPLY_VOLTAGE_NOW=8258000
SEQNUM=3526
SUBSYSTEM=power_supply
USEC_INITIALIZED=87587307

UDEV  [2924.713762] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:18/PNP0C09:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:18/PNP0C09:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=99
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=6190000
POWER_SUPPLY_CHARGE_FULL_DESIGN=6300000
POWER_SUPPLY_CHARGE_NOW=6162000
POWER_SUPPLY_CURRENT_NOW=850000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=TPS
POWER_SUPPLY_MODEL_NAME=S10
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=7400000
POWER_SUPPLY_VOLTAGE_NOW=8376000
SEQNUM=3527
SUBSYSTEM=power_supply
USEC_INITIALIZED=24713366
```
2) output of a trace of tlp restart where I did an unplug/plug event at the end. Note that despite this event showing up in udev-adm; nothing is visible in the output of tlp-stat -T:

```
[root@aurora:~] # tlp-stat -T
--- TLP 0.8 --------------------------------------------

Dec 23 10:16:33 aurora tlp[3120,3119]: +++ mode=init restart (0.8) ++++++++++++++++++++++++++++++++++++++++
Dec 23 10:16:33 aurora tlp[3120,3119]: PATH=/usr/local/sbin:/usr/sbin:/sbin:/usr/local/bin:/usr/bin:/bin:/usr/games:/usr/lib64/java/bin:/usr/lib64/java/jre/bin:/usr/lib64/qt/bin:/usr/share/texmf/bin
Dec 23 10:16:33 aurora tlp[3120,3119]: power_source=ac
Dec 23 10:16:33 aurora tlp[3120,3119]: set_radio_device_states(start): enable= disable=bluetooth wwan
Dec 23 10:16:33 aurora tlp[3120,3119]: device_switch(bluetooth, off)
Dec 23 10:16:33 aurora tlp[3120,3119]: get_devc(bluetooth).not_present
Dec 23 10:16:33 aurora tlp[3120,3119]: device_switch(wwan, off)
Dec 23 10:16:33 aurora tlp[3120,3119]: get_devc(wwan).not_present
Dec 23 10:16:33 aurora tlp[3120,3119]: set_laptopmode(0): 0
Dec 23 10:16:33 aurora tlp[3120,3119]: set_dirty_parms(0): 1500
Dec 23 10:16:33 aurora tlp[3120,3119]: set_cpu_perf_pct(0).min.not_configured
Dec 23 10:16:33 aurora tlp[3120,3119]: set_cpu_perf_pct(0).max.not_configured
Dec 23 10:16:33 aurora tlp[3120,3119]: set_cpu_boost_all(0).not_configured
Dec 23 10:16:33 aurora tlp[3120,3119]: set_sched_powersave(0).not_available
[... 10 lines cut ...]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:00.0 [0x060000 Host Bridge]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:02.0 [0x030000]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:14.0 [0x0c0330]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:14.2 [0x118000]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:16.0 [0x078000]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:17.0 [0x010601]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1c.0 [0x060400]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.0 [0x060100]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.2 [0x058000]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.3 [0x040300 Audio device]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:00:1f.4 [0x0c0500]
Dec 23 10:16:33 aurora tlp[3120,3119]: set_runtime_pm(0).on: /sys/bus/pci/devices/0000:01:00.0 [0x028000 Wireless]
Dec 23 10:16:33 aurora tlp[3120,3119]: check_tpacpi: rc=127
Dec 23 10:16:33 aurora tlp[3120,3119]: check_tp_smapi: rc=127
Dec 23 10:16:33 aurora tlp[3120,3119]: check_tp_battery(BAT0): idx=0; str=; tpacpi=127; tpsmapi=127; rc=255
Dec 23 10:16:33 aurora tlp[3120,3119]: check_tpacpi: rc=127
Dec 23 10:16:33 aurora tlp[3120,3119]: check_tp_smapi: rc=127
Dec 23 10:16:33 aurora tlp[3120,3119]: check_tp_battery(BAT1): idx=0; str=; tpacpi=127; tpsmapi=127; rc=255
```
3) attached is the output of tlp-stat -v with AC plugged in and one where AC is disconnected, but without the restart of tlp

[tlp_info_4.4.0rc5.txt](https://github.com/linrunner/TLP/files/70629/tlp_info_4.4.0rc5.txt)
[tlp_info_4.4.0rc5_unplug.txt](https://github.com/linrunner/TLP/files/70630/tlp_info_4.4.0rc5_unplug.txt)

*) this is a new machine, so running the latest BIOS

linrunner, 2015-12-23:
Hi,

your shown events are *not* the ones needed. TLP uses events with POWER_SUPPLY_NAME=**AC**
 Example posted [here](https://github.com/linrunner/TLP/issues/172#issuecomment-157490892).

There is a solution [posted above](https://github.com/linrunner/TLP/issues/172#issuecomment-157608621). You didn't try it yet or do you?

Which manufacturer / model is your laptop?

moesasji, 2015-12-23:
Thanks for the quick response. To answer:

1) The udev output I showed is from "udevadm monitor --udev --property --subsystem-match=power_supply". So exactly the command you link to; no clue why it looks different on this laptop.

2) I unfortunately only spotted the solution after posting (see my edit at the bottom). Removing ATTR{type}=="Mains" from the default udev-rule does indeed make things work for me. 

3) This is not a branded laptop, very limited details here: https://www.pcspecialist.co.uk/notebooks/lafiteII/ (most things work on recent kernel, but the BIOS definitely is far from perfect)

linrunner, 2015-12-23:
Fine. I'm glad the [above solution](https://github.com/linrunner/TLP/issues/172#issuecomment-157608621) works for you.

I didn't mean you're using a wrong command, it's just the laptop not producing the expected AC events.

I'll leave this open to gather more evidence before deciding about a change.

linrunner, 2016-01-02:
Committed the change to detect *all* power_supply change events. Had to implement additional checks to prevent duplicate invocation of 'tlp auto' when the power source changes.

thecompu, 2016-02-01:
I am experiencing a very similar issue but I came to discover it in a different way. I realized that when restarting the system on battery, tlp would start up in Performance mode. Re-running tlp (sudo tlp start) would re assert battery mode. I confirmed this by running tlp-stat after subsequent sudo tlp start runs. 

I have _also_ confirmed that changing from AC to Battery and back is not changing the energy profile. I did try the workaround above but it did not work for me. udevadm shows the changes:

`UDEV  [2480.125126] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/ACPI0003:00/power_supply/AC0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/ACPI0003:00/power_supply/AC0
POWER_SUPPLY_NAME=AC0
POWER_SUPPLY_ONLINE=0
SEQNUM=2639
SUBSYSTEM=power_supply
USEC_INITIALIZED=80012905

UDEV  [2480.213366] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=100
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CYCLE_COUNT=1
POWER_SUPPLY_ENERGY_FULL=56289000
POWER_SUPPLY_ENERGY_FULL_DESIGN=56041000
POWER_SUPPLY_ENERGY_NOW=56289000
POWER_SUPPLY_MANUFACTURER=ASUSTeK
POWER_SUPPLY_MODEL_NAME=ASUS Battery
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_POWER_NOW=5621000
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER= 
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11310000
POWER_SUPPLY_VOLTAGE_NOW=11310000
SEQNUM=2642
SUBSYSTEM=power_supply
USEC_INITIALIZED=132196

UDEV  [2482.696671] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=100
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CYCLE_COUNT=1
POWER_SUPPLY_ENERGY_FULL=56289000
POWER_SUPPLY_ENERGY_FULL_DESIGN=56041000
POWER_SUPPLY_ENERGY_NOW=56289000
POWER_SUPPLY_MANUFACTURER=ASUSTeK
POWER_SUPPLY_MODEL_NAME=ASUS Battery
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_POWER_NOW=5756000
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER= 
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11310000
POWER_SUPPLY_VOLTAGE_NOW=11310000
SEQNUM=2648
SUBSYSTEM=power_supply
USEC_INITIALIZED=576286

UDEV  [2482.696727] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/ACPI0003:00/power_supply/AC0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/ACPI0003:00/power_supply/AC0
POWER_SUPPLY_NAME=AC0
POWER_SUPPLY_ONLINE=1
SEQNUM=2645
SUBSYSTEM=power_supply
USEC_INITIALIZED=460380
`
My 85-tlp.rules:

`# handle change of power source ac/bat
ACTION=="change", SUBSYSTEM=="power_supply", RUN+="/usr/sbin/tlp auto"

# handle added usb devices (exclude subdevices via DRIVER=="USB")
ACTION=="add", SUBSYSTEM=="usb", DRIVER=="usb", ENV{DEVTYPE}=="usb_device", RUN+="/lib/udev/tlp-usb-udev %p"
`

Two other things:

1. I followed the instalaltion instructions posted to the website, but I am not getting tlp 0.9 via apt-get. Therefore, I am running 0.8.

2. I have followed the directions to enable tlp-stat -T. I am receiving the typical error message about creating the 90-debug.conf with the proper information, but I am still not able run a trace. 

` /etc/rsyslog.d $ ls -la
total 28
drwxr-xr-x   2 root root  4096 Jan 31 20:23 .
drwxr-xr-x 146 root root 12288 Jan 31 20:03 ..
-rw-r--r--   1 root root   311 Mar 17  2012 20-ufw.conf
-rw-r--r--   1 root root  1655 Apr 18  2013 50-default.conf
-rw-r--r--   1 root root    68 Jan 31 20:23 90-debug.conf
`
`$ cat 90-debug.conf 
*.=debug;\
auth,authpriv.none;\
news.none;mail.none -/var/log/debug`

My laptop is an ASUS Zenbook UX305. I am caught up on the BIOS.

linrunner, 2016-02-01:
@thecompu your symptoms may be similar but your udev events suggest a different cause. Please open a separate issue for your problem. I always need the full output of

    tlp-stat

via https://gist.github.com/ – in your case: **after boot on battery and before any manual intervention**.

indigopsy, 2016-03-10:
hello there
i have custom tlp config
I set max perfomance on AC and BAT
When I power on my Lenovo r61i my scaling is on max perfomance
when I unplug AC scaling stays on max perfomance
BUT when i connect my AC back, scaling goes to power saving
if i disconnect AC again, scaling goes to max perfomance ????
weird
Can you help me?
tell me what do you need, I will send you everything

And god bless you for this amazing app :+1:

linrunner, 2016-03-11:
Hi, what i need is written in the post above yours and in the [docs](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#support). For all of your three steps of course.

whirm, 2016-03-12:
Even with the commited udev changes TLP still thinks it's always on battery for me.

`tlp ac/batt` switches settings correctly, but `tlp auto` always switches to bat regardless of the laptop being plugged in or not.

My udev events aren't showing   `POWER_SUPPLY_NAME=AC` either BTW.
 
Thanks!

```
--- TLP 0.8.900 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = Dell Inc. XPS 13 9350
BIOS           = 1.2.3
Release        = Debian GNU/Linux unstable (sid)
Kernel         = 4.5.0-rc7-amd64 #1 SMP Debian 4.5~rc7-1~exp1 (2016-03-07) x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-4.5.0-rc7-amd64 root=/dev/mapper/1k--vg-root ro quiet splash dloglevel=3 rd.systemd.show_status=false rd.udev.log-priority=3 vga=current pcie_aspm=force
Init system    = systemd v229
Boot mode      = UEFI

+++ TLP Status
State          = enabled
/usr/bin/tlp-stat: 222: printf: AM: expected numeric value
Last run       = 10:22:03,      0 sec(s) ago205,      0 sec(s) ago
Mode           = battery
Power source   = battery

Notice: systemd-rfkill.service is not masked -- invoke "systemctl mask systemd-rfkill.service" to correct this!

[... 116 lines cut ...]

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 003 ID 0c45:670c control = auto, autosuspend_delay_ms =  2000 -- Microdia  (uvcvideo)
Bus 001 Device 002 ID 0a5c:6412 control = auto, autosuspend_delay_ms =  2000 -- Broadcom Corp.  (no driver)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = DELL JHXPY53
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   7570 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   7068 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   6880 [mAh]
/sys/class/power_supply/BAT0/current_now                    =    679 [mA]
/sys/class/power_supply/BAT0/status                         = Charging

Charge                                                      =   97.3 [%]
Capacity                                                    =   93.4 [%]
```

linrunner, 2016-03-12:
@whirm : you have a different problem. Please open your own issue, it's getting crowded here. 

I can only guess if AC was actually connected for your output, so please tell me.

linrunner, 2016-08-19:
Released with 0.9.

closed 2016-08-19 by linrunner