linrunner/TLP #173: Wake up from suspend crash!

PlugaruT, 2015-11-29:
I have Ubuntu 15.10 installed on my machine and when using tlp, sometimes machine crashes when waking up from suspend. If I uninstall tlp I don't have this bug.

linrunner, 2015-11-29:
Hi,

i don't have my crystal ball at hand right now. If you don't mind please show the [required output](http://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#support).

Does this happen on battery / AC / both?

Suspend/resume bugs almost are a pain to isolate. I suggest you try to disable Runtime PM as recently described in [Issue 162](https://github.com/linrunner/TLP/issues/162#issuecomment-149264270). 

In case the above doesn't help, next try is to [disable USB autosuspend](http://linrunner.de/en/tlp/docs/tlp-configuration.html#usb).

PlugaruT, 2015-12-11:
I have disabled USB autosuspend and runtime PM, none of this helped. The only solution was to uninstall tlp. :(

closed 2015-12-11 by linrunner