linrunner/TLP #174: external hard disk drive not working after installing tlp

avirajkhare00, 2015-12-05:
I have 500 GB Toshiba HDD which takes 5V DC, 1 Amps current. I think HDD is unable to draw needed current from USB 3.0 port. Do I need to change settings because it was working fine before installing top.

linrunner, 2015-12-05:
Hi,

try to blacklist the device id via USB_BLACKLIST --> http://linrunner.de/en/tlp/docs/tlp-configuration.html#usb

Please show the full output of

    sudo tlp-stat

via https://gist.github.com/

avirajkhare00, 2015-12-06:
I did USB_BLACKLIST=0
Nothing happened. Also none of my usb devices are working. i.e pendrives, external HDD
Blow is my tlp-stat output after changing USB_BLACKLIST to 0 and plugging my Ext. HDD in

--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=0
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = ASUSTeK Computer Inc. 1.0 K54C
BIOS           = K54C.205
Release        = Debian GNU/Linux 8.2 (jessie)
Kernel         = 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt11-1+deb8u6 (2015-11-09) x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-3.16.0-4-amd64 root=UUID=414f2590-828b-462f-99c9-63595fe05c68 ro initrd=/install/gtk/initrd.gz quiet
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = Intel(R) Pentium(R) CPU B960 @ 2.20GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2200000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   800000 [kHz]
[... 94 lines cut ...]
+++ USB
autosuspend        = disabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 003 Device 007 ID 0480:a007 control = on,   autosuspend_delay_ms =  2000 -- Toshiba America Info. Systems, Inc. External Disk USB 3.0 (usb-storage)
Bus 003 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 004 ID 058f:a014 control = auto, autosuspend_delay_ms =  2000 -- Alcor Micro Corp. Asus Integrated Webcam (uvcvideo)
Bus 001 Device 003 ID 0489:e03d control = on,   autosuspend_delay_ms =  2000 -- Foxconn / Hon Hai  (no driver)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTek
/sys/class/power_supply/BAT0/model_name                     = K53--44
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  48400 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  42020 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  31130 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  13145 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging

linrunner, 2015-12-06:
OK, let's try to disable runtime pm for the usb controllers. The setting is:

    RUNTIME_PM_ALL=1

Reboot after changing settings.

avirajkhare00, 2015-12-07:
Now it's working but wicd is not working after installing tlp. I finally purged down tlp since wicd is unable to connect wifi.
blow is output-

--- TLP 0.8 --------------------------------------------

+++ Runtime Power Management
device classes   = Ethernet, Wireless, Audio, Host Bridge, [SD] Card Reader, Firewire
device blacklist = (not configured)
driver blacklist = radeon nouveau

/sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, no driver)
/sys/bus/pci/devices/0000:00:02.0/power/control = on   (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:16.0/power/control = on   (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:1a.0/power/control = on   (0x0c0320, USB controller, ehci-pci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = auto (0x040300, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1c.0/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.1/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.3/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.5/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = on   (0x0c0320, USB controller, ehci-pci)
/sys/bus/pci/devices/0000:00:1f.0/power/control = on   (0x060100, ISA bridge, lpc_ich)
/sys/bus/pci/devices/0000:00:1f.2/power/control = on   (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1f.3/power/control = on   (0x0c0500, SMBus, no driver)
/sys/bus/pci/devices/0000:02:00.0/power/control = auto (0x028000, Network controller, ath9k)
/sys/bus/pci/devices/0000:03:00.0/power/control = on   (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:04:00.0/power/control = auto (0x020000, Ethernet controller, atl1c)

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 003 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 004 Device 002 ID 0480:a007 control = auto, autosuspend_delay_ms =  2000 -- Toshiba America Info. Systems, Inc. External Disk USB 3.0 (usb-storage)
Bus 004 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 004 ID 058f:a014 control = auto, autosuspend_delay_ms =  2000 -- Alcor Micro Corp. Asus Integrated Webcam (uvcvideo)
Bus 001 Device 003 ID 0489:e03d control = auto, autosuspend_delay_ms =  2000 -- Foxconn / Hon Hai  (no driver)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

closed 2015-12-07 by avirajkhare00

linrunner, 2015-12-07:
You're welcome. 

Just for the sake of knowledge i would try to narrow down on which of the usb controllers is causing the trouble. Revert to RUNTIME_PM_ALL=1 and blacklist them one by one:

    RUNTIME_PM_BLACKLIST="00:1a.0"
    RUNTIME_PM_BLACKLIST="00:1d.0"
    RUNTIME_PM_BLACKLIST="03:00.0"

One line only, please. To blacklist more than one pci id, separate with blanks.