linrunner/TLP #179: intel_pstate: Turbo disabled by BIOS or unavailable on processor

olologin, 2016-01-04:
I have i3-2310m processor, and it doesn't supports Intel Turbo Boost, but even if i disable it in config:

    CPU_BOOST_ON_AC=0
    CPU_BOOST_ON_BAT=0

Something is trying to enable it (?) and i see this message in dmesg output.

    intel_pstate: Turbo disabled by BIOS or unavailable on processor

Can i somehow disable these attempts to enable Turbo Boost?

linrunner, 2016-01-05:
Hi, 

comment out both config lines (with '#') and TLP won't touch turbo boost at all.

olologin, 2016-01-05:
Hmm, but both are commented out already, and i don't see these lines in tlp stat output too.

   #CPU_BOOST_ON_AC=0
   #CPU_BOOST_ON_BAT=0

closed 2016-01-05 by olologin

linrunner, 2016-01-06:
That's right. If you comment out a config line, it does'nt show up in tlp-stat -c.

See the boost settings code here: https://github.com/linrunner/TLP/blob/master/tlp-functions.in#L649