linrunner/TLP #180: Bluetooth doesn't fully turn on

FabianInostroza, 2016-01-09:
Hi, 

I have a x230 running Debian Stretch and I have configured tlp to shutdown bluetooth on boot.

After boot `# rfkill list` shows
```
0: tpacpi_bluetooth_sw: Bluetooth
	Soft blocked: yes
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
```

When I want to turn it on I use: `# bluetooth on`
Running `bluetoothctl` show the device but I cannot use it.
Running `power on` in `bluetoothctl` shows "Failed to set power on: org.bluez.Error.Blocked".
Running `hcitool dev` shows nothing and
```
# rfkill list
0: tpacpi_bluetooth_sw: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no
4: hci0: Bluetooth
	Soft blocked: yes
	Hard blocked: no
```

By running `# rfkill unblock bluetooth` I can use the device.

Searching the net I found a Archlinux user with a similar problem and the cause was 
an udev rule that was blocking the device but I don't have rules related to bluetooth.

From the documentation I expect that the only necessary command to reenable bluetooth is
`# bluetooth on`, is that correct?.

linrunner, 2016-01-09:
Yes. `bluetooth on/off` just calls `rfkill unblock/block bluetooth`, so let's see if there's a difference.

Please try the following and show all outputs:

    rfkill unblock bluetooth
    bluetooth off
    rfkill list

    bluetooth on
    rfkill list

    rfkill block bluetooth
    rfkill list

    rfkill unblock bluetooth
    rfkill list

FabianInostroza, 2016-01-09:
Ok, here is the gist
https://gist.github.com/FabianInostroza/0eca03ec700c54497b51

linrunner, 2016-01-09:
`rfkill unblock bluetooth` shows exactly the same result as `bluetooth on` : 
hci0 stays "soft blocked".

So this is a rfkill related bug – or some weird kernel issue. I suggest you file a bug report against rfkill or Debians rfkill package.

As a workaround you may try to invoke `bluetooth on` two times.

ps. just for curiosity i'll try to reproduce it here on my X220 with Stretch

FabianInostroza, 2016-01-10:
Tried `bluetooth on` two times but it doesn't work.
Added the device to USB_BLACKLIST with no effect.
Also updated the firmware (it isn't available in any package) but doesn't make any difference.

One thing that I noticed is that immediately after running `bluetooth on`, `rfkill list` says that the device isn't soft blocked, once the firmware is loaded (dmesg) it changes to soft blocked.

Thanks, I will fill a bug against rfkill in Debian.

linrunner, 2016-01-10:
Try enabling hci0 explicitly with 

    rfkill unblock 4

Does the firmware load each time you unblock bluetooth? 

Maybe you should file your bug against the kernel instead of rfkill. When the proprietary firmware is the cause, it may be difficult to get a bug report accepted in Debian.

FabianInostroza, 2016-01-10:
unblocking hci0 doesn't trigger a firmware load, only running `rfkill unblock 0` (tpacpi_bluetooth_sw) or 
`rfkill unblock bluetooth`.
Running `rfkill block bluetooth` or `rfkill block 0` disconnects the device from the usb bus.

linrunner, 2016-01-10:
I meant if `rfkill unblock 4` turns bt on fully?

FabianInostroza, 2016-01-10:
No, is does nothing, at boot I don't have a rfkill with id 4 and after fully enabling bluetooth it goes only to 3.

linrunner, 2016-01-10:
Oh my -- then try it with `rfkill unblock 3` ;-)

FabianInostroza, 2016-01-10:
That was what I first tried (I edited my last comment, also edited because the max id shown is 2, sometimes 3).
Running `rfkill unblock 2` turns bluetooth fully on and doesn't load firmware.

linrunner, 2016-01-11:
OK, with that workaround i close here. Please keep me posted about your other bug report.

closed 2016-01-11 by linrunner

ericjding, 2016-12-01:
I know this issue is closed, and is probably not a TLP bug, but for what it's worth, I have experienced this same issue on my x220 running Linux Mint 18 (kernel 4.4.0-21), and it didn't occur in Linux Mint 17.3 (up to 3.19 series of kernels).  I am not in the position of doing a kernel bisection just to chase this down, but I'm curious if you can reproduce it on your x220, for example, using a live USB image of Ubuntu 16.04 or Linux Mint 18...

linrunner, 2016-12-01:
> kernel 4.4.0-21

You may want to enable kernel updates for your Mint installation. 4.4.0-51 is current. Way easier than a bisect ...

FabianInostroza, 2016-12-02:
I tried a live usb of ubuntu 16.04 and I was unable to reproduce the problem, however when I was suffering from this problem after some kernel updates it worked fine _sometimes_. With the current kernel in debian testing (4.8.7-1) it always works as expected.

ericjding, 2016-12-03:
Thanks for hanging in there on a bug that's not your own to fix!  I took your advice to try a few other kernels and configurations, with disappointing results.  Updating to 4.4.0-51 didn't make bluetooth work any better.  I also downloaded a number of live images to verify my observations:

Linux Mint 17.3 (kernel 3.19): rfkill block/unblock works as expected
Ubuntu 16.04.1 (kernel 4.4.0) : rfkill block/unblock sometimes works, but mostly requires a second unblock
Ubuntu 16.10 (kernel 4.8.0) : rfkill block/unblock sometimes works, but mostly requires a second unblock

Timing issue, maybe?  I don't know.  It's a minor issue in the big picture -- I needed to upgrade to get WiFi working more reliably, which is a showstopper in comparison -- but it's still a little disappointing.  I mainly add this comment for the record, so that maybe others searching online will stumble into it and have an epiphany. :-)

ericjding, 2016-12-09:
I'm just talking to myself for posterity's sake at this point, I guess, but why not?

I just reverted to Linux Mint 17.3 for other reasons, and decided to install the latest LTS kernel series from Ubuntu Xenial (4.4) to try and resolve my WiFi issues. Lo and behold, with the 4.4.0-53 kernel on a Linux Mint 17.3 system, I am not experiencing this issue with bluetooth failing to turn back on completely. So maybe it's not a kernel issue after all, but something else?

amaothree, 2018-11-12:
I resolve a similar issue on my archlinux just add `RESTORE_DEVICE_STATE_ON_STARTUP=1` in the `/etc/default/tlp` and reboot