linrunner/TLP #181: Airplane mode on resume and no wireless

rishadafc, 2016-01-21:
I've had this problem happen on my laptop (Asus ZenBook UX305FA) a few times now and I just recovered from it. If I leave my laptop lid shut overnight, it suspends and resumes correctly most of the time. Every once in a while (like just now), I open the laptop lid to find the suspend light indicator stabilize but the screen does not come on. When I reboot to my desktop environment (GNOME 3), it shows airplane mode enabled. When I disable it, my bluetooth appears but no wireless. 

I tried all some settings in my BIOS to trigger my wireless to enable - toggling Fast Boot, CSM mode, Network Stack - but nothing worked. I uninstalled tlp and magically my wireless came back on. I'm somewhat familiar with the tlp configuration file, but in this case I'm not sure if it's a bug in the program, the driver or if there is some setting that needs to be changed.

I would really appreciate your help with this because this is a long time recurring issue. I'm not sure what log file output to provide, so please let me know if that would be helpful.

linrunner, 2016-01-21:
Hi, 

TLP does not touch wifi on suspend/resume and has no configuration to do so. 

What happens on suspend/resume with TLP uninstalled?

rishadafc, 2016-01-21:
```
--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
DISK_IOSCHED="noop deadline"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0
DEVICES_TO_ENABLE_ON_STARTUP="wifi"

+++ System Info
System         = ASUSTeK COMPUTER INC. 1.0 UX305FA
BIOS           = UX305FA.211
Release        = "Arch Linux"
Kernel         = 4.3.3-3-ARCH #1 SMP PREEMPT Wed Jan 20 08:12:23 CET 2016 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-linux root=UUID=ee938606-eee7-49f1-93a1-314f2a916b79 rw cryptdevice=/dev/disk/by-uuid/84adf26d-1175-40e5-a735-39e2f02dc4a7:cryptAntergos resume=UUID=5461a6e4-fc1e-4205-89a0-a2ed4be4fa84 quiet
Init system    = systemd

+++ System Status
TLP power save = enabled
***Warning: tlp.service is not enabled -- invoke "systemctl enable tlp.service" to correct this!
***Warning: tlp-sleep.service is not enabled -- invoke "systemctl enable tlp-sleep.service" to correct this!
power source   = battery

+++ Processor
CPU Model      = Intel(R) Core(TM) M-5Y10c CPU @ 0.80GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   500000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2000000 [kHz]

[... 97 lines cut ...]
/sys/bus/pci/devices/0000:00:1f.2/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x0c0500, SMBus, no driver)
/sys/bus/pci/devices/0000:02:00.0/power/control = auto (0x028000, Network controller, iwlwifi)

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 003 ID 064e:9700 control = auto, autosuspend_delay_ms =  2000 -- Suyin Corp. Asus Integrated Webcam (uvcvideo)
Bus 001 Device 002 ID 8087:0a2a control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = ASUS Battery
/sys/class/power_supply/BAT0/cycle_count                    =     80
/sys/class/power_supply/BAT0/energy_full_design             =  45030 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  45235 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  12654 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   4218 [mW]
/sys/class/power_supply/BAT0/status                         = Discharging
```

linrunner, 2016-01-21:
I still need your answer on my above question:

> What happens on suspend/resume with TLP uninstalled?

rishadafc, 2016-01-21:
Oh yes, sorry about that. Suspend/resume works normally with TLP uninstalled and wireless works fine. It *usually* works when TLP is installed as well, but sometimes when I resume (like today) the screen doesn't turn on and there is no response.

linrunner, 2016-01-21:
First of all your TLP installation is incomplete

> ***Warning: tlp.service is not enabled -- invoke "systemctl enable tlp.service" to correct this!
> ***Warning: tlp-sleep.service is not enabled -- invoke "systemctl enable tlp-sleep.service" to correct this!

TLP is neither invoked upon system start nor upon suspend/resume. Correct this with:

    systemctl enable tlp.service
    systemctl enable tlp-sleep.service
    systemctl mask systemd-rfkill@.service

and check if the symptoms change.

If not, next try is to disable Runtime PM on battery

    RUNTIME_PM_ON_AC=on
    RUNTIME_PM_ON_BAT=on

rishadafc, 2016-01-21:
Ok, thanks. I enabled the services and I will reply to this issue if there are any problems. I will close this issue for now.

There was no warning to run this command: `systemctl mask systemd-rfkill@.service`. Should that be something that tlp warns by default in addition to the other two services? The Arch wiki mentioned disabling rfkill which didn't seem to work for me. Does this command do the same thing?

closed 2016-01-21 by rishadafc

linrunner, 2016-01-21:
> There was no warning to run this command: systemctl mask systemd-rfkill@.service

Feature in v0.9.

rishadafc, 2016-01-30:
I had this issue again right now and I tried everything. Finally I disabled Runtime PM on battery and it worked on reboot. Do you have some idea of what this means?

linrunner, 2016-01-30:
Buggy kernel driver (as always). 

Instead of `RUNTIME_PM_ON_BAT=on` you may blacklist PCIe devices one by one in `RUNTIME_PM_BLACKLIST=` and re-check to isolate the offending device. Another option to test would be `RUNTIME_PM_ALL=0`.

[1] http://linrunner.de/en/tlp/docs/tlp-configuration.html#runtimepm