linrunner/TLP #185: When booting on battery, TLP starts as if on AC

thecompu, 2016-02-02:
ASUS ZenBook UX305 running Linux Mint 17.3 Cinnamon. 

I noticed that whenever I rebooted and manually ran tlp after bootup, my battery charge remaining time would increase by about 2-4 hours. This led me down a path that determined that even though the default configuration tlp was to run on powersave energy profile on battery, it was in performance on bootup. When I ran sudo tlp start after bootup, tlp would switch to powersave. 

Here is the output of tlp-stat after bootup:
https://gist.github.com/thecompu/da6c0be37099e98572d2

I've also noticed that tlp doesn't always switch the energy profile automatically when switching power-source. 

Here's udevadm going from battery to AC:
https://gist.github.com/thecompu/d89cdd746cd5c6361a9f

Here's udevadm going from AC to battery:
https://gist.github.com/thecompu/54516b8150a2e8f1a39a

Here's tlp-stat immediately after pulling AC cord (note we've changed to powersave):
https://gist.github.com/thecompu/c632c026bfdcba1b30f4

And now tlp-stat after plugging AC back in (note we're still on powersave):
https://gist.github.com/thecompu/eef44a27a0696a570656

It's important to note that the UX305 battery will not charge unless the battery percentage is below 95%. As I am testing this, the battery is at 96%. I will test switching from battery to AC when my battery drops below 95% and I will report back if there's any difference.

Lastly, please note that I cannot run a tlp-stat --trace despite following the instructions. The most I've been able to do is get it to output a banner but then it stops displaying anything.

linrunner, 2016-02-02:
Hi, thanks for your outputs.

Good news first: your assumption that TLP does not execute upon start and change of power source is wrong. Just compare these excerpts from your tlp-stat outputs 

**Battery**

    +++ Storage Devices
    /dev/sda:
              APM Level = 128

    +++ SATA Aggressive Link Power Management
    /sys/class/scsi_host/host0/link_power_management_policy  = min_power

    +++ Wireless
    wlan0(iwlwifi)      : connected, power management = on

    +++ Audio
    /sys/module/snd_hda_intel/parameters/power_save            = 1

    +++ Runtime Power Management
    /sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, no driver)`

**AC**

    +++ Storage Devices
    /dev/sda:
              APM Level = 254

    +++ SATA Aggressive Link Power Management
    /sys/class/scsi_host/host0/link_power_management_policy  = max_performance

    +++ Wireless
    wlan0(iwlwifi)      : connected, power management = off

    +++ Audio
    /sys/module/snd_hda_intel/parameters/power_save            = 0

    +++ Runtime Power Management
    /sys/bus/pci/devices/0000:00:00.0/power/control = on   (0x060000, Host bridge, no driver)

Your remaining issue is the  **CPU scaling governor:** 
The reason it doesn't change as you expect is quite simple: TLP's  default configuration doesn't touch it because most distros will apply a reasonable default "ondemand" or "powersave" upon system startup themselves.

If you'd like to have TLP change the governor you have to explicitly enable the [associated lines](http://linrunner.de/en/tlp/docs/tlp-configuration.html#scaling) in the config by uncommenting them i.e. remove the leading '#':

    CPU_SCALING_GOVERNOR_ON_AC=powersave  # or performance 
    CPU_SCALING_GOVERNOR_ON_BAT=powersave

And yes, i do recommend *not to* use "performance" but "powersave" on AC.

*Important:* you also need to disable your distro's governor default setting --> [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#scaling) 

**Trace:** as your output shows didn't insert the necessary line in the config --> [Trace Mode](http://linrunner.de/en/tlp/docs/tlp-configuration.html#tracemode)

But we don't need a trace because there is no issue anymore :-).

thecompu, 2016-02-02:
Wonderful.

I take it I need to alter /etc/default/tlp to uncomment the the scaling lines and to add the trace configuration, correct?

Seeing as I am a little new to running Linux full time, I am uncertain which command to run to disable the distro's default governer settings. Mint is based on Ubuntu. So, perhaps that.

closed 2016-02-22 by linrunner