linrunner/TLP #186: Linux Mint 17.3: Freeze after boot

f380cedric, 2016-02-16:
Hi,
I have installed tlp and tlp-rw but my pc freezes ~3 min after boot :/
```{r, engine='shell', count_lines}
--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0

+++ System Info
System         = Hewlett-Packard 087E1000003B5900000320100 HP Pavilion Sleekbook 15
BIOS           = F.1B
Release        = Linux Mint 17.3 Rosa
Kernel         = 3.19.0-32-generic #37~14.04.1-Ubuntu SMP Thu Oct 22 09:41:40 UTC 2015 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-3.19.0-32-generic root=UUID=e177d741-77c9-4501-95dd-fbaa1906fc67 ro quiet splash vt.handoff=7
Init system    = upstart

+++ System Status
TLP power save = enabled
power source   = battery

+++ Processor
CPU Model      = AMD A8-4555M APU with Radeon(tm) HD Graphics

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = ondemand
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =  1100000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  1600000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies = 1600000 1400000 1300000 1100000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = ondemand
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =  1100000 [kHz]
[... 111 lines cut ...]
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 004 Device 002 ID 04f2:b35f control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 004 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 006 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 003 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 005 Device 001 ID 1d6b:0001 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 1.1 root hub (hub)
Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = Hewlett-Packard
/sys/class/power_supply/BAT0/model_name                     = Primary
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   1632 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   1632 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =    320 [mAh]
/sys/class/power_supply/BAT0/current_now                    =   1057 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

```
If I go in tty1 just before the freeze (totally not responding), terminal prints (continuously) something like `radeon 0000:00:00.0: ring 0 stalled for more than 10008msec` with the timer increasing, and after a short time, it freezes.
An idea ?
Thx

linrunner, 2016-02-16:
Hi,

please uninstall TLP, power off and and check if the symptom disappears.

f380cedric, 2016-02-16:
Yes, I already did it, no symptom. It is the TLP installation which makes that...

f380cedric, 2016-02-18:
Any ideas how I can solve this ? ^^

linrunner, 2016-02-19:
Does it happen on battery only? Then change configuration to

    RUNTIME_PM_ON_BAT=on

apply the setting with

    sudo tlp start

and check agian.

f380cedric, 2016-02-20:
Hello,
No issue in AC mode, it works. On battery, the change `RUNTIME_PM_ON_BAT=on` changes nothing, still the same issue :(

f380cedric, 2016-02-20:
Questions: why comment the AC if it works perfectly ? And what about the `RUNTIME_PM_ALL=1`, must I set it to 0 ?

linrunner, 2016-02-20:
With commenting AC and BAT we make sure Runtime PM isn't touched at all (some devices are that touchy). RUNTIME_PM_ALL=0 just skips some device classes.

Next on the list of settings differing between AC and BAT is radeon dpm. Try

    RADEON_DPM_STATE_ON_BAT=performance

f380cedric, 2016-02-20:
Question before this modification, must I uncomment PM lines or not ?

linrunner, 2016-02-21:
Doesn't matter at this point. Go ahead.

f380cedric, 2016-02-22:
Ok, it seems to work with `RADEON_DPM_STATE_ON_BAT=performance`, but why ?

linrunner, 2016-02-22:
Not all kernel drivers implement power saving properly causing the user trouble when TLP activates it. Look for issues marked with "kernel quirk". There are quite a few. 

You may revert to

    RUNTIME_PM_ON_AC=on
    RUNTIME_PM_ON_BAT=auto

now.

closed 2016-02-22 by linrunner