linrunner/TLP #188: USB charging function dont work with tlp enabled

SanKro, 2016-03-15:
as long as tlp is enabled i cant recharge my android phone via USB Port on my computer.

for 1-2 secounds it starts, but then it stops charging. Seems like its bcause of the usb autosuspend function. 

there are 2 workarounds: open a folder on you phone via filemanager on linux so that it has to hold the connection or disable tlp/autosuspend.

it would be nice if tlp would ignore devices like this from usb autosuspend

dywisor, 2016-03-15:
You can exclude individual devices from USB autosuspend, see [FAQ#usb](http://linrunner.de/en/tlp/docs/tlp-faq.html#usb).

linrunner, 2016-03-15:
Hi, good idea. Unfortunately i don't have the slightest idea how to determine that a device is a smartphone.

Feel free to make concrete suggestions how to implement the feature – or even better: send pull requests :-)

prantlf, 2016-04-10:
Actually, I'd like not to charge my smartphone, when it is connected to the USB port of my notebook.

I use the smartphone to provide the internet connection over USB (tethering) from time to time.  I need to save the battery in the notebook as much as possible, because I'm travelling for a long time.  If you wanted to change the functionality to allow the phone charging, make it configurable, please, so that the people could turn the charging off, if they need it.  Thanks!

SanKro, 2016-04-10:
@linrunner yeah seems like this is some problem..
however in KDE Plasma, there is a widget called "Devicenotifier". it can simply mount/open new attached devices like smartphones, HDD/SDD, Mediaplayers und blah.

the interesting thing is that it identifies a smartphone correctly as smartphone. So there has to be a way to do this..

@prantlf if both, smartphone and PC have at least Bluetooth 4.x you should use this because its more energyfriendly than USB. i travel every monday 5h with a train (parents -> home). with bluetooth i can use internet all the time and still have battery left. with usb my phone battery is on 5% after only 3-4h

linrunner, 2016-12-09:
Feature is now [implemented](https://github.com/linrunner/TLP/commit/4f39d7f74332aeca08d42d6fa9641d9153c5dc91).

linrunner, 2017-05-25:
Released with 1.0.

closed 2017-05-25 by linrunner