linrunner/TLP #190: brightness control

acornejo, 2016-03-30:
a big factor in power consumption, especially in laptops, is screen brightness.

it would be fairly easy to add this feature to TLP, so that the user can specify default values for brightness when in AC/battery and have TLP apply them when appropriate. I would be happy to contribute a pull-request with this feature if there are changes of having it included in TLP.

Thoughts?

FadeMind, 2016-03-30:
In KDE Plasma this is controled via `plasma-workspace` and `powerdevil`  package and settings are stored in `powermanagementprofilesrc` file, so this feautre is manage via current used DE.

acornejo, 2016-03-30:
@fademind, but in gnome shell or i3 this is not available.

linrunner, 2016-03-30:
Hi,

such a feature would inevitably conflict with several desktop environments implementing their own kind brightness control. From a user's perspective it would always be TLP that "doesn't work" leading to many support questions or issues. As an additional complication these issues would be difficult to explain and/or solve. 

Conclusion: no, thanks.

Therefore I decided long ago not to have brightness control in TLP --> see [FAQ](http://linrunner.de/en/tlp/docs/tlp-faq.html#featuresexcluded).

acornejo, 2016-03-30:
I think these concerns would be addressed by having this feature disabled by default, so users have to opt in and read a comment before enabling it; but I understand and respect your decision.

FWIW this feature has been removed from the latest gnome (I don't know why, it used to work rather well), but it is certainly available in other DE or other gnome versions.

linrunner, 2016-03-30:
Reopened to show up on the issues page :-)

linrunner, 2016-03-30:
@acornejo: no way.

closed 2019-01-05 by linrunner