linrunner/TLP #192: wifi pm issue - not sure if to report, but now I do

pressure679, 2016-04-10:
Hi.

The main issue is I set line 131-132's value to 1 like this (uncommented):
# WIFI_PWR_ON_AC=1
# WIFI_PWR_ON_BAT=1

But somehow the wifi power management was still on (overriding settings in /etc/modprobe.d/<wifi module>.conf). Now the lines are commented out and it works.
settings were "options rtl8192se ips=N fwlps=N swlps=N debug=1 disable_watchdog=Y swenc=Y".
/etc/modprobe.d/<wifi module>.conf is not needed when commenting out line 131-132.

Even by setting line 131-132's value to 1 caused the error.
Even if I blacklisted the wifi module's address received from "lspci | grep Wireless" at line 164 the error still happened.

I have Runtime Power Management, line 155-164 with no issue:
# Runtime Power Management for PCI(e) bus devices: on=disable, auto=enable
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto

# Runtime PM for *all* PCI(e) bus devices, except blacklisted ones:
#   0=disable, 1=enable
RUNTIME_PM_ALL=1

# Exclude PCI(e) device adresses the following list from Runtime PM
# (separate with spaces). Use lspci to get the adresses (1st column).
RUNTIME_PM_BLACKLIST="04:00.0"

The issue is also addressed here: https://github.com/lwfinger/rtlwifi_new/issues/112

linrunner, 2016-04-10:
Hi, please be specific in your issue title (edited)!

TLP uses the following commands to switch wifi power save:

    iw dev wlan0 set power_save on
    iw dev wlan0 set power_save off

When iw is not installed 
    
    iwconfig wlan0 power on
    iwconfig wlan0 power off

is used (substitute wlan0 with your interface if necessary). 

You should verify that above off commands cannot disable power save for your wifi card – then file a bug report against the kernel driver not implementing the power save off switching correctly.

Conclusion: TLP cannot help here, the kernel driver must be fixed.

Remarks:
* Of course you have to disable wifi pm in TLP's config when using module options; otherwise TLP will overwrite the setting upon first invocation
* Runtime pm has no relationship with wifi pm – provided the wifi driver is implemented properly

pressure679, 2016-04-16:
I tried that, it did not help.
I think several errors occured which I reinstalled the OS to undo the bug.
In the end when I used the command "e4defrag /" caused the unstable connection, so I only defrag $HOME directory now.

closed 2016-04-16 by linrunner