linrunner/TLP #193: tlp.service wants Network Manager

SammysHP, 2016-04-10:
The systemd unit wants NetworkManager.service in https://github.com/linrunner/TLP/blob/master/tlp.service.in#L9, but Network Manager is not a dependency of tlp, but only of tlp-rdw (which is optional). This is a problem if one has installed Network Manger as a dependency for something else, but does not use it. In this case Network Manger is started, but conflicts with another manager.

I suggest to create a static tlp-rdw.service in the tlp-rdw package which is wanted by tlp.service.

Another possibility for the user would be to mask NetworkManager.service (or do not install it at all if it is not in use), but IMHO it is wrong that tlp wants Network Manger.

linrunner, 2016-04-12:
If it just could be that easy ...

For tlp-rdw NM is a must. tlp itself uses NM when it is installed and *likes* to have it started before doing radio switching. So how to implement that?

SammysHP, 2016-04-12:
If tlp itself uses NM when it is available then it might be better to keep the current implementation. Users who don't want NM to start should mask it anyway.

ArchangeGabriel, 2016-05-01:
Yes, but I might not want to install NM at all. And currently, it’s not a dependency of tlp in ArchLinux btw. Thing is, I don’t use enable or use the tlp.service at all (since it works perfectly without it), so…

linrunner, 2016-05-02:
@ArchangeGabriel: enabling tlp.service is not "nice to have" but a prerequisite for TLP working properly – and receiving support here. It works fine even without NM installed.

Most important purpose is to apply power save settings upon system startup (see [code](https://github.com/linrunner/TLP/blob/04ae361129dc57bc9d56959f483994d7db7055ea/tlp#L92)) and i take the freedom to include new features there without notice.

ArchangeGabriel, 2016-05-02:
I saw that y-day while looking at change already in for 0.9. Seems like I misunderstood something about systemd Wants. Enabled.

Gigadoc2, 2018-07-10:
> For tlp-rdw NM is a must. tlp itself uses NM when it is installed and likes to have it started before doing radio switching. So how to implement that?

@linrunner Wouldn't removing `NetworkManager.service` from `Wants=` but keeping it in `After=` achieve exactly that?

If the user uses NetworkManager they either have it enabled directly or something is depending on it; in both cases `After=` alone will delay the start of TLP until after NetworkManager is running.

If NetworkManager is not enabled it is not dbus-activatable either, so AFAIK the only way to use NetworkManager if it is not enabled or a dependency is to manually run `systemctl start NetworkManager` at some point. And if a user - for some strange reason - starts their NetworkManager by hand, they probably don't want TLP to start it for them either (and they should be aware that manually starting services will defeat any ordering done with `After=` or `Before=`).

So, apologies for being late (I just now discovered this behaviour), but I think this decision should be reevaluated, as the current behaviour does not make sense in any scenario I can think of.

I'm under the impression that use-cases like this are exactly why `Wants=` and `After=` are separated (along with requiring something to be started, but afterwards).

linrunner, 2019-01-05:
Done. Better late than never ... :-)

@Gigadoc2 : thanks for your helpful explanation.

closed 2019-01-05 by linrunner

linrunner, 2019-03-11:
Released with TLP 1.2.