linrunner/TLP #194: AE_NOT_FOUND messages on Lenovo B51 series

xypista, 2016-04-23:
`acpi_call: Cannot get handle: Error: AE_NOT_FOUND` 
I'm using Arch Linux with default kernel 4.5.1 and TLP 0.8 and I get a bunch of those messages on boot.
[dmesg](https://gist.github.com/xypista/cf3a839bb554b6643386397a6381cfda) | [systemd journal](https://gist.github.com/xypista/f8f3fce07daa288c7b0f395d160f9fc7)

Laptop model: Lenovo B51-80, not ThinkPad

linrunner, 2016-04-23:
Battery functions do not work on non-ThinkPads.

closed 2016-04-23 by linrunner

ActualizeInMaterial, 2016-06-07:
Btw, I am getting this error 3 times on ThinkPad T400 (where battery functions do work, apparently) during `tlp-stat`:

```
--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=600
DISK_IDLE_SECS_ON_BAT=60
MAX_LOST_WORK_SECS_ON_AC=600
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
DISK_IOSCHED="noop noop"
SATA_LINKPWR_ON_AC=min_power
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=powersave
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=low
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=battery
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=low
RADEON_DPM_PERF_LEVEL_ON_BAT=low
WIFI_PWR_ON_AC=5
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=1
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=1
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=auto
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=0
USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN=0
RESTORE_DEVICE_STATE_ON_STARTUP=0
START_CHARGE_THRESH_BAT0=40
STOP_CHARGE_THRESH_BAT0=50
START_CHARGE_THRESH_BAT1=40
STOP_CHARGE_THRESH_BAT1=50

+++ System Info
System         = LENOVO ThinkPad T400 64741AG
BIOS           = 7UET94WW (3.24 )
Release        = "Manjaro Linux"
Kernel         = 4.5.6-1-MANJARO #1 SMP PREEMPT Thu Jun 2 17:33:22 UTC 2016 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.5-x86_64 root=UUID=2888d969-3bfb-4f5e-9144-2e0261dd457b rw ipv6.disable=1 pnp.debug=1 loglevel=9 log_buf_len=10M printk.always_kmsg_dump=y printk.time=y mminit_loglevel=0 memory_corruption_check=1 fbcon=scrollback:4096k fbcon=font:ProFont6x11 apic=debug dynamic_debug.verbose=1 "dyndbg=file arch/x86/kernel/apic/* +pflmt ; file drivers/video/* +pflmt ; file drivers/input/* -pflmt ; file drivers/acpi/* -pflmt" rd.debug rd.udev.debug rd.memdebug=3 console=tty1 earlyprintk=vga slub_debug=U pax_sanitize_slab=full noefi resume=UUID=d29dfcdd-e381-43e3-b7bf-676aa0ed7bf0
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = AC

+++ Processor
CPU Model      = Intel(R) Core(TM)2 Duo CPU     T9400  @ 2.53GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = acpi-cpufreq
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = ondemand
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2534000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies = 2534000 2533000 1600000 800000 [kHz]

[... 166 lines cut ...]
license:        GPL
depends:        
vermagic:       4.5.6-1-MANJARO SMP preempt mod_unload modversions 
[root@mighty4 ~]# modinfo tp_smapi
filename:       /lib/modules/4.5.6-1-MANJARO/extramodules/tp_smapi.ko.gz
license:        GPL
version:        0.41
description:    ThinkPad SMAPI Support
author:         Shem Multinymous
srcversion:     B6841670771B2FF5222BFFD
depends:        thinkpad_ec
vermagic:       4.5.6-1-MANJARO SMP preempt mod_unload modversions 
parm:           debug:Debug level (0=off, 1=on) (int)
[root@mighty4 ~]# modinfo thinkpad_ec
filename:       /lib/modules/4.5.6-1-MANJARO/extramodules/thinkpad_ec.ko.gz
license:        GPL
version:        0.41
description:    ThinkPad embedded controller hardware access
author:         Shem Multinymous
srcversion:     3CC48C14EE079F4A4872469
depends:        
vermagic:       4.5.6-1-MANJARO SMP preempt mod_unload modversions 
parm:           force_io:Force IO even if region already reserved (0=off, 1=on) (bool)

```