linrunner/TLP #196: [Feature Request] AHCI Runtime PM support

ArchangeGabriel, 2016-05-01:
Hi,

Linux 4.6 is around the corner, and come with a great news for PM: https://www.phoronix.com/scan.php?page=news_item&px=Linux-4.6-AHCI-Runtime-PM.

TLP already supports SATA Runtime PM, so I think this shouldn’t be hard.

Thanks!

linrunner, 2016-05-01:
Hi,

in fact it is already in  the works for 0.9. Unfortunately it's causing system lockups on *every* invocation on my machine (ThinkPad T450s w/ Samsung 850 Evo, Kernel 4.6rc5). So it won't be enabled by default and will have a big EXPERIMENTAL WARNING sticker attached.

I would be grateful if you could test as soon as i commit :-).

ArchangeGabriel, 2016-05-01:
Sure, will let you know!

linrunner, 2016-05-03:
Please test the feature WITH CARE on a non-production system: potential system freezes/data loss!

Arch Linux: packages tlp-git and tlp-rdw-git tracking the master branch are available via AUR.

linrunner, 2016-08-19:
Released with 0.9.

closed 2016-08-19 by linrunner