linrunner/TLP #197: i915 parameters

ArchangeGabriel, 2016-05-01:
Hi,

It seems that more and more i915 parameters are being enforced on by default. Thus, I have this on my system (linux 4.5, but I think this started earlier):
```
/sys/module/i915/parameters/powersave        = (not available)
/sys/module/i915/parameters/lvds_downclock   = (not available)
```

Similarly, `enable_fbc` and `enable_psr` are getting enabled by default in 4.6 (but probably not enforced, 4.6 will land soon anyway if you’re like me and don’t want to dig into sources to know about it).

Also, new ones made their apparition, and you might want to track `/sys/module/i915/parameters/enable_dc` and `/sys/module/i915/parameters/enable_psr`.

I know all this is just about information, not settings, but if you want to keep this thing, rather keep it up-to-date. ;)

linrunner, 2016-05-03:
Please test.

FadeMind, 2016-05-03:
@linrunner https://pastebin.com/raw/advtuQ01
NOTE: I removed S.M.A.R.T. and details about disk here.

Great improvements in TLP. thanks.

linrunner, 2016-08-19:
Released with 0.9.

closed 2016-08-19 by linrunner