linrunner/TLP #198: Disabling WiFi on LAN connect does not work

zunder, 2016-05-06:
Looks like TLP does not disable the wifi device on lan connect, though it is set up through config:
```
DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"
DEVICES_TO_DISABLE_ON_DOCK="wifi"
```
Is that a current known issue? I am using Ubuntu Server 16.04 and `tlp-stat` yields following:
```
--- TLP 0.8 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=powersave
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC=max_performance
SATA_LINKPWR_ON_BAT=min_power
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=1
WIFI_PWR_ON_BAT=5
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_ALL=1
RUNTIME_PM_DRIVER_BLACKLIST="radeon nouveau"
USB_AUTOSUSPEND=1
USB_BLACKLIST_WWAN=1
RESTORE_DEVICE_STATE_ON_STARTUP=0
DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"
DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi wwan"
DEVICES_TO_DISABLE_ON_DOCK="wifi"
DEVICES_TO_ENABLE_ON_UNDOCK="wifi"

+++ System Info
System         = LENOVO ThinkPad S1 Yoga 20C0S0AX00
BIOS           = GQET48WW (1.28 )
Release        = Ubuntu 16.04 LTS
Kernel         = 4.4.0-21-generic #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.4.0-21-generic.efi.signed root=UUID=31ee377b-fec2-45cc-9b62-dff82d801882 ro
Init system    = systemd

+++ System Status
TLP power save = enabled
power source   = AC

+++ Processor
CPU Model      = Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
[... 149 lines cut ...]
Bus 002 Device 007 ID 1a40:0101 control = auto, autosuspend_delay_ms =     0 -- Terminus Technology Inc. Hub (hub)
Bus 002 Device 008 ID 17ef:304e control = on,   autosuspend_delay_ms =  2000 -- Lenovo  (snd-usb-audio, usbhid)
Bus 002 Device 005 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 003 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 002 ID 046d:c526 control = on,   autosuspend_delay_ms =  2000 -- Logitech, Inc. Nano Receiver (usbhid)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ ThinkPad Extended Battery Functions
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = inactive (kernel module 'acpi_call' not installed)

+++ Battery Status
/sys/class/power_supply/BAT0/manufacturer                   = SONY
/sys/class/power_supply/BAT0/model_name                     = 45N1705
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  47060 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  42550 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  42550 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

+++ Suggestions
* Install tp-smapi kernel modules for ThinkPad advanced battery functions

```

linrunner, 2016-05-06:
Package **tlp-rdw** is installed?

zunder, 2016-05-06:
Yes it is, `apt install tlp-rdw` yields following:

```
user@ubuntu:~$ sudo apt install tlp-rdw
Reading package lists... Done
Building dependency tree       
Reading state information... Done
tlp-rdw is already the newest version (0.8-1).
tlp-rdw set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 21 not upgraded.
```

linrunner, 2016-05-06:
I suspect the Yoga has ethernet via usb3 in the OneLink dock, most likely this device:
> Bus 003 Device 007 ID 17ef:304f control = auto, autosuspend_delay_ms =  2000 -- Lenovo  (r8152)

Please show (with lan cable connected):

   ifconfig
   nmcli dev

ps. could you do me a favor (for TLP development) and check out which usb devices belong to the OneLink dock natively i.e. remove all devices from the dock and compare the output of

    tlp-stat -u

docked vs. undocked? Which model is your dock exactly?

zunder, 2016-05-06:
That is very true. I am using a **Thinkpad OneLink Pro Dock, ModelNr: DU9033S1** which is connected via usb3. (actually it uses Lenovos own port technology, but I guess you can brake it down to usb3 + power supply)

Here `ifconfig`:

```
user@ubuntu:~$ ifconfig
enx0050b66cf1b4 Link encap:Ethernet  HWaddr XX:XX:XX:XX:XX:XX  
          inet addr:192.168.178.82  Bcast:192.168.178.255  Mask:255.255.255.0
          inet6 addr: 2a02:2028:635:6b01:bca3:d89f:ea3d:7d3a/64 Scope:Global
          inet6 addr: 2a02:2028:635:6b01:250:b6ff:fe6c:f1b4/64 Scope:Global
          inet6 addr: fd00::bca3:d89f:ea3d:7d3a/64 Scope:Global
          inet6 addr: fd00::250:b6ff:fe6c:f1b4/64 Scope:Global
          inet6 addr: fe80::250:b6ff:fe6c:f1b4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:9920 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5270 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:9262522 (9.2 MB)  TX bytes:1226648 (1.2 MB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:320 errors:0 dropped:0 overruns:0 frame:0
          TX packets:320 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:23680 (23.6 KB)  TX bytes:23680 (23.6 KB)

wlp4s0    Link encap:Ethernet  HWaddr XX:XX:XX:XX:XX:XX  
          inet addr:192.168.178.93  Bcast:192.168.178.255  Mask:255.255.255.0
          inet6 addr: fe80::ab3d:fb94:920a:4b80/64 Scope:Link
          inet6 addr: fd00::48ef:c7ef:e8dc:bdd7/64 Scope:Global
          inet6 addr: 2a02:2028:635:6b01:93:27b:65a5:5a0d/64 Scope:Global
          inet6 addr: 2a02:2028:635:6b01:c83:561c:94aa:8017/64 Scope:Global
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6456 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3433 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5920174 (5.9 MB)  TX bytes:976871 (976.8 KB)
```

`nmcli dev` yields:

```
DEVICE           TYPE      STATE      CONNECTION 
wlp4s0           wifi      connected  myhomenetwork
enx0050b66cf1b4  ethernet  unmanaged  --         
lo               loopback  unmanaged  --         
```

`tlp-stat -u` **docked** without any other usb devices (but DP, DVI, Ethernet still connected) yields:
```
user@ubuntu:~$ tlp-stat -u
--- TLP 0.8 --------------------------------------------

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 001 Device 004 ID 056a:00ec control = on,   autosuspend_delay_ms =  2000 -- Wacom Co., Ltd  (usbhid)
Bus 001 Device 003 ID 0483:91d1 control = on,   autosuspend_delay_ms =  2000 -- STMicroelectronics Sensor Hub (usbhid)
Bus 001 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 004 ID 17ef:304f control = auto, autosuspend_delay_ms =  2000 -- Lenovo  (r8152)
Bus 003 Device 003 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 002 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 009 ID 04f2:b39f control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
[... 6 lines cut ...]
Bus 002 Device 002 ID 046d:c526 control = on,   autosuspend_delay_ms =  2000 -- Logitech, Inc. Nano Receiver (usbhid)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```
`tlp-stat -u` **undocked** yields:
```
user@ubuntu:~$ tlp-stat -u
--- TLP 0.8 --------------------------------------------

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 001 Device 004 ID 056a:00ec control = on,   autosuspend_delay_ms =  2000 -- Wacom Co., Ltd  (usbhid)
Bus 001 Device 003 ID 0483:91d1 control = on,   autosuspend_delay_ms =  2000 -- STMicroelectronics Sensor Hub (usbhid)
Bus 001 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 009 ID 04f2:b39f control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 002 Device 006 ID 04f3:0254 control = on,   autosuspend_delay_ms =  2000 -- Elan Microelectronics Corp.  (usbhid)
Bus 002 Device 004 ID 8087:07dc control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 002 Device 002 ID 046d:c526 control = on,   autosuspend_delay_ms =  2000 -- Logitech, Inc. Nano Receiver (usbhid)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```

linrunner, 2016-05-06:
Which test case(s) did you try:

1. plug/unplug the cable or
2. dock/undock the ThinkPad?

Case 2 is likely not to work with 0.8 because after undocking the usb device is gone and the type cannot be detected. There is a however a [contribution](https://github.com/linrunner/TLP/commit/9f5f6866a3485a10a63344ea3e5fed6eb44f98d7) for 0.9 addressing this.

If you didn't try 1. please do so now.

zunder, 2016-05-06:
Do you mean the Ethernet cable in case 1.? If so, yes I did only try case 2 - I did remove all usb devices (except Logitech's dongle which was connected to laptop directly: `Bus 002 Device 002 ID 046d:c526 control = on,   autosuspend_delay_ms =  2000 -- Logitech, Inc. Nano Receiver (usbhid)`, sorry for that) and ran `tlp-stat -u` with dock pluged in. Then I just unplugged the dock and did `tlp-stat -u` again.

So here case 1. If I unplug all usb devices from Dock and Laptop, except for the Dock itself and additionally remove the Ethernet cable from Dock, I get:

```
user@ubuntu:~$ tlp-stat -u
--- TLP 0.8 --------------------------------------------

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 001 Device 004 ID 056a:00ec control = on,   autosuspend_delay_ms =  2000 -- Wacom Co., Ltd  (usbhid)
Bus 001 Device 003 ID 0483:91d1 control = on,   autosuspend_delay_ms =  2000 -- STMicroelectronics Sensor Hub (usbhid)
Bus 001 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 011 ID 17ef:304f control = auto, autosuspend_delay_ms =  2000 -- Lenovo  (r8152)
Bus 003 Device 010 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 009 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 009 ID 04f2:b39f control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 002 Device 006 ID 04f3:0254 control = on,   autosuspend_delay_ms =  2000 -- Elan Microelectronics Corp.  (usbhid)
Bus 002 Device 004 ID 8087:07dc control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 002 Device 015 ID 1a40:0101 control = auto, autosuspend_delay_ms =     0 -- Terminus Technology Inc. Hub (hub)
Bus 002 Device 019 ID 17ef:304e control = on,   autosuspend_delay_ms =  2000 -- Lenovo  (snd-usb-audio, usbhid)
Bus 002 Device 018 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 014 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```
Plugging the Ethernet cable back in yields:
```
alex@ubuntu:~$ tlp-stat -u
--- TLP 0.8 --------------------------------------------

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 001 Device 004 ID 056a:00ec control = on,   autosuspend_delay_ms =  2000 -- Wacom Co., Ltd  (usbhid)
Bus 001 Device 003 ID 0483:91d1 control = on,   autosuspend_delay_ms =  2000 -- STMicroelectronics Sensor Hub (usbhid)
Bus 001 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 011 ID 17ef:304f control = auto, autosuspend_delay_ms =  2000 -- Lenovo  (r8152)
Bus 003 Device 010 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 009 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 009 ID 04f2:b39f control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 002 Device 006 ID 04f3:0254 control = on,   autosuspend_delay_ms =  2000 -- Elan Microelectronics Corp.  (usbhid)
Bus 002 Device 004 ID 8087:07dc control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 002 Device 015 ID 1a40:0101 control = auto, autosuspend_delay_ms =     0 -- Terminus Technology Inc. Hub (hub)
Bus 002 Device 019 ID 17ef:304e control = on,   autosuspend_delay_ms =  2000 -- Lenovo  (snd-usb-audio, usbhid)
Bus 002 Device 018 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 014 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```
There seems to be no difference in here, so I am not sure if it helps you in any way.

zunder, 2016-05-06:
Do you mean the Ethernet cable in case 1.? If so, yes I did only try case 2 - I did remove all usb devices (except Logitech's dongle which was connected to laptop directly: `Bus 002 Device 002 ID 046d:c526 control = on,   autosuspend_delay_ms =  2000 -- Logitech, Inc. Nano Receiver (usbhid)`, sorry for that) and ran `tlp-stat -u` with dock pluged in. Then I just unplugged the dock and did `tlp-stat -u` again.

So here case 1. If I unplug all usb devices from Dock and Laptop, except for the Dock itself and additionally remove the Ethernet cable from Dock, I get:

```
user@ubuntu:~$ tlp-stat -u
--- TLP 0.8 --------------------------------------------

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 001 Device 004 ID 056a:00ec control = on,   autosuspend_delay_ms =  2000 -- Wacom Co., Ltd  (usbhid)
Bus 001 Device 003 ID 0483:91d1 control = on,   autosuspend_delay_ms =  2000 -- STMicroelectronics Sensor Hub (usbhid)
Bus 001 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 011 ID 17ef:304f control = auto, autosuspend_delay_ms =  2000 -- Lenovo  (r8152)
Bus 003 Device 010 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 009 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 009 ID 04f2:b39f control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 002 Device 006 ID 04f3:0254 control = on,   autosuspend_delay_ms =  2000 -- Elan Microelectronics Corp.  (usbhid)
Bus 002 Device 004 ID 8087:07dc control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 002 Device 015 ID 1a40:0101 control = auto, autosuspend_delay_ms =     0 -- Terminus Technology Inc. Hub (hub)
Bus 002 Device 019 ID 17ef:304e control = on,   autosuspend_delay_ms =  2000 -- Lenovo  (snd-usb-audio, usbhid)
Bus 002 Device 018 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 014 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```
Plugging the Ethernet cable back in yields:
```
--- TLP 0.8 --------------------------------------------

+++ USB
autosuspend        = enabled
device whitelist   = (not configured)
device blacklist   = (not configured)
wwan blacklist     = enabled

Bus 001 Device 004 ID 056a:00ec control = on,   autosuspend_delay_ms =  2000 -- Wacom Co., Ltd  (usbhid)
Bus 001 Device 003 ID 0483:91d1 control = on,   autosuspend_delay_ms =  2000 -- STMicroelectronics Sensor Hub (usbhid)
Bus 001 Device 002 ID 8087:8000 control = auto, autosuspend_delay_ms =     0 -- Intel Corp.  (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 003 Device 014 ID 17ef:304f control = auto, autosuspend_delay_ms =  2000 -- Lenovo  (r8152)
Bus 003 Device 013 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 012 ID 2109:0812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 003 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 3.0 root hub (hub)
Bus 002 Device 009 ID 04f2:b39f control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd  (uvcvideo)
Bus 002 Device 006 ID 04f3:0254 control = on,   autosuspend_delay_ms =  2000 -- Elan Microelectronics Corp.  (usbhid)
Bus 002 Device 004 ID 8087:07dc control = auto, autosuspend_delay_ms =  2000 -- Intel Corp.  (btusb)
Bus 002 Device 015 ID 1a40:0101 control = auto, autosuspend_delay_ms =     0 -- Terminus Technology Inc. Hub (hub)
Bus 002 Device 019 ID 17ef:304e control = on,   autosuspend_delay_ms =  2000 -- Lenovo  (snd-usb-audio, usbhid)
Bus 002 Device 018 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 014 ID 2109:2812 control = auto, autosuspend_delay_ms =     0 -- VIA Labs, Inc. VL812 Hub (hub)
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
```
There seems to be no difference in here, so I am not sure if it helps you in any way.

linrunner, 2016-05-06:
Sorry, i think i confused you completely by changing the subject.

Actually there are two problems with your hardware:

A. DEVICES_TO_DISABLE_ON_LAN_CONNECT --> detected via network manager
B. DEVICES_TO_DISABLE_ON_DOCK --> detected via udev events and usb ids

Me asking for a "favour" was about B. B cannot work because TLP doesn't know about the usb id of your dock.

So let's finish the analysis of A. first:

* Dock your ThinkPad with the ethernet cable *not* plugged in
* Plugin the ethernet cable
* Check if wifi gets disabled

zunder, 2016-05-06:
Now I see what you were after. Here is what I have to offer:

- Docked my ThinkPad with the ethernet cable _not_ plugged in -> WiFi connection on start up
- Pluged in the ethernet cable -> WiFi is still on, Ethernet connection not active (so basically nothing happened)

Did you expect that behavior?

linrunner, 2016-05-06:
Not really, just wanted to have a proper test setup.

In the meantime i spotted something odd in your above output:
> DEVICE           TYPE      STATE      CONNECTION 
> enx0050b66cf1b4  ethernet  unmanaged  -- 

This means NM doesn't handle your ethernet connection at all and as such tlp-rdw will never be called for status changes.

Please check your **/etc/network/interfaces** and remove all instances of "enx0050b66cf1b4". Maybe you need to reboot to get out of the "unmanaged" state.

Then re-test: plug/unplug the ethernet cable while still docked.

zunder, 2016-05-06:
Sweet, I was already on my journey through google in order to fix long boot times due to a message: `a start job is running for raise network interfaces` and your suggestion just did fix that. :) I am wondering how it got there in first place.
Anyway, now Ethernet gets enabled and WiFi gets disabled as you were expecting. If I unplug Dock though, WiFi stays disabled.

linrunner, 2016-05-06:
As i said above: unplugging the whole dock won't work with 0.8.

We're not finished with A. yet: what happens when you just unplug the ethernet _cable from the dock_. That should enable wifi.

zunder, 2016-05-06:
Yes it did. So if I am not mistaken, 0.9 should provide the same correct behavior with udev? (meaning docking/undocking would disable/enable WiFi)

zunder, 2016-05-07:
Yes it did.

Regarding case B: if I am not mistaken, 0.9 should provide the same correct behavior with udev? (meaning docking/undocking would disable/enable WiFi) If so, I guess this issue can be closed for now.

linrunner, 2016-05-07:
I'd like to test the changes concerning your hardware *before* the 0.9 release. I've uploaded packages for 16.04 [here](http://download.linrunner.de/packages/):

* **tlp_0.8.900-1~sid_all.deb**
* **tlp-rdw_0.8.900-1~sid_all.deb**

In fact the docking/undocking scenario consists of two test cases that must be distinguished carefully via the configuration:

A2. Detect ethernet disconnect/connect via network manager when docking/undocking

A2.1 Change your config like this

    DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi"
    DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi"
    #DEVICES_TO_DISABLE_ON_DOCK="wifi"
    #DEVICES_TO_ENABLE_ON_UNDOCK="wifi"

A2.2 Dock your ThinkPad and connect the ethernet cable
A2.3 Disable wifi (`wifi off`)
A2.4 Undock ThinkPad and check if wifi is enabled
A2.5 Dock ThinkPad and check if wifi is disabled

Report results of A2.4 and .5

B. Detect docking/undocking directly via udev

B.1 Change your config like this

    #DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi"
    #DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi"
    DEVICES_TO_DISABLE_ON_DOCK="wifi"
    DEVICES_TO_ENABLE_ON_UNDOCK="wifi"

B.2 Dock your ThinkPad and connect the ethernet cable
B.3 Disable wifi (`wifi off`)
B.4 Undock ThinkPad and check if wifi is enabled
B.5 Dock ThinkPad and check if wifi is disabled

Report results of B.4 and .5

zunder, 2016-05-07:
A2.4 WiFi is enabled
A2.5 WiFi is disabled

B.4 WiFi is enabled
B.5 WiFi is disabled

I did have some issues though. On first try A2.5 WiFi was still enabled (I did a restart beforehand), on second try it worked as expected.

Having B caused some issues on start up, something regarding USB3.0, it kept prompting a message in the shell. It did go away on next start up. However, I am having troubles restarting having B enabled. The screen stays black and I have to turn it down manually.

linrunner, 2016-05-09:
> On first try A2.5 WiFi was still enabled (I did a restart beforehand)
A restart between A2.4 and A2.5 or when?

On my own systems Network Manager shows the following quirk: after system start and TLP disabling Wifi, NM re-enables Wifi but only the rfkill switch, it doesn't connect. Does this resemble your symptom? I consider this behaviour a bug in NM, i can't (and do not want) to work around it in TLP.

Concerning your "troubles" with B: i don't believe TLP can cause blackscreens. It never has in 5 years lifetime. 
> it kept prompting a message in the shell.
Which shell?

Apart from your temporary, non-persistent symptoms i conclude that all test cases worked as designed. 

For *your* use case i suggest these settings e.g. case B disabled:

    DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi"
    DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi"
    #DEVICES_TO_DISABLE_ON_DOCK=""
    #DEVICES_TO_ENABLE_ON_UNDOCK=""

I use this setup myself. As a side effect it disables Wifi when the system starts docked, you don't need "wifi" in DEVICES_TO_DISABLE_ON_STARTUP.

zunder, 2016-05-09:
> A restart between A2.4 and A2.5 or when?
Before A2.4. But as you said, I do not find this very important (as it is non-persistent as you said) and agree that this might be caused by some side effects with NM or something else. I am happy with the solution as it is right now and did also decide for case A, being more general solution.

linrunner, 2016-05-09:
I like happy users :-)

linrunner, 2016-08-19:
Released with 0.9

closed 2016-08-19 by linrunner