linrunner/TLP #199: Intel graphics not supported

adamos101, 2016-05-07:
I am running Sandy Bridge Mobile  with Intel HD 3000 Graphics built into CPU. When using the TLP my power consumption remains high and the CPU very often goes up to 3.1GHz when on battery even if the pstate driver is setup to use only 32%.
I found this is because when Firefox, KDE are using graphics (lot's of things is happening with graphics hardware today), the CPU goes up, but when doing e.g. compression (which is only CPU and not GPU intensive), the power consumption remains low (so does the CPU frequencies as well).

I found this table in `/sys/kernel/debug/dri/0/i915_ring_freq_table`:

```
GPU freq (MHz)  Effective CPU freq (MHz)        Effective Ring freq (MHz)                                                 
650             800                             0                                                                         
700             800                             0                                                                         
750             2200                            0    
800             2300                            0
```                
Note that this table is presented based on the current setting i915_max_freq, so if the max_freq is 750, the 800 option will not be shown in table.

After doing `echo 750  >/sys/kernel/debug/dri/0/i915_max_freq`, the CPU frequency (as reported by i7z), is not going very high and not even hitting specified 2.2GHz. the 750 setting is a lot better than the 800 one.

The behavior of the graphics can be tested here: [](http://www.kevs3d.co.uk/dev/shaders/)
These demos can test maximum utilization of the graphics driver so while running these demos, you can check frequency with i7z (thus on newer kernels I think this utility is no longer required as the frequency is now properly reported by kernel in procfs).

Note htat the CPU scaling does not only happen when 3D is involved, this happens during normal Firefox and KDE operation. Firefox is using graphics when opening new tabs and rendering pages, which results also in spikes up to 3.1GHz. KDE sometimes does too many refreshes on some controls and it also causes spikes up to 3.1GHz, and during that time the CPU itself is not used but GPU and in this condition it draws over 40W of power.

linrunner, 2016-05-10:
Hi, 

thanks for your research. I take this as a feature request :-)

I wonder how to *briefly* explain the feature to the "ordinary user" in the configuration docs for TLP. How does she find out a suitable frequency value for her particular processor hardware?

Mancooo, 2017-01-02:
Hi, 

I also have a sandybridge laptop (with HD3000), I can confirm the issue as described by the OP. Is this feature implemented yet in TLP?

Btw thanks for this great program, it is one of the first things I install on my laptops!

I think a regular user would first have to sudo cat /sys/kernel/debug/dri/0/i915_ring_freq_table
and then do echo X >/sys/kernel/debug/dri/0/i915_max_freq . Users can only achieve the "best" setting by testing with for example Powertop.

This falls in the advanced category and obviously should be enabled by the user instead of standard.

Mancooo, 2017-01-02:
FYI

I did some extensive testing these couple of hours with the provided website in the OP (with some of the aggressive power saving features that are not recommended) by doing:

echo 650 >/sys/kernel/debug/dri/0/i915_max_freq  (the lowest possible in my case)

with the following in my i915.conf:

options i915 enable_rc6=7 semaphores=1 enable_psr=1 enable_dc=2 modeset=1 enable_fbc=1

There was some artifacting with these extreme settings. I tested them one by one and found out that the option enable_fbc=1 was causing the artifacting (on my system). So I removed that and now I have:

options i915 enable_rc6=7 semaphores=1 enable_psr=1 enable_dc=2 modeset=1

The system is rock solid on the lowest GPU Freq with full battery saving (CPU_MAX_PERF_ON_BAT=65 and no turbo and powersaving governor etc.)

There is definitely some powersaving and gains to be made by lowering the GPU freq. There is not much impact on the system, it would be awesome to implement this in TLP.

linrunner, 2017-01-02:
How much (mW) does it save **in Idle** compared to the default setting?

Mancooo, 2017-01-02:
I will test it (with Powertop) in idle and report back, but I don't know if it is supposed to save battery in idle, isn't it supposed to save battery when on (light/medium) GPU load? 

Like I tested it with watching (a small part) of a 1080p movie with MPV and hardware acceleration. With the lowest GPU freq. the movie plays fine and the desktop experience is still smooth. 

But I do see that the freq. does not go above CPU_MAX_PERF_ON_BAT=65 whereas with GPU freq. at 1300 Mhz it does and there basically no point of setting CPU_MAX_PERF_ON_BAT lowering than 100 in the case of GPU operations.

Let me first test it.

Mancooo, 2017-01-02:
Ok, I did quite some testing. All tests were done with the laptop (Dell Latitude e6320 with an iGPU HD3000) on battery, wifi off, brightness low and TLP bat mode and tested with Powertop (which I recalibrated). 

During the tests I **only** changed the GPU freq (and nothing else) with:

`echo 650 >/sys/kernel/debug/dri/0/i915_max_freq `(the lowest possible in my case) and
`echo 1300 >/sys/kernel/debug/dri/0/i915_max_freq `(the default state in my case).

1. Idle
I let the laptop be idle for approx. 10 min. in both cases, as I expected, I got the same results with 650 GPU freq. and 1300 GPU freq. Both were averaging between 6.22 W and 6.24 W.
So changing the GPU freq. did nothing for idle. I expected this.

2. Under load
I watched a 10 min. part of the movie Heat (the bank shootout scene) in both cases (the file is 1080p h.264). In this test I got a **significant** difference. 

With the standard GPU freq. of 1300 Mhz, the highest power usage I noted was 11.3 W and the lowest was 10.8 W.

With the GPU freq set at 650 Mhz, the highest was 10.5 W and the lowest was 9.67 W.

So in conclusion I can definitely say that there is at least 0.5 W or more powersaving possible by reducing the GPU freq. The reduction of the GPU freq. absolutely does not impact the desktop smoothness. 

I even tested the GPU at 650 Mhz while watching a downloaded clip (with youtube-dl) from Youtube that was 1080p at 60 frames. The desktop was smooth during playback! and I got no skipped frames (I loaded the movie on the cli with MPV).

So I think TLP should really have this feature, it will provide real world battery savings with no real negative and noticeable side effects.

I hope I helped and I hope you will implement this feature. I think the user should be able to set a GPU freq on bat and ac.

waldo99, 2018-05-31:
Thanks Mancooo for the suggestion! I hope this gets implemented in tlp.

I'm on a T440s with the iGPU. This is an interesting feature, which I tested a bit, as well.
I don't believe `echo 650 >/sys/kernel/debug/dri/0/i915_max_freq` correctly sets the max speed. You need to set the boost speed as well. I found 500 mhz worked for me after looking at the freq table.
This is done via (after sudo -s):
```
echo 500 > /sys/class/drm/card0/gt_boost_freq_mhz
echo 500 > /sys/class/drm/card0/gt_max_freq_mhz
```
Or this works, as well: 
```
echo 500 | sudo tee /sys/class/drm/card0/gt_boost_freq_mhz
echo 500 | sudo tee /sys/class/drm/card0/gt_max_freq_mhz
```
If you only set i915_max_freq, it clocks up to the boost speed.
I monitored it with `watch -n1 grep \"MHz\" /sys/kernel/debug/dri/0/i915_frequency_info`

I found the solution on the following forum: [stackexchange](https://unix.stackexchange.com/questions/404049/how-to-stop-intel-hd4000-from-running-at-max-clock-speed)

TiGR, 2018-08-02:
Having exact same problem. The problem is: it seems that GPU frequencies control take priority over intel_pstate's CPU control, at least on SandyBridge processors. So, when system starts using GPU, CPU frequency cranks up to turbo boost frequencies, rendering any CPU-related settings in TLP almost useless. And you don't actually need to do something GPU-intensive. Just open [this](https://tetris.com/play-tetris) in chrome (no need to start the game).

So, if you open that page, or anything else similar in battery mode, and you have turbo boost disabled and max frequencies set to minimum (~800MHz in my case), you'll notice that the CPU frequency is constantly in turbo boost ranges (2900MHz+).

Having no control over turbo or max frequency means you can't limit CPU overheating and get all sorts of problems.

Kudos to everyone in this thread, that's the only place where I've found solution to my overheating problem and intel_pstate seemingly not working at all.

linrunner, 2018-12-14:
Done.

adamos101, 2018-12-18:
I wrote original post long time ago and the kernel parameters since then have changed. Now I have these in /sys/devices/pci0000:00/0000:00:02.0/drm/card0.

```
[root@localhost card0]# ls -l gt*
-r--r--r-- 1 root root 4096 Dec 18 13:48 gt_act_freq_mhz
-rw-r--r-- 1 root root 4096 Dec 18 13:49 gt_boost_freq_mhz
-r--r--r-- 1 root root 4096 Dec 18 13:48 gt_cur_freq_mhz
-rw-r--r-- 1 root root 4096 Dec 18 13:49 gt_max_freq_mhz
-rw-r--r-- 1 root root 4096 Dec 18 13:48 gt_min_freq_mhz
-r--r--r-- 1 root root 4096 Dec 18 13:48 gt_RP0_freq_mhz
-r--r--r-- 1 root root 4096 Dec 18 13:48 gt_RP1_freq_mhz
-r--r--r-- 1 root root 4096 Dec 18 13:48 gt_RPn_freq_mhz

```

And both gt_boost_freq_mhz and gt_max_freq_mhz need to be configured to make it work.
The frequency table is still in /sys/kernel/debug/dri/0/i915_ring_freq_table

linrunner, 2018-12-18:
@adamos101 : **/sys/devices/pci0000:00/0000:00:02.0/drm/card0** should be equivalent to **/sys/class/drm/card0** and @jsalatas 's implementation uses the latter. Please test.

jsalatas, 2018-12-18:
Indeed, I used /sys/class/drm/card0 based on the https://github.com/jmechnich/intel-power-control 
Not sure if the path mentioned by @adamos101  is different. At least the current path (/sys/class/drm/card0) seems to work for me :\

TiGR, 2018-12-19:
BTW, sometimes my system boots in some different way, so that Intel card becomes `card1` (there is Radeon card also). Not sure why this happens, but it happens sometimes. So, hardcoded path won't work I suppose. I noticed that since my code did not work every time, so I had to find out why.

jsalatas, 2018-12-19:
Hmmm... So probably we need to check if the card0 is indeed an intel one :\

linrunner, 2018-12-19:
@TiGR : hybrid graphics is always fun ;-) 

Is i915_ring_freq_table at /sys/kernel/debug/dri/**1** then?

jsalatas, 2018-12-19:
Maybe something like the patch I committed in my forked repo? :\

@TiGR maybe you can give it a try?

adamos101, 2018-12-19:
OK it's in the /sys/class/drm/card0

TiGR, 2018-12-19:
> Is i915_ring_freq_table at /sys/kernel/debug/dri/**1** then?

Yes.

@jsalatas your code seems to be working. At least it did change the corresponging values after tlp start. Can't figure how to stop it (in order to reset values).

BTW, when I boot up system after shutdown - it is `card0`, but after reboot it may vary.

linrunner, 2018-12-19:
@jsalatas : thanks. I decided to do some refactoring.

linrunner, 2018-12-19:
@TiGR : sorry, could you test my variant too and provide 

    tlp-stat -g

please?

jsalatas, 2018-12-19:
> @jsalatas : thanks. I decided to do some refactoring.
@linrunner  Was expecting you to do so and that's why I didn't do a pull request :)
Thanks!

closed 2019-01-05 by linrunner

linrunner, 2019-03-11:
Released with TLP 1.2.