linrunner/TLP #203: [feature request] explicit per-chip default

mattia-b89, 2016-05-16:
under intel graphics parameters,
show the TRUE value of the parameter instead of "per-chip default"

linrunner, 2016-05-16:
How do you propose to get that "true value", dear?

mattia-b89, 2016-05-16:
unfortunately, I don't know...
I tried to look under /sys/... with no luck
I think if we are not able to check it "downstream" the hardware 
we can read it "upstream" in kernel file that will contain a table or a list of the selected hardware; in this way the "check" is not real we need to take care that user hasn't added a kernel parameter.
these is my idea and it is rough
I think this (and other) variable is exposed somewhere but I don't know where, maybe using `intel-gpu-tools` is the answer

linrunner, 2016-05-16:
Can't implement "somewhere" or "maybe". Sorry. If I knew how it would be implemented long ago. You#re welcome to do some more research.

mattia-b89, 2016-05-16:
yes, I try to contact an intel developer

mattia-b89, 2016-05-23:
(intel developers haven't answered me) but I was googling and I found this bug report:
https://bugs.freedesktop.org/show_bug.cgi?id=95176

that can be useful for this purpose;
e.g. `sudo cat /sys/kernel/debug/dri/0/i915_fbc_status`

linrunner, 2021-04-09:
Never say never ...

closed 2021-06-07 by linrunner